evcc-io/evcc #9941: Taycan invalid captcha value

normanmaurer, 2023-09-17:
### Describe the bug

Seit gestern wird bei meine Taycan immer "(offline)" im Namen mit angezeigt.

Bis vor dem Wochenende ging noch alles. Habe auch auf die neuste Version. Denke das Problem ist mit dem Captcha (siehe log).

### Steps to reproduce

Ist immer der Fall.

### Configuration details

```yaml
- name: taycan
    type: template
    template: porsche
    title: Porsche Taycan
    user: xxxx
    password: xxxxx
    vin: xxxx
    capacity: 93
```

### Log details

```text
ERROR 2023/09/17 20:27:31 creating vehicle taycan failed: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: Invalid captcha value
evcc    | [site  ] WARN 2023/09/17 20:27:31 vehicle 'Porsche Taycan (offline)' does not support automatic detection
```

### What type of operating system are you running?

Docker container

### Version

0.120.0

normanmaurer, 2023-09-17:
Habe auch gerade fest-gestellt das er nicht nach einem captcha fragt wenn ich mich via Webinterface einloggen will. Vielleicht ist das neu ?

andig, 2023-09-17:
Codeseitig hat sich ausser Token Refresh nichts geändert. Funktioniert ioBroker mit Porsche Plugin bei Dir?

normanmaurer, 2023-09-17:
@andig der Fehler war auch vor dem update da... somit denke nicht nicht das es was mit einer Änderung zutun hat. Vielleicht hat Porsche nun ein captcha eingefügt ?

normanmaurer, 2023-09-17:
ioBroker hab ich nicht

andig, 2023-09-17:
Dann wäre das jetzt der richtige Zeitpunkt. Wenn es keine funktionierende Referenz gibt gibts hier leider nichts zu tun.

normanmaurer, 2023-09-18:
@andig das captcha scheint nur zu kommen wenn ich mich über meine Internetleitung anmelde. Wenn ich über LTE gehe kommt das nicht... Ich frage mich ob evcc vielleicht zuviele "anfragen" gemacht hat und somit da captcha zur Sicherheit aktiviert wurde. Würde dann wohl wieder wegfallen sobald sich meine IP ändert.

andig, 2023-09-18:
Dann schliessen wir hier. Den Prozess können wir leider nicht beeinflussen. Wenn die Daten stimmen gibts genau 1 Anfrage alle 12 Stunden- das sollte egtl. kein Problem sein.

closed 2023-09-18 by andig

normanmaurer, 2023-09-26:
Nur zur info falls jemand das gleiche problem bekommt.... Irgendwann ist das captcha dann wieder verschwunden. Also einfach abwarten