evcc-io/evcc #9942: After setting maxCurrent via the API, it immediately reverts to the default (maximum)

dpaeme, 2023-09-17:
### Describe the bug

Hi EVCC team,

I've encountered a bug with the platform. I've tried setting the maxCurrent via the API, but it keeps reverting to the default setting after a couple of seconds.

The description is below.

Thanks for taking a look!

### Steps to reproduce

1. Check the State of EVCC via `% curl -X GET https://evcc/api/state`
```
...
"enabled": true,
"guardAction": "inactive",
"guardRemaining": 0,
"maxCurrent": 32,
"minCurrent": 2.2,
"minSoc": 0,
"mode": "now",
"phaseAction": "inactive",
"phaseRemaining":0,
"phasesActive": 3,
...
```
You can see that maxCurrent is at 32A

2. Change the maxCurrent via the API:
```
% curl -X POST https://evcc/api/loadpoints/1/maxcurrent/12
{"result":12}
```
The result looks good, so you'd expect that the maxCurrent would be set to 12 after this.

3. After a couple of seconds, you call the "State" again: `% curl -X GET https://evcc/api/state`
```
...
"guardRemaining": 0,
"maxCurrent": 32,
"minCurrent": 2.2,
...
```
And we're back to 32A.

### Configuration details

```yaml
log: info

chargers:
  - name: PMCC
    type: template
    template: pmcc
    ski: **redacted**
  
vehicles:
  - type: template
    template: porsche
    user: "**redacted**"
    password: "**redacted**"
    vin: **redacted**
    targetsoc: 85
    capacity: 84
    phases: 3
    name: Taycan

meters:
  - name: Fluvius
    type: custom
    power:
      source: http
      uri: **redacted**
      method: GET
      jq: .current_energy
  - name: SMA
    type: template
    template: sma-inverter
    usage: pv
    host: **redacted**
    password: **redacted**

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.39
  feedin:
    type: fixed
    price: 0.18

eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      **redacted**
      -----END CERTIFICATE-----
      
    private: |
      -----BEGIN EC PRIVATE KEY-----
      **redacted**
      -----END EC PRIVATE KEY-----
      
plant: **redacted**

site:
  - title: Home
    meters:
      grid: Fluvius
      pv: SMA

loadpoints:
  - title: Buiten
    charger: PMCC
    vehicle: Taycan
    mode: pv
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Version

0.120.0

andig, 2023-09-17:
Please add logfile. What is the expected vs the actual behavior?

@DerAndereAndi does the pmcc set any limits by itself?

dpaeme, 2023-09-17:
Hi @andig,

Expected behaviour would be that EVCC keeps the parameter as set via the API.

Logs: what level would you want them as? Debug?

DerAndereAndi, 2023-09-17:
You need to set the value on the connected vehicle instead. Setting it on the load point won’t work with this charger. 

If your Taycan supports Plug&Charge and it is enabled, then SoC is also available via the charger and you don’t need to use the Porsche online API. Use the iso15118 vehicle template for that. 

And as another note: the SKI of the PMCC is no critical information and does not need to be redacted. When providing log data it is helpful to know which one is configured