evcc-io/evcc #9948: Charging does not stop with openWB in PV Mode

MarkusGH, 2023-09-18:
### Describe the bug

When charging stops in PV mode due to PV timer expired, charging disable command is sent to openWB, but due to delayed response charging resumes immediately.

### Steps to reproduce

1. Use openWB
2. Use PV mode
3. EV is charging, PV power drops below threshold
4. Charging stops, then resumes immediately

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: localhost
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 3s # control cycle interval

log: info
levels:
  porsche: trace
  lp-1: debug
  lp-2: debug
  openwb: trace

mqtt:
  broker: mqtt.xxx.de:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.1570 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- name: grid1
  type: movingaverage
  decay: 1 # 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power
      scale: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    energy:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
[... 187 lines cut ...]
Sep 18 12:11:14 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:14 ----
Sep 18 12:11:14 homebridge evcc[1771769]: [lp-1  ] DEBUG 2023/09/18 12:11:14 charge power: 0W
Sep 18 12:11:14 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:14 pv power: 3794W
Sep 18 12:11:14 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:14 grid power: 1130W
Sep 18 12:11:14 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:14 grid powers: [-1120 1038 -1049]W
Sep 18 12:11:14 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:14 grid currents: [-3.5 4.5 -4.9]A
Sep 18 12:11:14 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:14 site power: 1130W
Sep 18 12:11:17 homebridge evcc[1771769]: [openwb] TRACE 2023/09/18 12:11:17 recv openWB/lp/2/W: '1783'
Sep 18 12:11:17 homebridge evcc[1771769]: [openwb] TRACE 2023/09/18 12:11:17 recv openWB/lp/2/APhase1: '7.8'
Sep 18 12:11:17 homebridge evcc[1771769]: [openwb] TRACE 2023/09/18 12:11:17 recv openWB/lp/2/kWhCounter: '3072.352'
Sep 18 12:11:17 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:17 ----
Sep 18 12:11:17 homebridge evcc[1771769]: [lp-1  ] DEBUG 2023/09/18 12:11:17 charge power: 1783W
Sep 18 12:11:17 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:17 pv power: 3792W
Sep 18 12:11:17 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:17 grid power: 1145W
Sep 18 12:11:17 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:17 grid powers: [-1120 1031 -1056]W
Sep 18 12:11:17 homebridge evcc[1771769]: [site  ] DEBUG 2023/09/18 12:11:17 grid currents: [-3.5 4.5 -4.9]A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.5 (8abaea79)

VolkerK62, 2023-09-18:
>but due to delayed response charging resumes immediately.

`interval: 3s # control cycle interval`

Hast du es mal mit einem größeren Intervall probiert?
3s ist schon sehr sportlich.

MarkusGH, 2023-09-18:
Ich wusste dass das kommt ;-)
Ich kann das testen, aber dennoch stellt sich die Frage, ob die an der Stelle kürzlich geänderte Logik "wasserdicht" ist oder angepasst werden sollte. Werde mit aktueller Version nachtesten, gerade eben war der Fehler weg.
Eventuell hat ja https://github.com/evcc-io/evcc/pull/9932 das Problem behoben.

MarkusGH, 2023-09-18:
Ich würde meinen, dass das die Folge einer Race Condition ist, bei der der vermeintliche oder echte Status des Chargers auf den Loadpoint übernommen wird. Wenn der Charger z.B. trotz vorherigem Enable(false) das Laden noch nicht beendet hat meldet er "Charging" und "Enabled". Im Falle der openWB bleibt der "Enabled" State sogar auf Ebene des Chargers erhalten wenn das Laden schon beendet wurde.
Vorgeschlagene Korrektur: https://github.com/evcc-io/evcc/pull/9959

andig, 2023-09-21:
Hast du mal probiert, ob ein normal großes Intervall das Problem auch löst?

GrimmiMeloni, 2023-09-21:
#9932 hat das Problem nur in eine Richtung behoben, nämlich für den Fall das der Charger lädt obwohl er disabled ist. Ich war hier primär von einer Restart Situation bei laufendem Charge ausgegangen. Dabei ist es dann OK, den Charger nochmal zu aktivieren, weil es nichts verändert.
Markus hat hier ein anderes mögliches Szenario für den Zustand gefunden. Der Charger schafft es nicht bis zum nächsten Regelinterval physikalisch abzuschalten. Dann haben wir das gleiche Bild wie beim Restart: Charger lädt (Status() liefert C) und gleichzeitig ist er disabled. #9932 enabled() dann aber wieder.

Ich habe mir #9959 angeschaut. Der Change am Loadpoint sieht erstmal richtig aus. Der Loadpoint.enabled folgt dem Charger dann nur nach Ablauf der Grace Period. Ich denk das ist OK. Allerdings hatte ich jetzt keine Zeit das Zusammenspiel mit Loadpoint.setLimit() zu durchdenken. Testen war um diese Uhrzeit ebenfalls nicht drin. 😉

MarkusGH, 2023-09-22:
Danke - freut mich dass ich mit meinen Überlegungen durchdringe. Ich habe gerade im Loadpoint noch eine Änderung gemacht, denn wenn am Charger Enabled und Charging auseinanderfallen gibt es dieselbe Race Condition. Sei so nett und schau Dir das auch mal an ob das so passen könnte.

MarkusGH, 2023-09-22:
So sieht das dann übrigens in Grafana aus
[image]
Man beachte auch dass die homePower jedes mal beim Umschalten auf 0 geht.

VolkerK62, 2023-09-22:
Netzbezug reagiert viel schneller, als der Carport

MarkusGH, 2023-09-22:
Genau - deswegen der PR https://github.com/evcc-io/evcc/pull/10011

closed 2023-10-01 by andig