evcc-io/evcc #9951: OpenWB Software 2.0 lässt sich nicht einbienden

chrisk982, 2023-09-18:
Hallo zusammen,

hat noch jemand das Problem das sich die openWB mit Software 2 nicht einbinden lässt?

cannot create charger 'wallbox4'
cannot create charger 'template'
cannot create charger 'openwb'
openWB loadpoint 1 is not configured

Meine Config der WB in evcc.ymal

chargers:  
- type: template  
  template: openwb   
  host: 192.168.1.136    
  name: wallbox4

StefanSchoof, 2023-09-18:
Ist das deine ganze Config? Der Abschnitt ist auf jeden Fall unvollständig und ohne komplette Config (als Code formatiert) ist es schwer zu sagen, wo das Problem ist

chrisk982, 2023-09-19:
`# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
meters:  
- type: template  
  template: fronius-gen24   
  usage: grid    
  host: 192.168.1.157    
  port: 502    
  name: grid1  
- type: template  
  template: fronius-gen24   
  usage: pv    
  host: 192.168.1.157    
  port: 502    
  name: pv2  
- type: template  
  template: fronius-gen24   
  usage: battery    
  host: 192.168.1.157    
  port: 502    
  name: battery3  
  
chargers:  
- type: template  
  template: openwb   
  host: 192.168.1.136    
  name: wallbox4  
  
loadpoints:  
- title: Garage  
  charger: wallbox4  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true  
  
site:  
  title: Mein Zuhause  
  meters:  
    grid: grid1  
    pv:  
    - pv2  
    battery:  
    - battery3  `

chrisk982, 2023-09-19:
hier dann mal meine Config. Ich habe die Ladepunkte auch bereits geändert aber es kommen immer die gleichen Fehler.

`# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
meters:  
- type: template  
  template: fronius-gen24   
  usage: grid    
  host: 192.168.1.157    
  port: 502    
  name: grid1  
- type: template  
  template: fronius-gen24   
  usage: pv    
  host: 192.168.1.157    
  port: 502    
  name: pv2  
- type: template  
  template: fronius-gen24   
  usage: battery    
  host: 192.168.1.157    
  port: 502    
  name: battery3  
  
chargers:  
- type: template  
  template: openwb   
  host: 192.168.1.136    
  name: wallbox4  
  
loadpoints:  
- title: Garage  
  charger: wallbox4  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true  
  
site:  
  title: Mein Zuhause  
  meters:  
    grid: grid1  
    pv:  
    - pv2  
    battery:  
    - battery3  `

StefanSchoof, 2023-09-19:
Siehe #9556, gibt gerade keine ausreichende Documentation von openwb für 2.0

chrisk982, 2023-09-19:
Danke für die Info

andig, 2023-10-21:
Mittlerweile ich die Modbus Doku veröffentlicht: https://openwb.de/main/wp-content/uploads/2023/10/ModbusTCP-openWB-series2-Pro-1.pdf. Damit lässt sich auch die 2.0 unterstützen.

closed 2023-10-21 by andig