evcc-io/evcc #9953: Phoenix Contact EM-CP-PP-ETH   Error: cannot create charger 'phoenix-em-eth': 1 error(s) decoding:  * '' has invalid keys: rtu

Sami0178, 2023-09-18:
### Describe the bug

Bei Version 0.120.0 bekomme ich den EM-CP-PP-ETH Controller leider nicht mehr zum laufen.
Selbst neu Konfiguration geht nicht.  (Downgrade auf Vorversion geht) 
Hat sich da etwas geändert?
Konfigversuch:
 Choose a charger: Phoenix Contact EM-CP-PP-ETH (Modbus TCP)
Please provide the following settings:
 Modbus ID 255
 IP address or hostname 192.168.2.12
 Port 502
Testing the configuration of Phoenix Contact EM-CP-PP-ETH (Modbus TCP) ...
   Error: cannot create charger 'phoenix-em-eth': 1 error(s) decoding:
* '' has invalid keys: rtu
 Testing Phoenix Contact EM-CP-PP-ETH (Modbus TCP) failed. Do you want to add it anyway? (y/N)    

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
  template: phoenix-em-eth 
  host: 192.168.2.12  
  port: 502  
  name: wallbox6
- type: template
  template: phoenix-em-eth 
  host: 192.168.2.13  
  port: 502  
  name: wallbox7

loadpoints:
- title: Ladepunkt 1
  charger: wallbox6
  vehicle: m3
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: false
  guardduration: 10m     # Schaltpause 10m

- title: Ladepunkt 2
  charger: wallbox7
  vehicle: id
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: false
  guardduration: 5m     # Schaltpause 5m
```

### Log details

```text
[main  ] INFO 2023/09/18 17:57:15 evcc 0.120.0
[main  ] INFO 2023/09/18 17:57:15 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/18 17:57:15 starting ui and api at :7070
[db    ] INFO 2023/09/18 17:57:15 using sqlite database: /home/pbjm11/.evcc/evcc.db
[main  ] FATAL 2023/09/18 17:57:16 cannot create charger 'wallbox7': cannot create charger 'template': cannot create charger 'phoenix-em-eth': 1 error(s) decoding:

* '' has invalid keys: rtu
[main  ] FATAL 2023/09/18 17:57:16 will attempt restart in: 5m0s
[server] FATAL 2023/09/18 17:57:16 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.0

StefanSchoof, 2023-09-18:
Ist im nightly gefixt, siehe #9913

Sami0178, 2023-09-26:
Läuft bei mir auch in Version 0.120.2 nicht. 
Muss bei V0.199.5 bleiben, sonst keine Modbus Kommunikation zu Phoenix EM-CP-PP-ETH Ladecontroller.

[lp-1  ] ERROR 2023/09/26 17:57:25 charger: read tcp 192.168.2.61:44396->192.168.2.12:502: i/o timeout

petosak, 2023-09-27:
> Läuft bei mir auch in Version 0.120.2 nicht. Muss bei V0.199.5 bleiben, sonst keine Modbus Kommunikation zu Phoenix EM-CP-PP-ETH Ladecontroller.
> 
> [lp-1 ] ERROR 2023/09/26 17:57:25 charger: read tcp 192.168.2.61:44396->192.168.2.12:502: i/o timeout

Same issue. I have to stay on 0.119.5

maikheinrich, 2023-09-27:
https://github.com/evcc-io/evcc/issues/9913#issuecomment-1722435803

Schaut mal hier...

petosak, 2023-09-27:
This work:
```
chargers:
  - name: Wallbox
    type: phoenix-ev-eth
    uri: 192.168.0.35:502

```
Thanks!

closed 2023-09-27 by andig