evcc-io/evcc #9954: Ladeleistung wird nicht reduziert

MarSlz, 2023-09-18:
### Describe the bug

Die Ladeleistung wird im Modus Min+PV nicht auf das Minimum reduziert bzw. Springt während der Ladung auf einmal auf das Max.

da meine PV Leistung zu klein ist, lass ich Über „residualPower: -1200 #“ 1200W Netzbezug zu (1200W Netzbezug + Max. 600W PV=> maximale Ladeleistung 1,8kW  somit sollte doch die Ladeleistung zwischen 1,4kW (min) und 1,8kW liegen.

Zu meinem Verständnis sollte im Modus PV das laden, dann bei weniger als 200W Pv Leistung unterbrochen werden, oder?

### Steps to reproduce

1. starte Ladung im Modus Min+ Pv
2. warten 
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3ODMxNjQ5MTQsImlhdCI6MTY4ODU1NjkxNCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6Ik1hclNseiJ9.lkfAFTk0DGizXFyyA86NK-oGFxWwBq8p6fKo_W-mlD4

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  ID.4: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
[... 201 lines cut ...]
[lp-2  ] DEBUG 2023/09/18 19:22:04 charge total import: 0.157kWh
[lp-2  ] DEBUG 2023/09/18 19:22:04 charger status: A
[lp-1  ] DEBUG 2023/09/18 19:22:10 set charge mode: off
[lp-1  ] DEBUG 2023/09/18 19:22:10 pv timer reset
[lp-1  ] DEBUG 2023/09/18 19:22:10 pv timer inactive
[site  ] DEBUG 2023/09/18 19:22:10 ----
[lp-1  ] DEBUG 2023/09/18 19:22:10 charge power: 10600W
[lp-2  ] DEBUG 2023/09/18 19:22:10 charge power: 0W
[site  ] DEBUG 2023/09/18 19:22:10 pv power: 4W
[site  ] DEBUG 2023/09/18 19:22:10 site power: 9396W
[lp-1  ] DEBUG 2023/09/18 19:22:10 charge currents: [15.4 15.5 15.4]A
[lp-1  ] DEBUG 2023/09/18 19:22:10 detected active phases: 3p
[lp-1  ] DEBUG 2023/09/18 19:22:10 charge total import: 1506.310kWh
[lp-1  ] DEBUG 2023/09/18 19:22:13 charger status: C
[lp-1  ] DEBUG 2023/09/18 19:22:13 soc estimated: 93.15% (vehicle: 92.00%)
[lp-1  ] DEBUG 2023/09/18 19:22:13 vehicle soc: 93%
[lp-1  ] DEBUG 2023/09/18 19:22:13 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/18 19:22:13 vehicle range: 359km
[lp-1  ] DEBUG 2023/09/18 19:22:19 charger disable
[lp-1  ] DEBUG 2023/09/18 19:22:19 wake-up timer: stop
```

### What type of operating system are you running?

HomeAssistant Add-on

VolkerK62, 2023-09-18:
Evtl ist ein CO2-Wert bei Smartes Netzladen gesetzt?
Klick mal um UI auf die g(ramm) Anzeige beim Netzbezug.

Gut wäre ein Log passend zu den Grafiken. 
https://docs.evcc.io/docs/guides/setup#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen
Aber, ich glaube, bei HA geht das nicht.

MarSlz, 2023-09-18:
Hallo Volker,

[image]

ich glaub das war ein guter Tipp :-).
Habe das mal ausgestellt und werde beobachten.

Vll. Lässt sich dass wenn aktiv, farblich hervorheben, sodass man eher daran denkt?

VolkerK62, 2023-09-19:
Allerdings ist das eigentlich nur im Modus `PV` aktiv.
Du hast doch aber den Modus `Min+PV` genutzt?

MarSlz, 2023-09-19:
Ja genau, war im Modus min+pv