evcc-io/evcc #9955: Vehicle is not woken up correctly

kscholty, 2023-09-18:
### Describe the bug

I am using a vehicle of type template saic2mqtt (it's an MG4)
The vehicle is not correctly woken up during the start sequence.
This happens both when initially connected (the vehicle is not woken up during detection) or when the charger interrupted charging due to insufficient PV.

My assumption is that this is caused by the code in `Loadpoint.Update()`

``` go
// Wake-up checks
	if lp.enabled && lp.status == api.StatusB &&
		int(lp.vehicleSoc) < lp.Soc.target && lp.wakeUpTimer.Expired() {
		lp.wakeUpVehicle()
	}
```
As we can see the wakeup is only executed in `StatusB ` and after the timeout expires.
The timeout is started in `loadpoint.setLimit()` when the carger gets enabled.
So directly after after enabling the charger, the timeout is not expired, so no wakeup is executed.
The Timeout will expire after 30s. When this happens the charger probably already changed to `StatusC`, so again no wakeup is executed.

### Steps to reproduce

1. Attach MG4 to charger while the gateway is not refrehing
2. The Gateway will not start refreshing and the vehicle will never be detected and SOC will never be correctly read
3.
...

### Configuration details

```yaml
The charger:

chargers:
- type: custom
  name: wallbox_custom
  tos: true
  status:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 100
      type: input
      decode: int16
  enabled:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 400
      type: coil
      decode: uint8
  enable:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 400
      type: writecoil
      decode: uint8
  maxcurrent:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 300
      type: writeholding
      decode: uint16
  phases1p3p:
    source: script
    cmd: /etc/switch-phases.sh ${phases}
    timeout: 15s

The Vehicle

- name: MG4
  title: MG4
  type: template
  template: mg2mqtt
  user: XXXXXXXXX
  vin: XXXXXXXXX
  capacity: 64
  phases: 3
  maxcurrent: 23
  targetsoc: 80
```

### Log details

```text
You can see here the logs of evcc and the saic2mqtt gateway
UInfortunately I have no debug output...

Sep 18 13:55:01 evcc evcc[121238]: [lp-1  ] INFO 2023/09/18 13:55:01 stop charging <-
Sep 18 13:55:25 evcc python3.11[146503]: 2023-09-18 13:55:25,797 Updating vehicle status
Sep 18 13:55:41 evcc python3.11[146503]: 2023-09-18 13:55:41,611 Updating charging status
Sep 18 13:55:57 evcc python3.11[146503]: 2023-09-18 13:55:57,410 Refreshing vehicle status succeeded...
Sep 18 13:55:57 evcc python3.11[146503]: 2023-09-18 13:55:57,539 0 messages received

*** One wakeup after stopping to charge ***
Sep 18 13:56:01 evcc python3.11[146503]: 2023-09-18 13:56:01,051 Setting refresh mode to force
Sep 18 13:56:01 evcc python3.11[146503]: 2023-09-18 13:56:01,421 Setting refresh mode to periodic
Sep 18 13:56:01 evcc python3.11[146503]: 2023-09-18 13:56:01,421 Updating vehicle status
Sep 18 13:56:17 evcc python3.11[146503]: 2023-09-18 13:56:17,718 Updating charging status
Sep 18 13:56:33 evcc python3.11[146503]: 2023-09-18 13:56:33,533 Refreshing vehicle status succeeded...
Sep 18 13:56:57 evcc python3.11[146503]: 2023-09-18 13:56:57,665 0 messages received
Sep 18 13:57:57 evcc python3.11[146503]: 2023-09-18 13:57:57,836 0 messages received
Sep 18 13:58:33 evcc python3.11[146503]: 2023-09-18 13:58:33,886 Updating vehicle status
Sep 18 13:58:50 evcc python3.11[146503]: 2023-09-18 13:58:50,183 Updating charging status
Sep 18 13:59:06 evcc python3.11[146503]: 2023-09-18 13:59:06,003 Refreshing vehicle status succeeded...
Sep 18 13:59:06 evcc python3.11[146503]: 2023-09-18 13:59:06,136 0 messages received
Sep 18 14:00:06 evcc python3.11[146503]: 2023-09-18 14:00:06,284 0 messages received
Sep 18 14:01:06 evcc python3.11[146503]: 2023-09-18 14:01:06,182 Updating vehicle status
Sep 18 14:01:22 evcc python3.11[146503]: 2023-09-18 14:01:22,494 Updating charging status
Sep 18 14:01:38 evcc python3.11[146503]: 2023-09-18 14:01:38,282 Refreshing vehicle status succeeded...
Sep 18 14:01:38 evcc python3.11[146503]: 2023-09-18 14:01:38,414 0 messages received
Sep 18 14:02:38 evcc python3.11[146503]: 2023-09-18 14:02:38,567 0 messages received
Sep 18 14:03:38 evcc python3.11[146503]: 2023-09-18 14:03:38,456 Updating vehicle status
Sep 18 14:03:54 evcc python3.11[146503]: 2023-09-18 14:03:54,866 Updating charging status
Sep 18 14:04:10 evcc python3.11[146503]: 2023-09-18 14:04:10,667 Refreshing vehicle status succeeded...

*** Gateway stops refresh since vehicle is no longer charging  ***

Sep 18 14:04:10 evcc python3.11[146503]: 2023-09-18 14:04:10,795 0 messages received
Sep 18 14:05:10 evcc python3.11[146503]: 2023-09-18 14:05:10,953 0 messages received
Sep 18 14:06:11 evcc python3.11[146503]: 2023-09-18 14:06:11,076 0 messages received
Sep 18 14:07:11 evcc python3.11[146503]: 2023-09-18 14:07:11,218 0 messages received
Sep 18 14:08:11 evcc python3.11[146503]: 2023-09-18 14:08:11,363 0 messages received
Sep 18 14:09:11 evcc python3.11[146503]: 2023-09-18 14:09:11,492 0 messages received

*** No wakeup after charging is resumed ***

Sep 18 14:09:30 evcc evcc[121238]: [lp-1  ] INFO 2023/09/18 14:09:30 start charging ->
Sep 18 14:10:11 evcc python3.11[146503]: 2023-09-18 14:10:11,621 0 messages received
Sep 18 14:11:11 evcc python3.11[146503]: 2023-09-18 14:11:11,802 0 messages received
Sep 18 14:12:11 evcc python3.11[146503]: 2023-09-18 14:12:11,939 0 messages received
Sep 18 14:13:12 evcc python3.11[146503]: 2023-09-18 14:13:12,068 0 messages received
Sep 18 14:14:12 evcc python3.11[146503]: 2023-09-18 14:14:12,219 0 messages received
Sep 18 14:15:12 evcc python3.11[146503]: 2023-09-18 14:15:12,346 0 messages received
Sep 18 14:16:12 evcc python3.11[146503]: 2023-09-18 14:16:12,471 0 messages received
Sep 18 14:17:12 evcc python3.11[146503]: 2023-09-18 14:17:12,609 0 messages received
Sep 18 14:18:12 evcc python3.11[146503]: 2023-09-18 14:18:12,734 0 messages received
Sep 18 14:19:12 evcc python3.11[146503]: 2023-09-18 14:19:12,862 0 messages received
Sep 18 14:20:13 evcc python3.11[146503]: 2023-09-18 14:20:13,024 0 messages received
```

### What type of operating system are you running?

Linux

### Version

0.120.0

andig, 2023-09-21:
Ich verstehe nicht was hier die Erwartung ist. Eine bestimmte Nachricht über MQTT? Welche?

/cc @tisoft

kscholty, 2023-09-21:
Die Erwartung ist ein Aufruf von 
`lp.wakeUpVehicle()` wenn mit dem Laden begonnen wird.

andig, 2023-09-21:
Das wird aufgerufen wenn nach Enable nicht innerhalb Zeit X ein Ladevorgang beginnt. OB es hier aufgerufen wird ist nicht ersichtlich da das Logfile fehlt.

kscholty, 2023-09-21:
Es ist ersichtlich, dass es nicht aufgerufen wird.
Man kann beim Abschalten sehen, dass dort einmal `Setting refresh mode to force` im logfile auftaucht. Das wird durch das (dort eigentlich unnötige) Wakeup ausgelöst. Das fehlt beim Start. Und du sagst ja selber, wakeup wird aufgerufen, wenn nach `enable` kein Ladevorgang startet. Der Ladevorgang startet ja, aber die Daten werden nicht aktualisiert, da das Gateway im Schlafmodus ist.

Das führt dann in diesem konkreten Fall irgendwann zu `outdated` Fehlern aus dem mqtt Plugin, da die Daten zu lange nicht aktualisiert wurden, so wie übrigens auch bei der Erkennung beim Anschließen des Autos (das andere issue).

andig, 2023-09-21:
Zeig uns bitte ein evcc log- das können wir lesen :)

andig, 2023-09-21:
Bzw: das Ziel von Wakeup ist das Laden. Genau das scheint ja- auch ohne Wakeup- zu funktionieren. Folgerichtig wird hier kein Wakeup aufgerufen.

andig, 2023-09-21:
> Man kann beim Abschalten sehen, dass dort einmal Setting refresh mode to force im logfile auftaucht. Das wird durch das (dort eigentlich unnötige) Wakeup ausgelöst.

das wäre ein Fehler, braucht aber ein evcc log file.

kscholty, 2023-09-21:
> Bzw: das Ziel von Wakeup ist das Laden. Genau das scheint ja- auch ohne Wakeup- zu funktionieren. Folgerichtig wird hier kein Wakeup aufgerufen.

Das Laden funktioniert, sonst aber leider vieles nicht, da keine Daten vorliegen.
Ich versuche mal ein debug Log herzustellen. Das wird aber auch nicht mehr zeigen, als das was wir jetzt schon wissen.

andig, 2023-09-21:
Das klingt nach einem Speziallfall mangels Api. Jedes andere Fahrzeug liefert Daten wenn es lädt und damit wach ist. Daher wontfix.

closed 2023-09-21 by andig

kscholty, 2023-09-21:
Schade. Dann ist aber auch das saic2mqtt Template recht nutzlos.

premultiply, 2023-09-21:
Ich hab es so verstanden hier ein Datenrefresh vom Fahrzeug benötigt wird.
Das was evcc unter Wakeup versteht ist etwas anderes.

kscholty, 2023-09-21:
Ja, so ist es. In diesem Fall hier soll nicht das Fahrzeug (oder die Wallbox) aufgeweckt werden, sondern die Datenverbindung des Fahrzeugs. Und das erwähnte Template nutzt die Funktion auch so, bzw. versucht es, was aber aufgrund der spezifischen Semantik der Wakeup-Funktion nicht funktioniert.