evcc-io/evcc #9956: Vehicle is not woken up in detection process

kscholty, 2023-09-18:
### Describe the bug

As also descirbed in #9955 I am using the saic2mqtt Gateway in order to connect an MG45 to evcc

When the car is sleeping while attached to the charger, it is not detected, since invalid (or even outdated) values are read.
#9955 causes the vehicle never to be activated even after charging started.
Also manually selecting the vehicle does not wake it up.

I propose to call  `WakeupVehicle()` in an appropriate position in the detection algorithm.

### Steps to reproduce

1. Connect sleeping MG4 to charger
2. No wakeup is issued
3.
...

### Configuration details

```yaml
See #9955
```

### Log details

```text
See also #9955

Sep 18 12:13:01 evcc evcc[121238]: [lp-1  ] INFO 2023/09/18 12:13:01 car connected
Sep 18 12:13:05 evcc python3.11[32191]: 2023-09-18 12:13:05,185 0 messages received
Sep 18 12:13:05 evcc evcc[121238]: [main  ] ERROR 2023/09/18 12:13:05 vehicle status: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/charging outdated: 17h7
m58s
Sep 18 12:13:32 evcc evcc[121238]: [main  ] ERROR 2023/09/18 12:13:32 vehicle status: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/charging outdated: 17h8
m24s
Sep 18 12:14:02 evcc evcc[121238]: [main  ] ERROR 2023/09/18 12:14:02 vehicle status: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/charging outdated: 17h8
m55s
Sep 18 12:14:05 evcc python3.11[32191]: 2023-09-18 12:14:05,312 0 messages received
Sep 18 12:14:31 evcc evcc[121238]: [main  ] ERROR 2023/09/18 12:14:31 vehicle status: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/charging outdated: 17h9
m23s
Sep 18 12:15:02 evcc evcc[121238]: [main  ] ERROR 2023/09/18 12:15:02 vehicle status: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/charging outdated: 17h9
m55s
Sep 18 12:15:05 evcc python3.11[32191]: 2023-09-18 12:15:05,440 0 messages received
Sep 18 12:15:31 evcc evcc[121238]: [main  ] ERROR 2023/09/18 12:15:31 vehicle status: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/charging outdated: 17h1
0m24s

**** Manually changed the vehicle, still no wakeup ***
Sep 18 12:15:45 evcc evcc[121238]: [lp-1  ] INFO 2023/09/18 12:15:45 vehicle updated: unknown -> MG4
Sep 18 12:16:00 evcc evcc[121238]: [lp-1  ] ERROR 2023/09/18 12:16:00 vehicle odometer: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/mileage outdated: 17h10m53s
Sep 18 12:16:01 evcc evcc[121238]: [lp-1  ] ERROR 2023/09/18 12:16:01 vehicle soc: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/soc outdated: 17h10m38s
Sep 18 12:16:05 evcc python3.11[32191]: 2023-09-18 12:16:05,583 0 messages received
Sep 18 12:17:05 evcc python3.11[32191]: 2023-09-18 12:17:05,709 0 messages received
Sep 18 12:18:00 evcc evcc[121238]: [lp-1  ] INFO 2023/09/18 12:18:00 start charging ->
Sep 18 12:18:00 evcc evcc[121238]: [lp-1  ] ERROR 2023/09/18 12:18:00 vehicle soc: saic/XXXXXXX/vehicles/XXXXXXXX/drivetrain/soc outdated: 17h12m37s
.
.
.
```

### What type of operating system are you running?

Linux

### Version

0.120.0

andig, 2023-09-21:
> Manually changed the vehicle, still no wakeup

Bei manueller Änderung wird nichts aufgeweckt- warum auch? Lass uns erstmal in das andere Thema Klarheit bringen.

closed 2023-09-21 by andig

kscholty, 2023-09-21:
Es geht hier auch mehr um den automatischen Erkennungsvorgang. Bei einem "schlafenden" Fahrzeug kann der aktuelle Zustand nicht abgefragt werden und das Fahrzeug wird nie erkannt. Und selbst wenn ich es manuell auswähle, weil es nicht erkannt wird, wird es nicht aufgeweckt, um den aktuellen Zustand abfragen zu können. 
Zusammen mit dem anderen Fehler wird es nie aufgeweckt, was die gesamte Einbindung nutzlos werden lässt.