evcc-io/evcc #9961: Dynamische Strompreise wie bei Smartem Laden mit einbeziehen

DavidNemecek, 2023-09-18:
**Is your feature request related to a problem? Please describe.**
Wenn smartes Laden mit einer Preisobergrenze aktiviert wurde, ist das Auto möglicherweise schon geladen, obwohl es später billigeren Strom gibt. 

**Describe the solution you'd like**
Wie bei der Ladeplanung sollten die billigsten Stunden herausgesucht werden und zur entsprechenden Zeit geladen werden.
Die Ladedauer ist schließlich bekannt.

**Beispiel**
Preisobergrenze für Smartes Laden: 20 Cent/kWh
benötigte Zeit zum Laden: 1 Stunden
Preise von 01:00 bis 05:00 (in Cent/kWh): 25, 15, 5, 20
aktuelles Verhalten: Laden von 01:00 bis 02:00
gewünschtes Verhalten: Laden von 03:00 bis 04:00

andig, 2023-09-20:
Dafür solltest du die Ladeplanung nutzen. Diese verwendet immer den günstigsten Preis.

tzatti1, 2023-09-20:
Kann so eine dynamische Planung auch in Kombination mit PV genutzt werden? Angenommen ich habe nur PV-Überschussladen aktiviert (1-Phasig), nun ist aber der Strompreis unter einem bestimmten Grenzwert und es sollte automatisch auf 11 kW 3-phasig springen, bis eben der Grenzwert wieder unterschritten wird, so dass wieder auf reines PV-Überschussladen umgestellt werden sollte. Ist das möglich?

naltatis, 2023-09-20:
@tzatti1 Ja automatischer Phasenwechsel bei PV Überschuss und Preisgenze (max Power) funktioniert zusammen. Er schaltet je nach Situation um.

@DavidNemecek Wie Andi sagt, denn du eine Abfahrtszeit setzt wird er günstigste Slot verwendet. Wenn keine Abfahrtszeit bekannt ist muss evcc davon ausgehen, dass du das Auto so schnell wie möglich gefüllt haben willst. In deinem Beispiel könnte es ja sein, dass du um 3 Uhr los möchtest. Hier können/sollten wir nicht anfangen zu raten.

DavidNemecek, 2023-09-20:
Naja, aber wenn ich um 3 Uhr losfahren möchte, würde ich ja genau um diese Uhrzeit die Abfahrtszeit setzen und die Ladeplanung verwenden. 
Aber ja, da habt ihr recht. Es würde zu Verwirrung führen, nachdem es ja einfach unter einer gewissen Grenze anfangen soll zu laden.

andig, 2023-09-20:
Klingt nach Konsens 👍🏻

closed 2023-09-20 by andig