evcc-io/evcc #9968: Alfen Single WB - Kein Identifier

89Amaterasu, 2023-09-19:
### Describe the bug

Ich versuche derzeit an meiner Alfen Single WB den übertragenen RFID Identifier zu finden. evcc charger liefert allerdings keinen Identifier. Ist das ein Fehler meinerseits, ein Bug, oder noch nicht implementiert?

`[main  ] INFO 2023/09/19 18:54:56 evcc 0.120.0
[main  ] INFO 2023/09/19 18:54:56 using config file: C:\Users\Svenja\Desktop\Neuer Ordner\evcc.yaml
[mqtt  ] INFO 2023/09/19 18:54:57 connecting evcc-423767314 at tcp://10.168.1.2:1883
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 234V 232V
Charge status:  A
Enabled:        true`

### Steps to reproduce

run 'evcc charger' in terminal

### Configuration details

```yaml
not needed
```

### Log details

```text
[main  ] INFO 2023/09/19 18:54:56 evcc 0.120.0
[main  ] INFO 2023/09/19 18:54:56 using config file: C:\Users\Svenja\Desktop\Neuer Ordner\evcc.yaml
[mqtt  ] INFO 2023/09/19 18:54:57 connecting evcc-423767314 at tcp://10.168.1.2:1883
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 234V 232V
Charge status:  A
Enabled:        true
```

### What type of operating system are you running?

Linux

### Version

0.120.0

premultiply, 2023-09-19:
Das kommt darauf an ob überhaupt und wenn ja wie und unter welchen Umständen die Box diese Infos via Modbus (oder OCPP) exportiert.
Müsste ich auch erst in der Doku nachschauen ob da eine theoretische Chance besteht mit der aktuellen Firmware daran zu kommen.

premultiply, 2023-09-19:
Nach Blick in die Modbus-Doku: Sieht schlecht aus.

Bliebe wohl nur noch die Variante OCPP zu testen oder mit Alfen in die Diskussion für eine entsprechende Protokollerweiterung einzusteigen.

premultiply, 2023-09-19:
Erstmal können wir hier nichts tun.
Daher schließe ich hier.
Sollte es weitere Erkenntnisse geben gerne nochmal hier weiter.

closed 2023-09-19 by premultiply