evcc-io/evcc #9970: Doku: template BMW/Mini missleading

VolkerK62, 2023-09-19:
yep, DAS ist es! 

  region: EU 
  - NA
  - EU 
  name: ev4
  
  ist kaputt. Wenn ich die Zeilen 2 und 3 lösche, geht der Start sauber durch!

_Originally posted by @stm9x9 in https://github.com/evcc-io/evcc/discussions/9946#discussioncomment-7049571_

VolkerK62, 2023-09-19:
Die Auswahlliste bei ´region` ist irreführend, weil sie in der Konfig zu Fehlern führt.

stm9x9, 2023-09-19:
ist das nur ein Doku-Issue? Ich hab das ja mit der configurate-Option erstellen lassen.

VolkerK62, 2023-09-20:
Danke für den Hinweis, ich dachte das wäre manuell konfiguriert worden.

andig, 2023-09-20:
Helft mir- was ist der konkrete Fehler?

VolkerK62, 2023-09-20:
Konfig wurde mit `configure` erstellt. 
Danach kommt der Fehler
```
While parsing config
yaml: line 47: did not find expected key
```
Löscht man
```
- NA
- EU
```
startet evcc ohne Fehler

VolkerK62, 2023-09-21:
Also doch "nur" ein Template-Fehler?

andig, 2023-09-21:
Ja- hab aber nicht versucht, den "schön" zu lösen sondern einfach die zusätzlich Funktion der Listenauswahl raus genommen. Dann gehts wieder.