evcc-io/evcc #9974: Feature Request: zusätzliche Grenzwerte zum Laden von Haus Akku und EV gleichzeitig

patsch9, 2023-09-20:
Ich habe ein Hybridwechselrichter mit Haus Akku welcher mit bis zu 5KW lädt. Wenn die Solarerzeugung jetzt zum Beispiel 6 KW Beträgt und der Haushalt 1KW Leistung bezieht habe ich keinen Überschuss und EVCC erkennt keinen Grund den Ladevorgang Automatisch zu starten. 

**Describe the solution you'd like**
Es wäre nützlich zusätzliche Grenzwerte für den Haus Akku angeben zu können um das Laden Automatisch zu starten wenn der Grenzwert zu Laden des Akku überschritten wird.
Das ganze könnte analog der aktuellen Batterieeinstellung gemacht werden (siehe Screenshot). Logisch müsste noch über die Abhängigkeiten zu den vorhanden Grenzwerten des Batteriegestützten Ladens nachgedacht werden.

**Beispiel:**

Ist:
Akkustand 60%
Solar Leistung: 5 KW
Haus Verbrauch:  500 W
Ladung Akku: 4,5 KW

Soll bei Akkustand über (30)%: 
Akkustand 60%
Solar Leistung: 5 KW
Haus Verbrauch:  500 W
-- Unterer Grenzwert Akku Ladung: 900 W --> Ab diesem Wert wird bei aktiver Ladung der EV Ladevorgang beendet. (Haus Akku Ladung bevorzugen)
-- Oberer Grenzwerk Akku Ladung: 3,7 KW --> Ab diesem Wert wird bei Bedarf die EV Ladung aktiviert. (EV Ladung bevorzugen)
Ladung Akku: 1,8 KW
Ladung EV: 2,7 KW 

Soll bei Akkustand unter (30)%: 
Akkustand 25%
Solar Leistung: 5 KW
Haus Verbrauch:  500 W
-- Unterer Grenzwert Akku Ladung: 900 W --> Ab diesem Wert wird bei aktiver Ladung der EV Ladevorgang beendet. (Haus Akku Ladung bevorzugen) 
-- Oberer Grenzwerk Akku Ladung: 3,7 KW --> Ab diesem Wert wird bei Bedarf die EV Ladung aktiviert. (EV Ladung bevorzugen)
Ladung Akku: 4,5 KW

[image]

Ergänzt das Thema gerne bei Bedarf um z.B. Variable Ladeleistung etc. Hier habe ich nicht so den Überblick, da mein PHEV nur mit den Ladeziegel und einem Relay gesteuert wird.

mdkeil, 2023-09-20:
Irgendwie verstehe ich deine Problematik nicht.. In deinem Bild.. wenn dein Hausakku > BufferSOC ist, sprich der Hausakku aus der Prio rausgefallen ist, startet der EV-Ladevorgang, egal ob der Speicher weiter lädt-- diese Ladeleistung des Hausakkus wird dann natürlich durch den WR reduziert..

patsch9, 2023-09-21:
> Irgendwie verstehe ich deine Problematik nicht.. In deinem Bild.. wenn dein Hausakku > BufferSOC ist, sprich der Hausakku aus der Prio rausgefallen ist, startet der EV-Ladevorgang, egal ob der Speicher weiter lädt-- diese Ladeleistung des Hausakkus wird dann natürlich durch den WR reduziert..

Ich meine das ganze so, dass aktuell das Laden nur beginnt wenn du über den Ladestand des Akkus kommst der in den Akku Einstellungen definiert ist. Das ganze hat aus meiner sich zwei Effekte:

1.
Wie in dem Screenshot zu sehen wird der Akku mit 5KW geladen und es ist eine Einspeisung von 1,2 vorhanden. Die Ladung erfolgt nicht, da der Akku noch nicht über (bei mir) 85% ist und nicht genug Überschuss ins Netz geht.

[image]
--
Mit den von mit vorgeschlagenen Grenzwerten könnte man das EV Laden nun (ab erreichen des oberen Grenzwertes) starten und der Haus Akku würde mit verringerter Leistung Laden. Zusätzlich würde bei nicht genügend PV Strom (unterer Grenzwerk für Laden des Haus Akku) das EV Laden beendet um das Laden des Haus Akku zu Priorisieren.

2.
Wenn ich den Wert für "Fahrzeug zuerst dann das Haus" jetzt Runter setze erreiche ich zwar, dass das EV Laden beginnt, es ist aber nicht sichergestellt, dass der Haus Akku Parallel auch Lädt. So kann es passiere, dass das EV zwar am Ende des Tages voll ist, aber der Haus Akku nur einen geringen Speicherstand hat. 
Mit dem von mir vorgeschlagenen unterem Grenzwert würde Sichergestellt, dass wenn der Haus Akku nicht einen gewissen Ladestand hat, immer eine Mindestmenge in den Haus Akku fliest bevor das EV Laden startet.

mdkeil, 2023-09-21:
Da dein Fahrzeug/Ladepunkt scheint ja auch einphasig laden zu können.. generell verstehe ich den Use-Case nun.. wobei ein relativ einfacher Workaround wäre mit einer externen Steuerung evcc in dieser Konstellation einfach auf min+pv zu setzen oder mit einer externen Steuerung die Ladeleistung des WR aktiv zu reduzieren.

patsch9, 2023-09-21:
> Da dein Fahrzeug/Ladepunkt scheint ja auch einphasig laden zu können.. generell verstehe ich den Use-Case nun.. wobei ein relativ einfacher Workaround wäre mit einer externen Steuerung evcc in [...]

Ladeleistung des WR lässt sich nicht aktiv Reduzieren, nur über Steigerung des AC Verbrauchs.

Über eine Externe Ansteuerung (z.B. Home Assistant) würde das natürlich funktionieren, setzt allerdings voraus das man so etwas einsetzt. Ich persönlich habe Home Assistant im Einsatz, denke aber zeitgleich das es eine Elegante Lösung wäre das ganze aus ECVV heraus zu machen um die Logik nicht an 3 verschiedenen stellen zu haben, denn wofür sollte ich dann noch EVCC nutzen, wenn ich die halbe Logik bereits im Home Assistant selber gebaut habe. 

Das ganze setzt natürlich voraus, dass andere auch Interesse an so einer Logik haben.

VolkerK62, 2023-09-21:
"Einspeisung vermeiden" ist eines von vielen Themen https://github.com/evcc-io/evcc/discussions/3530

Ich habe das z. Zt. extern gelöst, indem ich zwischen PV-Modus mit niedrigem prioSoc und Min+PV-Modus mit hohem prioSoc umschalte, gesteuert durch die Ladeleistung.
https://github.com/evcc-io/evcc/discussions/5239

patsch9, 2023-09-21:
> "Einspeisung vermeiden" ist eines von vielen Themen #3530
> 
> Ich habe das z. Zt. extern gelöst, indem ich zwischen PV-Modus mit niedrigem prioSoc und Min+PV-Modus mit hohem prioSoc umschalte, gesteuert durch die Ladeleistung. #5239

das spiegelt so ziemlich genau meinen Gedanken wieder und trifft genau den Punk "noch smarteres laden". Gerne meinen FRQ verlinken zu den Themen. Sehe das Thema hat doch schon mehrfach Beachtung/Interesse gefunden.

andig, 2023-09-21:
Können wir die Diskussion in einem der verlinkten Themen weiter führen? Gerne den relevanten Teil von hier übernehmen. Danke!

patsch9, 2023-09-22:
Ich habe mein FRQ mal in den Diskussionen verlink. Sehe das Thema ist einem größerem zuzuordnen und mache den FRQ erstmal zu. Danke.

closed 2023-09-22 by patsch9