evcc-io/evcc #9978: LG ESS Abfrage sorgt für Logout aus der EnerVu App

patsch9, 2023-09-20:
### Describe the bug

In der Komponente meter/lgpcs/lgpcs.go ist zu sehen das beim LG ESS System der Login über den Installer Login erfolgt "/v1/login". Durch dieses vorgehen wird man in der LG App, immer dann wenn EVCC abfragt, geschmissen da sich der "auth_key" mit jedem Login ändert.

Da von EVCC nur Daten abgefragt werden welche auch als Rolle User Sichtbar sind wäre es gut wenn es eine Möglichkeit gäbe zu entscheiden den Login über den User Login "/v1/user/setting/login" durchzuführen. Wie man das Passwort für den User herausfindet ist hier beschrieben: https://github.com/Morluktom/ioBroker.lg-ess-home/tree/master#getting-the-password

Der Restliche Weg der Authentifizierung ist gleich der über den Installer Login.

### Steps to reproduce

1. Starten von EVCC mit LG ESS
2. Login in der LG EnerVu App als Installer
3. Mit der Abfrage von EVCC wird man aus der App geschmissen

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: lg-ess-home-8-10
    usage: grid
    host: 192.168.X.X
    registration: DEXXXX
  - name: my_pv
    type: template
    template: lg-ess-home-8-10
    usage: pv
    host: 192.168.X.X
    registration: DEXXXX
  - name: my_battery
    type: template
    template: lg-ess-home-8-10
    usage: battery
    host: 192.168.X.X
    registration: DEXXXX
```

### Log details

```text
keine
```

### What type of operating system are you running?

Linux

### Version

0.120.0

andig, 2023-09-20:
Was heisst das konkret für die Implementierung? Willst du einen PR versuchen?

patsch9, 2023-09-21:
> Was heisst das konkret für die Implementierung? Willst du einen PR versuchen?

Für die Implementierung selber macht es keinen Unterschied welche Anmeldung genutzt wird, da die Daten die für EVCC relevant sind mit beiden wegen Abgefragt werden können. Der Unterschied ist, dass sich als "User" mehrere Clients gleichzeitig anmelden können, jedoch als Installateur nur einer. So wird man bei der Abfrage von EVCC an LGESS bei gleichzeitiger Nutzung der Installateur Einstellungen in der App, aus der App geworfen. Der einzige unterschied ist, dass Anwender der Implementierung vorher das User Password auslesen müssen, statt die Registrierungsnummer des WR als Passwort zu nutzen (diese ist das Standard Passwort, kann aber vom Installateur auch geändert worden sein)

Ich kann es mit einem PR gerne versuchen, da ich aber mit GO nicht so vertraut bin würde ich nur die Login URL ändern ohne eine Condition. So ist es allerdings nur mit dem Passwort und nicht mehr mit der RegNo. möglich sich anzumelden.

andig, 2023-09-21:
Verstehe. Dann müsste egtl. das Template geändert werden um zu entscheiden welcher Parameter- user oder regno- angegeben ist.

patsch9, 2023-09-21:
> Verstehe. Dann müsste egtl. das Template geändert werden um zu entscheiden welcher Parameter- user oder regno- angegeben ist.

ich denke das passt. Logisch müsste ja so gehen, dass man folgende Konfiguration in der evcc.yaml hat:

  - name: my_battery
    type: template
    template: lg-ess-home-8-10
    usage: battery
    host: 192.168.XX.XX
    registration: DE21XXXXXX

oder

  - name: my_battery
    type: template
    template: lg-ess-home-8-10
    usage: battery
    host: 192.168.XX.XX
    userpassword: abcd1234abd

Über eine If condition könnte man dann ja entscheiden welche login uri genutzt wird: 

if $registration >="" set
const (
	LoginURI = "/v1/login"
	MeterURI = "/v1/user/essinfo/home"
)

if $userpassword >= "" and $registration = "" 
const (
	LoginURI = "/v1/user/setting/login"
	MeterURI = "/v1/user/essinfo/home"
)

andig, 2023-09-21:
Genau. Und dabei die bisher existierenden Parameter beibehalten.

patsch9, 2023-09-21:
> Genau. Und dabei die bisher existierenden Parameter beibehalten.

Ich habe mich mal dran versucht. Wenn das so passt kann ich gerne einen PR machen.

https://github.com/evcc-io/evcc/compare/master...patsch9:evcc:master

patsch9, 2023-09-22:
#10001

hab mal ein PR aufgemacht

andig, 2023-09-24:
@patsch9 probierst du es mal bitte?

closed 2023-09-29 by andig

patsch9, 2023-10-09:
> @patsch9 probierst du es mal bitte?

ist der fix im aktuellen release schon enthalten? Wenn ja funktioniert es nicht: [main  ] FATAL 2023/10/09 12:00:08 cannot create meter 'my_grid': cannot create meter 'template': invalid key: password

andig, 2023-10-09:
Siehst Du hier: https://github.com/evcc-io/evcc/compare/0.120.3...master. Nein...

patsch9, 2023-10-13:
sieht soweit erstmal gut aus