evcc-io/evcc #9985: EPEX Spot Preise

LarsTh86, 2023-09-21:
Wäre es möglich direkt die Epex Spot Preise zu integrieren? Tibber erlaubt nur gleichzeitig Verbindung mit den token und meine IP wurde es öfteren schon gesperrt, weil alle meine aystem gleichzeitig die Preise geholt haben. 
Die Einspeisung übertrage ich schon per mqtt und so wäre das evcc in der Theorie unabhängig vom tibber.
[image]
Man kann ja wie bei awatar noch konstanten zum Preis hinzugefügt und baut sich eine eigene unabhängige Prognose der Preise

andig, 2023-09-21:
Was sind die EPEX Stop Preise? Gibts dafür ein API? Reicht Dir sonst https://github.com/evcc-io/evcc/pull/9794?

LarsTh86, 2023-09-21:
https://www.eex.com/en/market-data/eex-group-datasource/api --> über die API. Die Plattform nutzen eigentlich alle Anbieter für Ihre Preisbildung, also wäre es nur die logische Lösung diese zu verwenden und  die entsoe transparent Plattform hat die Infos auch, das ist richtig. Aber sie würde sie auch eex ziehen. Die entsoe Plattform bietet deutlich mehr Daten, aber welche ja hier nicht gebraucht werden. Es ist ähnlich zu #9794 aber sicherlich wäre die Implementierung einfacher.

andig, 2023-09-21:
Hast Du da einen Account? Was kostet das?

LarsTh86, 2023-09-21:
Leider nicht. Ich hätte gedacht, da es auch eine homeassistant Integration zur epex gibt, daß es eine kostenlose api ist

LarsTh86, 2023-09-21:
https://github.com/mampfes/ha_epex_spot

LarsTh86, 2023-09-21:
https://www.epexspot.com/en/market-data

andig, 2023-09-21:
Mangels nutzbarem Api wontfix. Entso sollte analoge Daten liefern.

closed 2023-09-21 by andig