evcc-io/evcc #9986: [UI] Zielladen aktiv als einzige Infomeldung

mdkeil, 2023-09-21:
### Describe the bug

Wenn Zielladen aktiv ist, steh bei angeschlossenen Fahrzeug (Status B) im Modus PV unter dem Fahrzeug "Zielladen aktiv. Ladung startet heute [..]" - wenn nun ausreichender PV-Überschuss erkannt, wird "normaler Weise" dies mit einer entsprechenden Info angezeigt und einem Timer angezeigt. Dies ist bei aktivem Zielladen nicht der Fall.. nach Ablauf des internen Timers beginnt dann halt die Ladung mit der Info: "Zielladen aktiv".. das gleiche Verhalten auch, wenn kein ausreichender Überschuss mehr vorhanden ist.. dies wird nicht mit einem ablaufenden Timer angekündigt, sondern auch hier stoppt die Ladung und in der "Infobox" erscheint wieder nur "Zielladen aktiv. Ladung startet heute [..]" 

Fazit: Mit aktivem Zielladen werden sämtliche Informationen, die sonst "standardmäßig" angezeigt werden, durch "Zielladen aktiv [..]" überschieben.. Nun die Frage, ob es so gewollt ist?

### Steps to reproduce

Eine aktive Zielplanung.

### Configuration details

```yaml
nicht notwendig
```

### Log details

```text
kann ich bei Bedarf nachliefern.
```

### What type of operating system are you running?

Docker container

### Version

v0.120.0

andig, 2023-09-21:
> Nun die Frage, ob es so gewollt ist?

Vmtl. nicht...

naltatis, 2023-09-21:
Ja Zielladen hat aktuell Priorität. Egal ob geladen wird weil Überschuss oder weil noch Strom aus dem Netz gebraucht wird. Momentan können wir diese Unterscheidung in der UI auch nicht treffen. Die Frage ist was in welchem Szenario erwartet wäre. Die "Zielladen ..." Meldungen komplett rausnehmen? Dann ist ggf. nicht klar, warum das Auto auch bei Dunkelheit lädt.

andig, 2023-09-21:
Ich denke die Idee hier wäre, Zeilladen während des ladens nur anzuzeigen wenn es sich um einen aktiven plan slot handelt. Anderenfalls wären die normalen PV Meldungen anzuzeigen. Das würde natürlich verstecken, dass ein Plan aktiv ist.

naltatis, 2023-09-22:
Find ich gut. Ich würd vorschlagen, dass wir das grundsätzlicher angehen und sowas wie `chargingReason: surplus|minsoc|buffersoc|plan|smartcost` am LP zu publishen. Mit entsprechender Prioritätenreglung (bspw `plan` sticht `surplus`).

Momentan sammelt sich da Logik in der UI VehicleStatus-Meldungskomponente an (aktueller Preis < Smart Cost Limit, ...). Würde für die Komponente jetzt ungern noch den aktuellen Ladeplan regelmäßig pollen. Aus den gepublishten Daten kann man das aktuell nicht ablesen.

naltatis, 2023-10-15:
Die Information `planActive` gab es im UI doch bereits, wurde nur nicht ausgewertet. Das sollte nun das gewünschte Verhalten abbilden. Heißt der Status wir nun nur noch angezeigt, wenn man sich in einem geplanten Zeitfenster befindet.

Die vorgeschlagene Konsolidierung (siehe `enabledReason`) können wir dann gesondert noch mal angehen. Dieser Issue wäre damit aber behoben.

PR #10332 (2023-10-15): Show plan active status only in active timeslots
fixes #9986
closed 2023-10-15 by andig