evcc-io/evcc #9987: format (target)soc as °C if icon: waterheater

Razer2412, 2023-09-21:
Hallo,

ich betreibe einen 3-phasigen Heizstab zur Warmwasserbereitung.
Dieser hängt pro Phase an einem Shelly und wird abhängig vom vorhandenen PV-Überschuss gesteuert (ca. 666, 1333, 2000 Watt).

ich habe gesehen, dass es die Formatierung des SOC in °C bereits gibt, jedoch nur für Plugins. 

Gibt es die Möglichkeit, die Formatierung des SOC in °C generell für Charger mit dem "Icon: Waterheater" zu implementieren?

andig, 2023-09-21:
@naltatis wollen wir das am icon fest machen? Zumindest als Workaround?

naltatis, 2023-09-21:
Icon mit Funktion zu verknüpfen fänd ich nicht so geil. Außerdem geht Icon aktuell auch nur bei ausgewählten und nicht bei allen Chargern (`switchsocket`). Bei Switchsockets kann man aktuell schon Feature über das Template aktivieren (`integrateddevice`). Ich hab’ jetzt mal analog dazu `heating` hinzugefügt #9996. So richtig universell ist die Lösung nicht, aber für diesen konkreten Fall finde ich die relativ sauber. Müssen überlegen, ob wir, ähnlich wie beim Vehicle, einen Wrapper brauchen mit denen wir Icon+Feature generell einstellbar/überschreibbar machen wollen. Aber vielleicht brauchen wir das auch nicht.

PR #9996 (2023-09-22): Add heating option to switchsockets
fixes #9987
closed 2023-09-22 by andig