evcc-io/evcc #9993: Fiat "vehicle status: unexpected status: 403 (Forbidden)"

horst789github, 2023-09-21:
### Describe the bug

der status lässt sich nicht automatisch abfragen vom Fiat 500e. Früher ging es mal.

### Steps to reproduce

anstecken > homepage checken > status immernoch unbekannt (gastfahrzeug)

### Configuration details

```yaml
vehicles:
  - name: Fiat500e
    type: template
    template: fiat
    title: Fiat500e
    user: xxx
    password: xxx
    vin: ZFAEFAC47NXxxx
    pin: xxx
    mode: pv # enable PV-charging when vehicle is identified
    minSoc: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    targetSoc: 100 # limit charge to 90%
    priority: 3
```

### Log details

```text
normales log:
2023/09/21 15:51:13	stdout	[lp-3  ] ERROR 2023/09/21 15:51:13 vehicle soc: unexpected status: 403 (Forbidden)
2023/09/21 15:51:13	stdout	[lp-3  ] ERROR 2023/09/21 15:51:13 vehicle odometer: unexpected status: 403 (Forbidden)
2023/09/21 15:50:47	stdout	[lp-1  ] ERROR 2023/09/21 15:50:47 vehicle odometer: asleep
2023/09/21 15:50:43	stdout	[main  ] ERROR 2023/09/21 15:50:43 vehicle status: unexpected status: 403 (Forbidden) 

und im trace:
2023/09/21 15:49:30	stdout	[main  ] ERROR 2023/09/21 15:49:30 vehicle status: unexpected status: 403 (Forbidden)
2023/09/21 15:49:30	stdout	{"name":"FORBIDDEN","message":"Forbidden"}
2023/09/21 15:49:30	stdout	--
2023/09/21 15:49:30	stdout	[fiat  ] TRACE 2023/09/21 15:49:30 {"command":"DEEPREFRESH","pinAuth":"geheimer pinauth habe ich mal hier entfernt"}
2023/09/21 15:49:30	stdout	[fiat  ] TRACE 2023/09/21 15:49:30 POST https://channels.sdpr-01.fcagcv.com/v1/accounts/5c7f593c9ffe40bf89cab8884df16659/vehicles/***/ev
2023/09/21 15:49:30	stdout	{"token":"geheimer token habe ich mal hier entfernt","expiry":600000}
2023/09/21 15:49:30	stdout	--
2023/09/21 15:49:30	stdout	[fiat  ] TRACE 2023/09/21 15:49:30 {"pin":"geheime pin habe ich mal hier entfernt"}
2023/09/21 15:49:29	stdout	[fiat  ] TRACE 2023/09/21 15:49:29 POST https://mfa.fcl-01.fcagcv.com/v1/accounts/5c7f593c9ffe40bf89cab8884df16659/ignite/pin/authenticate
2023/09/21 15:49:29	stdout	[fiat  ] TRACE 2023/09/21 15:49:29 {"vehicleInfo":{"totalRangeADA":null,"odometer":{"odometer":{"value":"11787","unit":"km"}},"daysToService":"null","hvbatSOH":null,"fuel":{"fuelAmountLevel":null,"isFuelLevelLow":false,"distanceToEmpty":{"value":"97","unit":"km"},"fuelAmount":{"value":"null","unit":"null"}},"oilLevel":{"oilLevel":null},"tyrePressure":[{"warning":false,"pressure":{"value":"null","unit":"kPa"},"type":"FL","status":"NORMAL"},{"warning":false,"pressure":{"value":"null","unit":"kPa"},"type":"FR","status":"NORMAL"},{"warning":false,"pressure":{"value":"null","unit":"kPa"},"type":"RL","status":"NORMAL"},{"warning":false,"pressure":{"value":"null","unit":"kPa"},"type":"RR","status":"NORMAL"}],"batteryInfo":{"batteryStatus":"0","batteryVoltage":{"value":"14.4","unit":"volts"},"batteryStateOfCharge":"null"},"tripsInfo":{"trips":[{"totalElectricDistance":{"value":"null","unit":"km"},"name":"TripA","totalDistance":{"value":"11509","unit":"km"},"energyUsed":{"value":"null","unit":"kmpl"},"averageEnergyUsed":{"value":"null","unit":"kmpl"},"totalHybridDistance":{"value":"null","unit":"km"}},{"totalElectricDistance":{"value":"null","unit":"km"},"name":"TripB","totalDistance":{"value":"25","unit":"km"},"energyUsed":{"value":"null","unit":"kmpl"},"averageEnergyUsed":{"value":"null","unit":"kmpl"},"totalHybridDistance":{"value":"null","unit":"km"}}]},"batPwrUsageDisp":null,"distanceToService":{"distanceToService":{"value":"3213.0","unit":"km"}},"wheelCount":4,"hvacPwrUsageDisp":null,"mtrPwrUsageDisp":null,"tpmsvehicle":false,"isTPMSVehicle":false,"HVBatSOH":null,"timestamp":1695302835902},"evInfo":{"chargeSchedules":[],"ignitionStatus":"OFF","battery":{"stateOfCharge":45,"chargingLevel":"DEFAULT","plugInStatus":false,"timeToFullyChargeL3":0,"timeToFullyChargeL2":0,"chargingStatus":"NOT_CHARGING","totalRange":97,"distanceToEmpty":{"value":97,"unit":"km"}},"timestamp":1695302838519,"schedules":[{"chargeToFull":false,"scheduleType":"CHARGE","enableScheduleType":false,"scheduledDays":{"sunday":false,"saturday":false,"tuesday":false,"wednesday":false,"thursday":false,"friday":false,"monday":false},"startTime":"00:00","endTime":"00:00","cabinPriority":false,"repeatSchedule":true},{"chargeToFull":false,"scheduleType":"CHARGE","enableScheduleType":false,"scheduledDays":{"sunday":false,"saturday":false,"tuesday":false,"wednesday":false,"thursday":false,"friday":false,"monday":false},"startTime":"00:00","endTime":"00:00","cabinPriority":false,"repeatSchedule":true},{"chargeToFull":false,"scheduleType":"CHARGE","enableScheduleType":false,"scheduledDays":{"sunday":false,"saturday":false,"tuesday":false,"wednesday":false,"thursday":false,"friday":false,"monday":false},"startTime":"00:00","endTime":"00:00","cabinPriority":false,"repeatSchedule":true}]},"timestamp":1695302838519}
2023/09/21 15:49:29	stdout	[fiat  ] TRACE 2023/09/21 15:49:29 GET https://channels.sdpr-01.fcagcv.com/v2/accounts/5c7f593c9ffe40bf89cab8884df16659/vehicles/***/status
2023/09/21 15:49:29	stdout	[lp-3  ] DEBUG 2023/09/21 15:49:29 charge total import: 4739.000kWh
2023/09/21 15:49:29	stdout	[lp-3  ] DEBUG 2023/09/21 15:49:29 guard timer inactive
2023/09/21 15:49:29	stdout	[lp-3  ] DEBUG 2023/09/21 15:49:29 charger: guard elapse
2023/09/21 15:49:29	stdout	[lp-3  ] DEBUG 2023/09/21 15:49:29 pv timer inactive
2023/09/21 15:49:29	stdout	[lp-3  ] DEBUG 2023/09/21 15:49:29 pv timer elapse
2023/09/21 15:49:29	stdout	[lp-3  ] DEBUG 2023/09/21 15:49:29 vehicle api refresh
```

### What type of operating system are you running?

Docker container

### Version

0.120.0

andig, 2023-09-21:
Da lässt sich leider nichts machen- hat wohl Fiat geändert :/

closed 2023-09-21 by andig

horst789github, 2023-10-30:
> Die PIN ist für das Senden von Commands an das Fahrzeug notwendig.

super danke für den tipp, probier ich mal aus. Aber was hat das für ein Nachteil? weil Evcc kann doch sowie keine commands senden oder?

horst789github, 2023-11-01:
>  das Entfernen oder Auskommentieren der `pin:` Zeile hat bei mir geholfen.

hat bei mir leider nicht funktioniert. Der Fiat wird nach dem Anstecken weiterhin nicht erkannt, erst wenn ich zwanghaft (schnell oder Min/PV) anfange zu laden erkennt er das auto

horst789github, 2023-11-20:
> Da lässt sich leider nichts machen- hat wohl Fiat geändert :/

was soll ich liefern, damit die änderung erkannt wird und wieder in evcc eingebaut wird?
Der SoC wird ja erkannt von evcc, nur halt nicht am anfang sondern erst wenn man lädt.

thecem, 2024-08-03:
Habe das gleiche Problem, der Stand wird aber nicht nur abgefragt sondern im Log sichtbar auch geliefert! evcc verarbeitet diesen Wert nicht.

StefanSchoof, 2024-08-04:
@thecem Das ist ein geschlossenes Issue aus dem letzten Jahr. Ich denke du solltest ein neues Issue mit guter Beschreibung und Log das dein Problem zeigt aufmachen.