evcc-io/evcc #10020: evcc startet nicht mehr nach update auf 120.1

zwiebelxxl, 2023-09-23:
### Describe the bug

Nach update von 120.0 auf 121.1 started evcc nicht mehr. Keine Änderung an der Konfiguration gemacht.
Der Startvorgang bleibt stehen.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token: *****
sponsortoken: *****

# log settings
#log: error
log: debug
levels:
  #core: error
  #lp-1: error
  #lp-2: error
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: pv
  #type: custom
  #type: default
  type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.10.11 # IP-Adresse oder Hostname
    #power:
    #source: http
    #type: http
    #uri: http://192.168.10.11/solar_api/v1/GetPowerFlowRealtimeData.fcgi
    #jq: if .Body.Data.Site.P_PV == null then 0 else .Body.Data.Site.P_PV end
- name: pv2
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.10.126 # IP-Adresse oder Hostname

- name: pv2_battery
  type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.10.126 # IP-Adresse oder Hostname
[... 258 lines cut ...]
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] INFO 2023/09/23 11:15:13   mode:        pv
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] INFO 2023/09/23 11:15:13   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] INFO 2023/09/23 11:15:13   meters:      charge ✓
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] INFO 2023/09/23 11:15:13     charge:    power ✓ energy ✓ currents ✓
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] INFO 2023/09/23 11:15:13 loadpoint 2:
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] INFO 2023/09/23 11:15:13   mode:        pv
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] INFO 2023/09/23 11:15:13   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] INFO 2023/09/23 11:15:13   meters:      charge ✓
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] INFO 2023/09/23 11:15:13     charge:    power ✓ energy ✗ currents ✓
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] DEBUG 2023/09/23 11:15:13 phase timer inactive
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] DEBUG 2023/09/23 11:15:13 pv timer inactive
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] DEBUG 2023/09/23 11:15:13 guard timer inactive
Sep 23 11:15:13 as2 evcc[63158]: [lp-1  ] INFO 2023/09/23 11:15:13 vehicle updated: unknown -> IONIQ 5
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] DEBUG 2023/09/23 11:15:13 phase timer inactive
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] DEBUG 2023/09/23 11:15:13 pv timer inactive
Sep 23 11:15:13 as2 evcc[63158]: [lp-2  ] DEBUG 2023/09/23 11:15:13 guard timer inactive
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.1

andig, 2023-09-23:
Was startet da nicht? Läuft doch laut Log?

zwiebelxxl, 2023-09-23:
Der process ist noch da.
```
ps aux | grep evcc
root       63158  0.1  1.2 1320576 72900 ?       Ssl  11:15   0:14 /opt/evcc/evcc -c /opt/evcc/evcc.yaml --log error
```
Aber der Port der gui is nicht offen.

Log mit trace
```
[site  ] INFO 2023/09/23 14:11:20 site config:
[site  ] INFO 2023/09/23 14:11:20   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/23 14:11:20     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/09/23 14:11:20     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/23 14:11:20     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/23 14:11:20     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/23 14:11:20   vehicles:
[site  ] INFO 2023/09/23 14:11:20     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/09/23 14:11:20 loadpoint 1:
[lp-1  ] INFO 2023/09/23 14:11:20   mode:        pv
[lp-1  ] INFO 2023/09/23 14:11:20   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/09/23 14:11:20   meters:      charge ✓
[lp-1  ] INFO 2023/09/23 14:11:20     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/09/23 14:11:20 loadpoint 2:
[lp-2  ] INFO 2023/09/23 14:11:20   mode:        pv
[lp-2  ] INFO 2023/09/23 14:11:20   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/09/23 14:11:20   meters:      charge ✓
[lp-2  ] INFO 2023/09/23 14:11:20     charge:    power ✓ energy ✗ currents ✓
[cache ] TRACE 2023/09/23 14:11:20 siteTitle: Zuhause
[cache ] TRACE 2023/09/23 14:11:20 gridConfigured: true
[cache ] TRACE 2023/09/23 14:11:20 pvConfigured: true
[cache ] TRACE 2023/09/23 14:11:20 batteryConfigured: true
[cache ] TRACE 2023/09/23 14:11:20 bufferSoc: 90
[cache ] TRACE 2023/09/23 14:11:20 bufferStartSoc: 0
[cache ] TRACE 2023/09/23 14:11:20 prioritySoc: 50
[cache ] TRACE 2023/09/23 14:11:20 residualPower: 0
[cache ] TRACE 2023/09/23 14:11:20 smartCostLimit: 0
[cache ] TRACE 2023/09/23 14:11:20 smartCostType: <nil>
[cache ] TRACE 2023/09/23 14:11:20 smartCostType: pricestatic
[cache ] TRACE 2023/09/23 14:11:20 currency: EUR
[cache ] TRACE 2023/09/23 14:11:20 savingsSince: 2022-11-02 20:46:16 +0100 CET
[cache ] TRACE 2023/09/23 14:11:20 vehicles: [IONIQ 5]
[cache ] TRACE 2023/09/23 14:11:20 lp-1/title: Garage
[cache ] TRACE 2023/09/23 14:11:20 lp-1/minCurrent: 6
[cache ] TRACE 2023/09/23 14:11:20 lp-1/maxCurrent: 16
[cache ] TRACE 2023/09/23 14:11:20 lp-1/phasesConfigured: 0
[lp-1  ] DEBUG 2023/09/23 14:11:20 phase timer inactive
[lp-1  ] DEBUG 2023/09/23 14:11:20 pv timer inactive
[lp-1  ] DEBUG 2023/09/23 14:11:20 guard timer inactive
[lp-1  ] INFO 2023/09/23 14:11:20 vehicle updated: unknown -> IONIQ 5
[cache ] TRACE 2023/09/23 14:11:20 lp-1/phasesEnabled: 0
[cache ] TRACE 2023/09/23 14:11:20 lp-1/phasesActive: 3
[cache ] TRACE 2023/09/23 14:11:20 lp-1/phaseAction: inactive
[cache ] TRACE 2023/09/23 14:11:20 lp-1/phaseRemaining: 0s
[cache ] TRACE 2023/09/23 14:11:20 lp-1/pvAction: inactive
[cache ] TRACE 2023/09/23 14:11:20 lp-1/pvRemaining: 0s
[cache ] TRACE 2023/09/23 14:11:20 lp-1/guardAction: inactive
[cache ] TRACE 2023/09/23 14:11:20 lp-1/guardRemaining: 0s
[cache ] TRACE 2023/09/23 14:11:20 lp-1/chargerFeatureIntegratedDevice: false
[cache ] TRACE 2023/09/23 14:11:20 lp-1/chargerFeatureHeating: false
[cache ] TRACE 2023/09/23 14:11:20 lp-1/chargerIcon: <nil>
[cache ] TRACE 2023/09/23 14:11:20 lp-1/minSoc: 0
[cache ] TRACE 2023/09/23 14:11:20 lp-1/targetSoc: 100
[cache ] TRACE 2023/09/23 14:11:20 lp-1/targetEnergy: 0
[cache ] TRACE 2023/09/23 14:11:20 lp-1/vehiclePresent: true
[cache ] TRACE 2023/09/23 14:11:20 lp-1/vehicleTitle: IONIQ 5
[cache ] TRACE 2023/09/23 14:11:20 lp-1/vehicleIcon: car
[cache ] TRACE 2023/09/23 14:11:20 lp-1/vehicleCapacity: 58
[cache ] TRACE 2023/09/23 14:11:20 lp-1/phasesActive: 3
[cache ] TRACE 2023/09/23 14:11:20 lp-1/vehicleSoc: 0
[cache ] TRACE 2023/09/23 14:11:20 lp-1/vehicleRange: 0
[... 25 lines cut ...]
[mqtt  ] TRACE 2023/09/23 14:11:26 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:11:36 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:11:50 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:11:56 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:12:06 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:12:16 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[modbus] TRACE 2023/09/23 14:12:19 modbus: closing connection due to idle timeout: 1m0.000167752s
[mqtt  ] TRACE 2023/09/23 14:12:26 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:12:37 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:12:49 recv mqttGenericBridge/phoenix_meter/Current_L2__A: '0.00'
[mqtt  ] TRACE 2023/09/23 14:12:49 recv mqttGenericBridge/phoenix_meter/Current_L1__A: '0.00'
[mqtt  ] TRACE 2023/09/23 14:12:49 recv mqttGenericBridge/phoenix_meter/Current_L3__A: '0.00'
[mqtt  ] TRACE 2023/09/23 14:12:50 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:12:56 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:13:06 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:13:16 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[... cut ...]
[mqtt  ] TRACE 2023/09/23 14:13:36 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:13:53 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:13:56 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:14:06 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:14:16 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'
[mqtt  ] TRACE 2023/09/23 14:14:29 recv mqttGenericBridge/phoenix_meter/Power_Sum__W: '0.0'

```

andig, 2023-09-23:
Schau mal bitte auf /debug/pprof. Sieht nach Deadlock aus.

zwiebelxxl, 2023-09-23:
Du meinst mit dem Browser auf:
http://192.168.10.121:7070/debug/pprof

Aber der Port 7070 ist nicht offen:

```
as2:~# netstat -antp | grep 7020
as2:~#
```

andig, 2023-09-23:
Mhhhm- da haben wir ein echtes Problem mit der Diagnose. Kannst Du info@evcc.io irgendwie zu einem VPN einladen so dass ich eine Debug Session machen könnte?

andig, 2023-09-23:
grep 7070

?

zwiebelxxl, 2023-09-23:
Ich hab eine Google Meeting Einladung zu info@evcc.io für heute gesendet. Bin bereit. :)

zwiebelxxl, 2023-09-23:
Leider ist der Online Call nicht zustande gekommen.

Ich dachte das auf dem Port 7070 in dem Pfad: debug/pprof irgendeine art von logging gemacht wird. Ich lag falsch.

andig, 2023-09-23:
Ist denn der Port überhaupt offen? Oben hattest Du auf 7020 getestet.

andig, 2023-09-23:
Das Fahrzeug `ev1` ist fest beiden Ladepunkten zugewiesen- das kann nicht funktionieren. Allerdings steckt hier auch ein Bug drin der zum Deadlock führt.

zwiebelxxl, 2023-09-23:
Hallo Andreas,

danke für deine Hilfe! Super, dass ich jetzt weiß woran es lag und jetzt
auch die neue Version benutzen kann.
ssh Zugang hab ich wieder ausgeschaltet...

Viele Grüße,
Gabriel.

andig, 2023-09-23:
Closed in https://github.com/evcc-io/evcc/commit/3b1d61523998ae5cf876d5c7a4c24c0ec929abb8

closed 2023-09-23 by andig