evcc-io/evcc #10029: EVCC crash on unconfigured OCPP charger (Denial of Service)

c-po, 2023-09-24:
### Describe the bug

I have two ABB Terra AC W22-T-0, one is configured in EVCC the other is not. The configured one connects without an issue. As soon as the second (unconfigured) charger connects - EVCC will `SIGSEGV`

### Steps to reproduce

1. Let an unconfigured OCPP enabled charger connect to EVCC
2. EVCC will crash

### Configuration details

```yaml
chargers:
  - type: template
    template: ocpp
    stationid: TACW2243521T1625
    connector: 1
    idtag: 1
    connecttimeout: 15m
    timeout: 5m
    getconfiguration: false
    bootnotification: true
    meterinterval: 1m
    metervalues: 1m
    name: wallbox-2
```
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/09/24 08:36:10 evcc 0.120.2
[main  ] INFO 2023/09/24 08:36:10 using config file: /config/evcc.yaml
[main  ] INFO 2023/09/24 08:36:10 starting ui and api at :7070
[TACW2243521T1625] DEBUG 2023/09/24 08:36:11 waiting for chargepoint: 15m0s
[ocpp  ] DEBUG 2023/09/24 08:38:42 chargepoint connected: TACW2243521T1625
[ocpp  ] WARN 2023/09/24 08:38:42 chargepoint connected, unknown: TACW2243521T2288
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x175d412]
goroutine 31 [running]:
github.com/evcc-io/evcc/charger/ocpp.(*CP).BootNotification(0x40feda?, 0xc0009d0005?)
	github.com/evcc-io/evcc/charger/ocpp/cp_core.go:29 +0x12
github.com/evcc-io/evcc/charger/ocpp.(*CS).OnBootNotification(0xc00096cbd0?, {0xc0009d0005?, 0x10?}, 0x32f2ba0?)
	github.com/evcc-io/evcc/charger/ocpp/cs_core.go:76 +0xee
github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingRequest.func1()
	github.com/lorenzodonini/ocpp-go@v0.17.0/ocpp1.6/central_system.go:508 +0x4ea
created by github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingRequest in goroutine 48
	github.com/lorenzodonini/ocpp-go@v0.17.0/ocpp1.6/central_system.go:505 +0x5ac
```
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.120.2

andig, 2023-09-24:
@c-po kannst Du den PR testen?

closed 2023-09-24 by andig

c-po, 2023-09-24:
Works as expected!

```
(16:12) cpo lnx01:~/evcc [master] # ./evcc --version
evcc version 0.120.2 (d58706e8)
(16:12) cpo lnx01:~/evcc [master] # ./evcc
[main  ] INFO 2023/09/24 16:12:44 evcc 0.120.2 (d58706e8)
[main  ] INFO 2023/09/24 16:12:44 using config file: /home/cpo/evcc/evcc.yml
[main  ] INFO 2023/09/24 16:12:44 starting ui and api at :7070
[TACW2243521T1625] DEBUG 2023/09/24 16:12:45 waiting for chargepoint: 15m0s
[ocpp  ] WARN 2023/09/24 16:12:51 unknown charge point connected: TACW2243521T2288
[ocpp  ] DEBUG 2023/09/24 16:12:53 charge point connected: TACW2243521T1625
OnBootNotification id TACW2243521T1625
OnBootNotification id TACW2243521T1625
[site  ] INFO 2023/09/24 16:12:57 site config:
[site  ] INFO 2023/09/24 16:12:57   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/24 16:12:57     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/24 16:12:57     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/24 16:12:57     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/09/24 16:12:57   vehicles:
[site  ] INFO 2023/09/24 16:12:57     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/09/24 16:12:57 loadpoint 1:
[lp-1  ] INFO 2023/09/24 16:12:57   mode:        pv
[lp-1  ] INFO 2023/09/24 16:12:57   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/24 16:12:57   meters:      charge ✗
[lp-1  ] DEBUG 2023/09/24 16:12:57 phase timer inactive
[lp-1  ] DEBUG 2023/09/24 16:12:57 pv timer inactive
[lp-1  ] DEBUG 2023/09/24 16:12:57 guard timer inactive
[site  ] DEBUG 2023/09/24 16:12:57 ----
[lp-1  ] DEBUG 2023/09/24 16:12:57 charge power: 0W
[site  ] DEBUG 2023/09/24 16:12:57 pv power: 5722W
```

andig, 2023-09-24:
Super- danke für die hervorragende Fehlermeldung!