evcc-io/evcc #10041: Typo in https://docs.evcc.io/docs/reference/configuration/site

t-beech, 2023-09-24:
### Describe the bug

pv: myonlypv # **singele** pv meter reference

### Steps to reproduce

Open  https://docs.evcc.io/docs/reference/configuration/site and select the meters.pv section.

### Configuration details

```yaml
Not applicable, since this is a typo in the documentation.
```

### Log details

```text
Not applicable, since this is a typo in the documentation.
```

### What type of operating system are you running?

Windows

andig, 2023-09-24:
/cc @VolkerK62

VolkerK62, 2023-09-24:
@t-beech  would you like to do that? Otherwise I'll take care of it

t-beech, 2023-09-24:
@VolkerK62 Did not see that the docs were in a separate repo. Just created https://github.com/evcc-io/docs/pull/448.

closed 2023-09-24 by andig