evcc-io/evcc #10044: evcc cannot enable charger

pjjjv, 2023-09-24:
### Describe the bug

With version 0.120.2 versus 0.119.3, it looks like it is much harder for evcc to sync with the charger state. Because I don´t know how to disconnect and reconnect the car without starting a transaction, before I restart evcc, I usually end up starting evcc with a transaction running. This would usually result in a few errors, but then I can set evcc to mode Off and back to PV. Except now, I can´t seem to get evcc to get control. This is using an OCPP charger.

### Steps to reproduce

1. Stop evcc
2. Disconnect car
3. Start evcc
4. Connect car. A transaction starts which is not started by evcc.
5. Evcc reports:
[lp-1  ] WARN 2023/09/24 14:36:31 charger logic error: disabled but charging
[lp-1  ] ERROR 2023/09/24 14:36:31 charger: cannot enable: transaction already running

6. Set evcc to mode Off
7. Set evcc to mode PV

Evcc no longer executes any RemoteStopTransaction or RemoteStartTransaction I believe. Evcc never gets in control of the state.

### Configuration details

```yaml
OCPP charger
```

### Log details

```text
[lp-1  ] WARN 2023/09/24 14:36:31 charger logic error: disabled but charging
[lp-1  ] ERROR 2023/09/24 14:36:31 charger: cannot enable: transaction already running
```

### What type of operating system are you running?

Linux

### Version

0.120.2

andig, 2023-09-24:
> Connect car. A transaction starts which is not started by evcc.

That's ocpp. We cannot take over a foreign transaction, hence the error.

pjjjv, 2023-09-24:
Hmm, looks like you are right. With 0.119.3 (undoing commit b28c7b5 idem) I got:
[lp-1  ] WARN 2023/09/24 15:00:30 charger logic error: disabled but charging
[lp-1  ] ERROR 2023/09/24 15:00:30 charger disable: unknown transaction running, cannot disable

andig, 2023-09-24:
If you find anything in the spec that might help let me know. You can also play with Steve and try if you find anything to stop an unknown transaction.

wlanrouter, 2023-09-25:
Is there a trick to let evcc start the transaction?

I can not change any charging value.

Here is my log:
```
[main  ] INFO 2023/09/25 18:10:45 evcc 0.120.2
[main  ] INFO 2023/09/25 18:10:45 using config file: evcc.yaml
[main  ] INFO 2023/09/25 18:10:45 starting ui and api at :7070
[db    ] INFO 2023/09/25 18:10:46 using sqlite database: /home/pi/.evcc/evcc.db
[ocpp  ] DEBUG 2023/09/25 18:10:47 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2023/09/25 18:11:05 chargepoint connected, registering: EVB-P21124929
[ocpp  ] TRACE 2023/09/25 18:11:05 sent JSON message to EVB-P21124929: [2,"2101692415","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2023/09/25 18:11:06 received JSON message from EVB-P21124929: [2,"2ca46d9a-df12-4647-b7e3-40e52bf51481","Heartbeat",{}]
[ocpp  ] TRACE 2023/09/25 18:11:06 sent JSON message to EVB-P21124929: [3,"2ca46d9a-df12-4647-b7e3-40e52bf51481",{"currentTime":"2023-09-25T16:11:06Z"}]
[ocpp  ] TRACE 2023/09/25 18:11:06 received JSON message from EVB-P21124929: [3,"2101692415",{"status":"Accepted"}]
[ocpp  ] TRACE 2023/09/25 18:11:06 sent JSON message to EVB-P21124929: [2,"2430835522","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp  ] TRACE 2023/09/25 18:11:06 TriggerMessage MeterValues for EVB-P21124929: Accepted
[ocpp  ] TRACE 2023/09/25 18:11:06 received JSON message from EVB-P21124929: [3,"2430835522",{"status":"Accepted"}]
[ocpp  ] DEBUG 2023/09/25 18:11:06 enabling meter watchdog
[ocpp  ] TRACE 2023/09/25 18:11:07 received JSON message from EVB-P21124929: [2,"2ca46d9a-df12-4647-b7e3-40e5d3076f83","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2023-09-25T16:11:06Z","sampledValue":[{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"6193560","context":"Trigger","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Trigger","measurand":"Power.Active.Import","unit":"W"},{"value":"224","context":"Trigger","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"224","context":"Trigger","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"225","context":"Trigger","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp  ] TRACE 2023/09/25 18:11:07 sent JSON message to EVB-P21124929: [3,"2ca46d9a-df12-4647-b7e3-40e5d3076f83",{}]
[ocpp  ] TRACE 2023/09/25 18:11:07 received JSON message from EVB-P21124929: [2,"2ca46d9a-df12-4647-b7e3-40e5df25c284","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Available,NoError,No error.,0,Off,0,{0,7936,100},{6193560,0},{A,11936,12028,61568},0,0,390,2023-09-25T16:11:06Z,0,3,{224,224,225,0,0,0,0,0,0},290,0,0,136,0,340,900,5009"}]
[ocpp  ] TRACE 2023/09/25 18:11:07 sent JSON message to EVB-P21124929: [3,"2ca46d9a-df12-4647-b7e3-40e5df25c284",{"status":"Accepted"}]
[ocpp  ] TRACE 2023/09/25 18:12:06 received JSON message from EVB-P21124929: [2,"2ca46dd6-df12-4647-b7e3-40e55afb55e6","Heartbeat",{}]
[ocpp  ] TRACE 2023/09/25 18:12:06 sent JSON message to EVB-P21124929: [3,"2ca46dd6-df12-4647-b7e3-40e55afb55e6",{"currentTime":"2023-09-25T16:12:06Z"}]
[ocpp  ] TRACE 2023/09/25 18:12:06 sent JSON message to EVB-P21124929: [2,"4028331096","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":1}]
[ocpp  ] TRACE 2023/09/25 18:12:06 received JSON message from EVB-P21124929: [3,"4028331096",{"status":"Accepted"}]
[ocpp  ] TRACE 2023/09/25 18:12:06 TriggerMessage StatusNotification for EVB-P21124929: Accepted
[ocpp  ] TRACE 2023/09/25 18:12:08 received JSON message from EVB-P21124929: [2,"2ca46dd6-df12-4647-b7e3-40e5db282bea","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","timestamp":"2023-09-25T16:12:06Z"}]
[ocpp  ] TRACE 2023/09/25 18:12:08 sent JSON message to EVB-P21124929: [3,"2ca46dd6-df12-4647-b7e3-40e5db282bea",{}]
[site  ] WARN 2023/09/25 18:12:08 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/09/25 18:12:08 deprecated: use 'battery' instead of 'batteries'
[site  ] WARN 2023/09/25 18:12:08 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://https://docs.evcc.io/docs/reference/configuration/site#residualpower
[site  ] WARN 2023/09/25 18:12:08 bufferStartSoc must be larger than bufferSoc
[site  ] INFO 2023/09/25 18:12:08 site config:
[site  ] INFO 2023/09/25 18:12:08   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/25 18:12:08     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/25 18:12:08     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/25 18:12:08     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2023/09/25 18:12:08 loadpoint 1:
[lp-1  ] INFO 2023/09/25 18:12:08   mode:        pv
[lp-1  ] INFO 2023/09/25 18:12:08   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/25 18:12:08   meters:      charge ✓
[lp-1  ] INFO 2023/09/25 18:12:08     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/09/25 18:12:08 phase timer inactive
[lp-1  ] DEBUG 2023/09/25 18:12:08 pv timer inactive
[lp-1  ] DEBUG 2023/09/25 18:12:08 guard timer inactive
[site  ] WARN 2023/09/25 18:12:08 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[... cut ...]
        uri: http://192.168.81.12/getInfluxData.php?nurBatSoc=1   

 
chargers:
- name: wallbox1
  type: template
  template: ocpp 
  connecttimeout: 5m
  timeout: 2m  
  getconfiguration: false
  bootnotification: false
  metervalues: Current.Import.L1,Current.Import.L2,Current.Import.L3,Energy.Active.Import.Register,Power.Active.Import,Voltage.L1,Voltage.L2,Voltage.L3

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
```

wlanrouter, 2023-09-25:
> 

> > Connect car. A transaction starts which is not started by evcc.
> 
> That's ocpp. We cannot take over a foreign transaction since we don't know its id, hence the error. If you think evcc was able to take over an unknown transaction in an earlier version I'd be happy to look at a log file.

But @andig , isn't OCPP telling evcc the transaction ID? In this case "1"
```
[ocpp  ] TRACE 2023/09/25 18:13:32 received JSON message from EVB-P21124921: [2,"2ca46e2b-df12-4647-b7e3-40e5d6a267e4","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2023-09-25T16:13:31Z","sampledValue":[{"value":"5.8","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"5.6","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"5.7","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"6193572","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"3890","measurand":"Power.Active.Import","unit":"W"},{"value":"224","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp  ] TRACE 2023/09/25 18:13:32 sent JSON message to EVB-P21124921: [3,"2ca46e2b-df12-4647-b7e3-40e5d6a267e4",{}]
[ocpp  ] TRACE 2023/09/25 18:13:34 received JSON message from EVB-P21124921: [2,"2ca46e2d-df12-4647-b7e3-40e5ab61b9f1","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Charging,NoError,No error.,1,Blue,1,{20,7936,10},{6193572,3890},{C,11936,5932,12112},0,0,380,2023-09-25T16:13:31Z,1,3,{224,226,226,58,56,57,996,996,996},290,1,0,137,0,330,60,5009"}]
[ocpp  ] TRACE 2023/09/25 18:13:34 sent JSON message to EVB-P21124921: [3,"2ca46e2d-df12-4647-b7e3-40e5ab61b9f1",{"status":"Accepted"}]
[site  ] DEBUG 2023/09/25 18:13:38 ----
[lp-1  ] DEBUG 2023/09/25 18:13:38 charge power: 3890W
[site  ] DEBUG 2023/09/25 18:13:38 pv power: 570W
[site  ] DEBUG 2023/09/25 18:13:38 battery soc: 82%
[site  ] DEBUG 2023/09/25 18:13:38 battery power: 170W
[site  ] DEBUG 2023/09/25 18:13:38 grid power: 2419W
[site  ] DEBUG 2023/09/25 18:13:38 site power: 2589W
[lp-1  ] DEBUG 2023/09/25 18:13:38 charge currents: [5.8 5.6 5.7]A
[lp-1  ] DEBUG 2023/09/25 18:13:38 detected active phases: 3p
[lp-1  ] DEBUG 2023/09/25 18:13:38 charge total import: 6193.572kWh
[lp-1  ] DEBUG 2023/09/25 18:13:38 charger status: C
[lp-1  ] WARN 2023/09/25 18:13:38 charger logic error: disabled but charging
[lp-1  ] ERROR 2023/09/25 18:13:38 charger: cannot enable: transaction already running
```

binzman, 2023-09-25:
@wlanrouter I did have a look at your logfile. Seems like the transaction is started by the Wallbox and not by evcc. Thus evcc can't control that transaction even though it knows the transaction ID.

You need to make sure that the Wallbox does not start a transaction on its own. It should only start a transacting if evcc tells the Wallbox to do so.

How did you setup your Wallbox?

binzman, 2023-09-25:
If you connect to the evbox using the app, the first setting is "charger access control". This should not be set to "Auto" but needs to be set to "Swipe card/key fob to start".

You might also need to ad a card under the card setting for this to work.

I wrote a German comment about this a couple of month ago:

andig, 2023-09-25:
> But @andig , isn't OCPP telling evcc the transaction ID? In this case "1"

@wlanrouter interesting- we do actually get it via the meter values. I wasn't aware of that.

> Seems like the transaction is started by the Wallbox and not by evcc. Thus evcc can't control that transaction even though it knows the transaction ID.

@binzman if we do accidentally get it- as above- why don't we just grab it and try to stop it or take over? This won't work without meter apparently (unless we can find other messages containing the transaction id) but it seems worth a try?

binzman, 2023-09-25:
I was primarily thinking about how to help @wlanrouter to get his setup working with the current evcc version.

I actually also never noticed that we sometimes get the transaction ID "by accident". I guess we really could take over the transaction in those cases. However this might be very device specific and probably error prone. But yes it is definitely worth a try

wlanrouter, 2023-09-26:
Yes indeed, the wallbox settings are "automatically". Thank you @binzman 
But having evcc taking over the transaction ID would be pretty convenient. In case of evcc beeing unavailable (WiFi down, Raspi down, evcc down), the car would at least start charging. evcc could take over as soon as it is available again.

I will change it to "Swipe key" for now, but I would be happy to run some tests if the evcc team is interested in taking over a wallbox transaction! Could be maybe something more than only one wallbox model is doing. Please let me know if I could help anyhow @andig

wlanrouter, 2023-09-26:
Run a test. Changed wallbox to "swipe card". Started evcc...attached cable...nothing...swiped card...started full charging...transaction already running

Logfile:
```
evcc -c evcc.yaml
[main  ] INFO 2023/09/26 09:08:39 evcc 0.120.2
[main  ] INFO 2023/09/26 09:08:39 using config file: evcc.yaml
[main  ] INFO 2023/09/26 09:08:39 starting ui and api at :7070
[db    ] INFO 2023/09/26 09:08:39 using sqlite database: /home/pi/.evcc/evcc.db
[ocpp  ] DEBUG 2023/09/26 09:08:40 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2023/09/26 09:09:14 chargepoint connected, registering: EVB-P21124921
[ocpp  ] TRACE 2023/09/26 09:09:14 sent JSON message to EVB-P21124921: [2,"4160213522","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2023/09/26 09:09:16 received JSON message from EVB-P21124921: [2,"2ca5401b-df12-4647-b7e3-40e5b0ba0239","Heartbeat",{}]
[ocpp  ] TRACE 2023/09/26 09:09:16 sent JSON message to EVB-P21124921: [3,"2ca5401b-df12-4647-b7e3-40e5b0ba0239",{"currentTime":"2023-09-26T07:09:16Z"}]
[ocpp  ] TRACE 2023/09/26 09:10:14 sent JSON message to EVB-P21124921: [2,"2547959607","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
[ocpp  ] TRACE 2023/09/26 09:10:14 TriggerMessage MeterValues for EVB-P21124921:
[ocpp  ] TRACE 2023/09/26 09:10:14 received JSON message from EVB-P21124921: [3,"2547959607",{"status":"Accepted"}]
[ocpp  ] DEBUG 2023/09/26 09:10:14 enabling meter watchdog
[ocpp  ] TRACE 2023/09/26 09:10:15 received JSON message from EVB-P21124921: [2,"2ca54058-df12-4647-b7e3-40e51f61ce99","Heartbeat",{}]
[ocpp  ] TRACE 2023/09/26 09:10:15 sent JSON message to EVB-P21124921: [3,"2ca54058-df12-4647-b7e3-40e51f61ce99",{"currentTime":"2023-09-26T07:10:15Z"}]
[ocpp  ] TRACE 2023/09/26 09:11:14 sent JSON message to EVB-P21124921: [2,"916690258","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":1}]
[ocpp  ] TRACE 2023/09/26 09:11:15 received JSON message from EVB-P21124921: [3,"916690258",{"status":"Accepted"}]
[ocpp  ] TRACE 2023/09/26 09:11:15 TriggerMessage StatusNotification for EVB-P21124921: Accepted
[ocpp  ] TRACE 2023/09/26 09:11:15 received JSON message from EVB-P21124921: [2,"2ca54092-df12-4647-b7e3-40e5dd4cbafe","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","timestamp":"2023-09-26T07:11:14Z"}]
[ocpp  ] TRACE 2023/09/26 09:11:15 sent JSON message to EVB-P21124921: [3,"2ca54092-df12-4647-b7e3-40e5dd4cbafe",{}]
[site  ] WARN 2023/09/26 09:11:15 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/09/26 09:11:15 deprecated: use 'battery' instead of 'batteries'
[site  ] WARN 2023/09/26 09:11:15 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://https://docs.evcc.io/docs/reference/configuration/site#residualpower
[site  ] WARN 2023/09/26 09:11:15 bufferStartSoc must be larger than bufferSoc
[site  ] INFO 2023/09/26 09:11:15 site config:
[site  ] INFO 2023/09/26 09:11:15   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/26 09:11:15     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/26 09:11:15     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/26 09:11:15     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2023/09/26 09:11:15 loadpoint 1:
[lp-1  ] INFO 2023/09/26 09:11:15   mode:        pv
[lp-1  ] INFO 2023/09/26 09:11:15   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/09/26 09:11:15   meters:      charge ✓
[lp-1  ] INFO 2023/09/26 09:11:15     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/09/26 09:11:15 phase timer inactive
[lp-1  ] DEBUG 2023/09/26 09:11:15 pv timer inactive
[lp-1  ] DEBUG 2023/09/26 09:11:15 guard timer inactive
[site  ] WARN 2023/09/26 09:11:15 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/26 09:11:15 ----
[lp-1  ] DEBUG 2023/09/26 09:11:15 charge power: 0W
[ocpp  ] TRACE 2023/09/26 09:11:15 received JSON message from EVB-P21124921: [2,"2ca54092-df12-4647-b7e3-40e5d2b7f2fe","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Available,NoError,No error.,0,Off,0,{0,7920,100},{6193721,0},{A,11952,12028,61568},0,0,320,2023-09-26T07:11:14Z,0,3,{228,227,227,0,0,0,0,0,0},220,0,1,135,0,250,900,5009"}]
[ocpp  ] TRACE 2023/09/26 09:11:15 sent JSON message to EVB-P21124921: [3,"2ca54092-df12-4647-b7e3-40e5d2b7f2fe",{"status":"Accepted"}]
[site  ] DEBUG 2023/09/26 09:11:15 pv power: 1992W
[ocpp  ] TRACE 2023/09/26 09:11:15 received JSON message from EVB-P21124921: [2,"2ca54093-df12-4647-b7e3-40e55a216fff","Heartbeat",{}]
[ocpp  ] TRACE 2023/09/26 09:11:15 sent JSON message to EVB-P21124921: [3,"2ca54093-df12-4647-b7e3-40e55a216fff",{"currentTime":"2023-09-26T07:11:15Z"}]
[site  ] DEBUG 2023/09/26 09:11:15 battery soc: 9%
[site  ] DEBUG 2023/09/26 09:11:15 battery power: -290W
[site  ] DEBUG 2023/09/26 09:11:15 grid power: -994W
[site  ] DEBUG 2023/09/26 09:11:15 giving priority to battery charging at soc: 9%
[site  ] DEBUG 2023/09/26 09:11:15 site power: -994W
[lp-1  ] DEBUG 2023/09/26 09:11:15 charge currents: [0 0 0]A
[lp-1  ] ERROR 2023/09/26 09:11:15 charge total import: not available
[lp-1  ] DEBUG 2023/09/26 09:11:15 charger status: A
[lp-1  ] INFO 2023/09/26 09:11:15 car disconnected
[lp-1  ] DEBUG 2023/09/26 09:11:15 set charge mode: pv
[lp-1  ] DEBUG 2023/09/26 09:11:15 set min current: 6
[... cut ...]
[site  ] DEBUG 2023/09/26 09:17:35 pv power: 2087W
[site  ] DEBUG 2023/09/26 09:17:35 battery soc: 10%
[site  ] DEBUG 2023/09/26 09:17:35 battery power: -303W
[site  ] DEBUG 2023/09/26 09:17:35 grid power: -1072W
[site  ] DEBUG 2023/09/26 09:17:35 giving priority to battery charging at soc: 10%
[site  ] DEBUG 2023/09/26 09:17:35 site power: -1072W
[lp-1  ] DEBUG 2023/09/26 09:17:35 charge currents: [0.4 0.4 5.7]A
[lp-1  ] DEBUG 2023/09/26 09:17:35 charge total import: 6193.789kWh
[lp-1  ] DEBUG 2023/09/26 09:17:35 charger status: A

```

pjjjv, 2023-09-27:
I would have also expected the charger to send a StartTransaction to evcc, even delayed if offline.

wlanrouter, 2023-09-27:
For me this seems like the transaction was started by evcc. That why I dont get it:
```
[ocpp  ] TRACE 2023/09/26 09:14:33 received JSON message from EVB-P21124921: [2,"2ca54158-df12-4647-b7e3-40e596b00198","Authorize",{"idTag":"045D82BA857480"}]
[ocpp  ] TRACE 2023/09/26 09:14:33 sent JSON message to EVB-P21124921: [3,"2ca54158-df12-4647-b7e3-40e596b00198",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2023/09/26 09:14:34 received JSON message from EVB-P21124921: [2,"2ca5415a-df12-4647-b7e3-40e56cdfcba1","StatusNotification",{"connectorId":1,"status":"Preparing","errorCode":"NoError","timestamp":"2023-09-26T07:14:34Z"}]
[ocpp  ] TRACE 2023/09/26 09:14:34 sent JSON message to EVB-P21124921: [3,"2ca5415a-df12-4647-b7e3-40e56cdfcba1",{}]
[ocpp  ] TRACE 2023/09/26 09:14:34 received JSON message from EVB-P21124921: [2,"2ca5415a-df12-4647-b7e3-40e5e7ba05a2","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Preparing,NoError,No error.,0,Off,0,{20,7920,100},{6193721,0},{B,11952,9100,61590},0,0,320,2023-09-26T07:14:34Z,0,3,{229,227,227,0,0,0,0,0,0},220,0,0,125,0,250,900,5009"}]
[ocpp  ] TRACE 2023/09/26 09:14:34 sent JSON message to EVB-P21124921: [3,"2ca5415a-df12-4647-b7e3-40e5e7ba05a2",{"status":"Accepted"}]
```
@binzman You have changed it to "Swipe card" too. Can you send me your log so I can spot the difference?

binzman, 2023-09-27:
Based on your last messages the transactions are still startet by your wallbox.

>> 3.) Attached cable -> nothing
If the wallbox is set to "swipe card" and all conditions for evvc to start a transaction are met, this is where evcc should start the session. For some reason that didn't happen. Based on your logs I can't see why. 

You could try the following:
- set/keep the wallbox on "swipe card"
- attach the cable of the wallbox to the car
- open your local evcc webpage. It should say that the car is connected but not charging
- set the evcc mode to "Fast" (Schnell) using the evcc webpage
- evcc should start the loading session. The car should start charging

If this doesn't work, please describe what happend on the evcc webpage and provide a full log that covers all above steps.

>>4. Swiped some shopping card -> started full charging (which it should not given to the meter values and settings)
- If you do this, you will manually start a charging session which will be startet by the wallbox and not by evcc
- The wallbox needs to be set to "swipe card". However you should not start a session by actually swiping the card. The session should be started by evcc without the need to physically swipe the card

In the log shown in your comment https://github.com/evcc-io/evcc/issues/10044#issuecomment-1737270657 I do not see evcc starting the session. Instead the session is started by the wallbox

wlanrouter, 2023-09-28:
YAY 🥳 
That was is. I was not supposed to swipe any card but to let evcc start the transaction! In my log file the house battery was to empty, thats why evcc did not start ist. Now is working! Guys, you are stunning. Thank you so much!

If "talking over a wallbox transaction ID" is still something you want me to test something, let me know.

Thank you!

andig, 2023-09-29:
> If "talking over a wallbox transaction ID" is still something you want me to test something, let me know.

Let me see if I can prepare a prototype.

wlanrouter, 2023-09-29:
If so, let me know. I will help you as much as I can.

andig, 2023-09-29:
Please give https://github.com/evcc-io/evcc/pull/10109 a try. Should take over a running transaction when evcc starts and finds one.

cryptev, 2023-09-29:
> Please give #10109 a try. Should take over a running transaction when evcc starts and finds one.

Does it require a active meter in the Wallbox to test? If not, I can try testing it, too. Also using the same box but without meter.

andig, 2023-09-30:
TxnId is part of the meter updates, hence doing so requires a meter.

closed 2023-09-30 by andig