evcc-io/evcc #10045: Eastron SDM72DM-V2 wird nicht mehr über tcpip ausgelesen

gulchi, 2023-09-24:
### Describe the bug

Der Verbrauch der Wallbox wird bei mir über einen Zähler Eastron SDM72DM-V2 ausgeslesen. Der Zähler wird per TCP über ein extra ModbusTCP zu ModbusRTU Konverter ausgelesen. Der Konverter erledigt bereits die Umwandlung von ModbusRTU zu ModbusTCP.

Vor Version 0.120.0 konnte ich den Zähler über den Parameter _modbus: tcpip_ auslesen. Jetzt scheint es so als ob evcc auch mit diesen Einstellungen versucht via ModbusRTU (rs485tcpip) auf den Zähler zuzugreifen. Korrekt wäre in meinem fall aber direkt ModbusTCP (tcpip).

### Steps to reproduce

Modbus TCP Converter auf RS485 mit integrierter Umsetung verwenden.
Konfig wie unten einrichten.

### Configuration details

```yaml
meters:
- name: wallbox4meter
  type: template
  template: eastron
  usage: charge
  modbus: tcpip
  id: 1
  host: 172.16.14.81 # Hostname
  port: 502 # Port 

chargers:
- type: template
  template: pulsarplus
  name: wallbox4
```

### Log details

```text
Sep 24 14:48:24 evcc evcc[423]: [lp-1  ] ERROR 2023/09/24 14:48:24 charge meter: read failed: EOF
Sep 24 14:48:25 evcc evcc[423]: [lp-1  ] ERROR 2023/09/24 14:48:25 charge meter: read failed: modbus: response crc '3019' does not match expected '37066'
Sep 24 14:48:25 evcc evcc[423]: [lp-1  ] ERROR 2023/09/24 14:48:25 charge meter: read failed: dial tcp 172.16.14.81:502: connect: connection refused
Sep 24 14:48:25 evcc evcc[423]: [lp-1  ] ERROR 2023/09/24 14:48:25 charge total import: read failed: dial tcp 172.16.14.81:502: connect: connection refused
```

### What type of operating system are you running?

Linux

### Version

0.120.2

andig, 2023-09-24:
connect: connection refused

Dein Zähler antwortet nicht- da kann evcc nix machen.

closed 2023-09-24 by andig

ForwardThinkerDev, 2023-09-24:
[image]

ForwardThinkerDev, 2023-09-24:
[image]

ForwardThinkerDev, 2023-09-24:
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: da01e9dea4052c109fdb3daae0295d29f49ee39dfcd331508373461410801381

interval: 10s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: siemens-pac2200 
  id: 1  
  host: 192.168.0.56  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1

chargers:
- type: template
  template: abb 
  id: 49
  host: 192.168.0.49  
  port: 502  
  modbus: tcpip  
  name: wallbox1

vehicles:
- type: template
  template: bmw 
  title: BMW i3s  
  user:
  password: 
  vin: 
  capacity: 42.2  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  minSoc: 25  
  targetSoc: 90  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev1

loadpoints:
- title: Carport
  charger: wallbox1
  vehicle: ev1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: My home
  meters:
    pv: pv1