evcc-io/evcc #10047: "PV Min" Ladung beginnt immer 3-phasig obwohl zu wenig Überschuss zur Verfügung steht

phrozen77, 2023-09-24:
### Describe the bug

Wie im Titel - obwohl beim Anstecken vom Model 3 an der Go-E Gemini Flex (Modus PV Min, automatischer Phasen-Wechsel) wenig (zu wenig) Überschuss für 3-Phasiges Laden vorhanden ist, startet die Ladung scheinbar immer 3-Phasig und skaliert dann zurück auf 1-phasig.

Zum einen wird dadurch das Laden kurzzeitig wieder unterbrochen, es entsteht unnötiger Netzbezug - vor allem weil eben eigentlich auch schon vorher klar war, dass der vorhandene Überschuss nicht reicht.

Ist das so erwünscht / works as designed?

### Steps to reproduce

1. Überschussladen im Modus "PV Min" starten
2. Es steht zu wenig Überschuss für 3-phasiges Laden zur Verfügung
3. Ladung startet trotzdem direkt 3-phasig
4. nach 3 Minuten wird auf 1-phasig umgeschalten

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  lp-1: trace
  tesla: trace

# unique installation id
plant: ...

interval: 10s # control cycle interval

sponsortoken: ...

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.222.9
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.222.9  
  name: pv2
- type: template
  template: tibber-pulse
  usage: grid
  name: tibber
  token: ".."
  homeid: ".."

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.222.104  
  name: wallbox3

loadpoints:
- title: Go-E
  charger: wallbox3
  mode: minpv
  phases: 0
[... 551 lines cut ...]
Sep 24 09:05:31 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:31 pv charge current: 4.53A = 6A + -1.47A (337W @ 1p)
Sep 24 09:05:40 evcc evcc[5056]: [site  ] DEBUG 2023/09/24 09:05:40 ----
Sep 24 09:05:40 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:40 charge power: 1352W
Sep 24 09:05:41 evcc evcc[5056]: [site  ] DEBUG 2023/09/24 09:05:41 pv power: 1687W
Sep 24 09:05:42 evcc evcc[5056]: [site  ] DEBUG 2023/09/24 09:05:42 grid power: 343W
Sep 24 09:05:42 evcc evcc[5056]: [site  ] DEBUG 2023/09/24 09:05:42 site power: 343W
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 charge currents: [5.97 0 0]A
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 detected active phases: 1p
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 charge total import: 1360.203kWh
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 charger status: C
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 soc estimated: 20.32% (vehicle: 20.00%)
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 vehicle soc: 20%
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 vehicle soc limit: 100%
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 vehicle range: 86km
Sep 24 09:05:42 evcc evcc[5056]: [lp-1  ] DEBUG 2023/09/24 09:05:42 pv charge current: 4.51A = 6A + -1.49A (343W @ 1p)
Sep 24 09:05:50 evcc evcc[5056]: [site  ] DEBUG 2023/09/24 09:05:50 ----
```

### What type of operating system are you running?

Linux

### Version

0.120.2

GrimmiMeloni, 2023-09-25:
@andig maybe related to #9543 ? Not exactly the same, but here also no scaling happens when it would be beneficial.

TobiasHuber1980, 2023-09-29:
nicht der gleiche fehler, aber wohl ähnlich zu #10105

andig, 2023-09-29:
#9543 ist rein optisch. Wir schalten zu oft wenn die angebotene Leistung nicht genutzt wird. #10105 dürfte gar nicht passieren, ohne Log aber unklar.

**Hier** dürften wir nicht einschalten wenn wir wissen, dass wir prinzipiell 1p3p schalten müssten. Dazu gehört auch, dass wir damit nicht warten sollten wenn die letzte Schalthandlung lange genug her ist, also sofort machen statt auf Timer warten.

Beim Versuch die Timer zu redesignen bin ich leider schon 2x gescheitert :/

TobiasHuber1980, 2023-09-29:
Hier bei mir ähnlich, ich denke aber evcc startet keine Ladung, es läuft nur der Timer 3P->1P runter.

[main  ] INFO 2023/09/29 17:44:13 evcc 0.120.2 (fc0bcf95)
[main  ] INFO 2023/09/29 17:44:13 using config file: /etc/evcc.yaml
[main  ] INFO 2023/09/29 17:44:13 starting ui and api at :7070
[site  ] WARN 2023/09/29 17:44:19 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://https://docs.evcc.io/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/09/29 17:44:19 site config:
[site  ] INFO 2023/09/29 17:44:19   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/09/29 17:44:19     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/09/29 17:44:19     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/09/29 17:44:19     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/09/29 17:44:19   vehicles:
[site  ] INFO 2023/09/29 17:44:19     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2023/09/29 17:44:19     vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/09/29 17:44:19 loadpoint 1:
[lp-1  ] INFO 2023/09/29 17:44:19   mode:        pv
[lp-1  ] INFO 2023/09/29 17:44:19   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/09/29 17:44:19   meters:      charge ✓
[lp-1  ] INFO 2023/09/29 17:44:19     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/09/29 17:44:19 phase timer inactive
[lp-1  ] DEBUG 2023/09/29 17:44:19 pv timer inactive
[lp-1  ] DEBUG 2023/09/29 17:44:19 guard timer inactive
[site  ] WARN 2023/09/29 17:44:19 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/09/29 17:44:19 ----
[lp-1  ] DEBUG 2023/09/29 17:44:19 charge power: 0W
[main  ] INFO 2023/09/29 17:44:19 new version available: 0.120.3
[site  ] DEBUG 2023/09/29 17:44:19 pv power: 1056W
[site  ] DEBUG 2023/09/29 17:44:19 battery soc: 97%
[site  ] DEBUG 2023/09/29 17:44:19 battery power: 2W
[site  ] DEBUG 2023/09/29 17:44:19 grid power: -803W
[site  ] DEBUG 2023/09/29 17:44:19 site power: -801W
[lp-1  ] DEBUG 2023/09/29 17:44:19 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/29 17:44:19 charge total import: 110.676kWh
[lp-1  ] DEBUG 2023/09/29 17:44:19 charger status: B
[lp-1  ] DEBUG 2023/09/29 17:44:19 charge start energy: 111kWh
[lp-1  ] INFO 2023/09/29 17:44:19 car connected
[lp-1  ] DEBUG 2023/09/29 17:44:19 vehicle api refresh
[lp-1  ] DEBUG 2023/09/29 17:44:19 pv timer elapse
[lp-1  ] DEBUG 2023/09/29 17:44:19 pv timer inactive
[lp-1  ] DEBUG 2023/09/29 17:44:19 charger: guard elapse
[lp-1  ] DEBUG 2023/09/29 17:44:19 guard timer inactive
[lp-1  ] DEBUG 2023/09/29 17:44:19 charge total import: 110.676kWh
[lp-1  ] INFO 2023/09/29 17:44:21 vehicle updated: unknown -> BMW i3
[lp-1  ] DEBUG 2023/09/29 17:44:21 vehicle status: B
[lp-1  ] DEBUG 2023/09/29 17:44:21 vehicle soc: 92%
[lp-1  ] DEBUG 2023/09/29 17:44:21 vehicle range: 272km
[lp-1  ] DEBUG 2023/09/29 17:44:21 available power 801W < 4140W min 3p threshold
[lp-1  ] DEBUG 2023/09/29 17:44:21 start phase scale1p timer
[lp-1  ] DEBUG 2023/09/29 17:44:21 phase scale1p in 3m0s
[lp-1  ] DEBUG 2023/09/29 17:44:21 pv charge current: 1.16A = 0A + 1.16A (-801W @ 3p)
[lp-1  ] DEBUG 2023/09/29 17:44:21 pv enable timer reset
[lp-1  ] DEBUG 2023/09/29 17:44:21 pv timer inactive
[site  ] DEBUG 2023/09/29 17:44:34 ----
[lp-1  ] DEBUG 2023/09/29 17:44:34 charge power: 0W
[site  ] DEBUG 2023/09/29 17:44:34 pv power: 1062W
[site  ] DEBUG 2023/09/29 17:44:34 battery soc: 97%
[site  ] DEBUG 2023/09/29 17:44:34 battery power: 1W
[site  ] DEBUG 2023/09/29 17:44:34 grid power: -809W
[site  ] DEBUG 2023/09/29 17:44:34 site power: -807W
[lp-1  ] DEBUG 2023/09/29 17:44:34 vehicle odometer: 13816km
[lp-1  ] DEBUG 2023/09/29 17:44:34 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/29 17:44:34 charge total import: 110.676kWh
[lp-1  ] DEBUG 2023/09/29 17:44:34 charger status: B
[lp-1  ] DEBUG 2023/09/29 17:44:34 available power 807W < 4140W min 3p threshold
[lp-1  ] DEBUG 2023/09/29 17:44:34 phase scale1p in 2m47s
[lp-1  ] DEBUG 2023/09/29 17:44:34 pv charge current: 1.17A = 0A + 1.17A (-807W @ 3p)
[site  ] DEBUG 2023/09/29 17:44:49 ----
[lp-1  ] DEBUG 2023/09/29 17:44:49 charge power: 0W
[site  ] DEBUG 2023/09/29 17:44:49 pv power: 1032W
[site  ] DEBUG 2023/09/29 17:44:49 battery soc: 97%
[... 141 lines cut ...]
[site  ] DEBUG 2023/09/29 17:47:34 pv power: 1016W
[site  ] DEBUG 2023/09/29 17:47:34 battery soc: 97%
[site  ] DEBUG 2023/09/29 17:47:34 battery power: 2W
[site  ] DEBUG 2023/09/29 17:47:34 grid power: -766W
[site  ] DEBUG 2023/09/29 17:47:34 site power: -764W
[lp-1  ] DEBUG 2023/09/29 17:47:34 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/29 17:47:34 charge total import: 110.676kWh
[lp-1  ] DEBUG 2023/09/29 17:47:34 charger status: B
[lp-1  ] DEBUG 2023/09/29 17:47:34 available power 764W < 4140W min 3p threshold
[lp-1  ] DEBUG 2023/09/29 17:47:34 phase scale1p in 0s
[lp-1  ] DEBUG 2023/09/29 17:47:34 !! scalePhases: GetPhases 0p <> 1p wanted
[lp-1  ] DEBUG 2023/09/29 17:47:34 switched phases: 1p
[lp-1  ] DEBUG 2023/09/29 17:47:34 phase timer inactive
[lp-1  ] DEBUG 2023/09/29 17:47:34 pv charge current: 3.32A = 0A + 3.32A (-764W @ 1p)
[site  ] DEBUG 2023/09/29 17:47:49 ----
[lp-1  ] DEBUG 2023/09/29 17:47:49 charge power: 0W
[site  ] DEBUG 2023/09/29 17:47:49 pv power: 1008W
[site  ] DEBUG 2023/09/29 17:47:49 battery soc: 97%
[site  ] DEBUG 2023/09/29 17:47:49 battery power: 2W
[site  ] DEBUG 2023/09/29 17:47:49 grid power: -773W
[site  ] DEBUG 2023/09/29 17:47:49 site power: -772W
[lp-1  ] DEBUG 2023/09/29 17:47:49 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/29 17:47:49 charge total import: 110.676kWh
[lp-1  ] DEBUG 2023/09/29 17:47:49 charger status: B
[lp-1  ] DEBUG 2023/09/29 17:47:49 pv charge current: 3.35A = 0A + 3.35A (-772W @ 1p)

andig, 2023-09-29:
Bitte formatierne

TobiasHuber1980, 2023-09-29:
> Bitte formatierne

done!

GrimmiMeloni, 2023-09-30:
> Beim Versuch die Timer zu redesignen bin ich leider schon 2x gescheitert :/

Ich will mitmachen beim Scheitern! 😉 
Der erste Kaffee ist noch nicht im Kopf, aber vielleicht ist die Lösung relativ einfach? (siehe PR)

TobiasHuber1980, 2023-09-30:
nur dass bei @phrozen77 wohl auch tatsächlich geladen wird
```[lp-1  ] DEBUG 2023/09/24 09:01:30 charge power: 793W```

während bei mir
```[lp-1  ] DEBUG 2023/09/29 20:41:52 charge power: 0W```

PR #10113 (2023-10-01): Loadpoint: elapse PV scale timer immediately if not charging
fix #10047
closed 2023-10-01 by andig