evcc-io/evcc #10050: When vehicle is (re-) connected to wallbox, EEBus communication is interrupted and available current changes from 16A to 6A

dreisieben, 2023-09-24:
### Describe the bug

Wallbox: Elli Charger Connect
Car: Tesla Model Y

Without evcc: The car charges with 16A (3 phases). 
With evcc: 
- After connecting the car to the wallbox, the available current changes from 16A to 6A (as seen in the car or in the Tesla App). Also, the rightmost LED on the wallbox turns yellow, indicating loss of communication between wallbox and evcc. Charging then proceeds with 6A (3 phases).
- After interrupting the charging and restarting evcc, everything goes back to normal: the available current goes back to 16A, the LED on the wallbox turns green, and charging proceeds with 16A.
- After disconnecting and then reconnecting the car, the symptoms described above return (i.e., current=6A, LED=yellow).

Expected behavior:  On (re-) connect, the current should be 16A, and the LED should remain green.

In the attached log (from which I deleted most heartbeat messages to decrease the file size), there are two disconnect/reconnect events. After the first disconnect, there are EEBus messages indicating temporary loss of communication with the wallbox. After the second disconnect there are no such messages. The symptoms are the same in both cases.

Thank you very much for looking into this. And please let me know if I should provide more details.

### Steps to reproduce

1. Disconnect the car from the wallbox and then reconnect it.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  cache: error
  lp-1: debug
  tesla: debug

# unique installation id
plant: *****

interval: 30s

meters:
- name: grid1
  type: custom
  power:
    source: mqtt
    broker: 127.0.0.1:1883
    topic: vzlogger/data/chn0/raw
- name: dummyPV
  type: custom
  power:
    source: const
    value: 0

chargers:
- name: wallbox1
  type: template
  template: elliconnect 
  ski: 45CB D09E 1EFB 5807 E7CB 5D0A 4F16 8AE6 F061 AE39  
  ip: 192.168.188.102  

vehicles:
- name: tesla
  title: Model Y
  type: template
  template: tesla 
  mode: now
  accessToken: *****
  refreshToken: *****
  capacity: 62  
[... 414 lines cut ...]
[eebus ] DEBUG 2023/09/24 18:24:05 Send: d:_i:47859_Elli-Wallbox-2103A15G8Q notify 453 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/09/24 18:24:06 Send: d:_i:47859_Elli-Wallbox-2103A15G8Q notify 454 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/09/24 18:24:06 Recv: d:_i:47859_Elli-Wallbox-2103A15G8Q: to DeviceConfiguration notify 559 DeviceConfigurationKeyValueListData
[eebus ] DEBUG 2023/09/24 18:24:06 Recv: d:_i:47859_Elli-Wallbox-2103A15G8Q: to DeviceConfiguration notify 560 DeviceConfigurationKeyValueListData
[eebus ] DEBUG 2023/09/24 18:24:06 Recv: d:_i:47859_Elli-Wallbox-2103A15G8Q: to ElectricalConnection notify 561 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2023/09/24 18:24:06 Recv: d:_i:47859_Elli-Wallbox-2103A15G8Q: to ElectricalConnection notify 562 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2023/09/24 18:24:06 Recv: d:_i:47859_Elli-Wallbox-2103A15G8Q: to ElectricalConnection notify 563 ElectricalConnectionPermittedValueSetListData
[eebus ] DEBUG 2023/09/24 18:24:06 Recv: d:_i:47859_Elli-Wallbox-2103A15G8Q: to Measurement notify 564 MeasurementListData
[eebus ] DEBUG 2023/09/24 18:24:06 Send: d:_i:47859_Elli-Wallbox-2103A15G8Q notify 455 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2023/09/24 18:24:15 Send: d:_i:47859_Elli-Wallbox-2103A15G8Q notify 466 DeviceDiagnosisHeartbeatData
[site  ] DEBUG 2023/09/24 18:24:15 ----
[lp-1  ] DEBUG 2023/09/24 18:24:15 charge power: 0W
[site  ] DEBUG 2023/09/24 18:24:15 pv power: 0W
[site  ] DEBUG 2023/09/24 18:24:15 grid power: 327W
[site  ] DEBUG 2023/09/24 18:24:15 site power: 327W
[lp-1  ] DEBUG 2023/09/24 18:24:15 charger status: B
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.2

DerAndereAndi, 2023-09-25:
Hi,

some quick points:

- The first disconnect is due to a bug in the Elli software. There is nothing we can do.
- Whenever the limit is < the maximum (with or without a HEMS connected), the LED will be yellow
- Whenever a paired HEMS is not connected (or not compatible with the wallbox), the wallbox will set the limit to 6A
- No charge power/currents are ever provided, meaning evcc does not know if the car is actually charging. This can be because of a bug in the Elli software (happens when a car is connected and the connection to evcc is interrupted), or if you charger has no meter installed
- evcc never sets a limit in the log, so the limit you get is the default from the wallbox
- charger status is always B, meaning evcc knows an EV is connected, but does not know it is charging

This is not a problem of evcc, but of the buggy software in the wallbox. Try to get a connect to evcc without the EV being connected at the time of the connection. Then connect the EV and it might work better.

dreisieben, 2023-09-25:
Many thanks for the quick reply. The workaround you suggested (starting evcc without the EV being connected, and waiting for 30s between disconnecting and reconnecting the EV) indeed works, at least for the few times I tried. 

Two follow-up questions:

- Can I do something in evcc.yaml, like setting minCurrent or maxCurrent, to circumvent/overwrite the buggy wallbox software? (BTW, there are small inconsistencies in the evcc documentation. The parameters are defined with an uppercase C, but in the examples they appear with a lowercase c, like mincurrent. I don't know if this makes a difference.)
- Should I contact the wallbox manufacturer and ask them (as a customer) to fix the buggy software?

Thanks very much again!