evcc-io/evcc #10054: Tesla keeps disconnecting after x hours

arekpietrzyk, 2023-09-25:
### Describe the bug

Charger: Tesla Wallbox 3
Car: Tesla Y

When providing authorization and refresh tokens to the EVCC config file, UI reports that the car is connected and everything works well.

After a couple of hours, I'm still checking to confirm, but it looks like after 8 hours, UI reports my Tesla as disconnected.

Tesla is connected to the Internet - I can use connectivity inside the car. Wallbox is reachable in my local network. Overall local network performance is excellent, and there are no issues there.

My last test from this morning was to restart the EVCC process, and it started working again. Yesterday I replaced tokens (both) with fresh ones.

My feeling is, that the EVCC does not refresh the access token when invalid, but probably refreshes it when the EVCC is starting.

### Steps to reproduce

1. provide the tokens
2. start the application
3. connect the car
4. wait 8 or more hours
5. check if your car is still connected

### Configuration details

```yaml
default config, nothing special
```

### Log details

```text
no logs
```

### What type of operating system are you running?

Linux

### Version

0.120.2

StefanSchoof, 2023-09-25:
> My feeling is, that the EVCC does not refresh the access token when invalid, but probably refreshes it when the EVCC is starting.

And if you create a log, that shows this someone can check it and is not only a feeling.

(From the Bug Form report:
> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

)

arekpietrzyk, 2023-09-25:
I'll try to generate them today and add.

In general system logs I found only timeouts to the wallbox local IP address, but it was different issue and related strictly to my local network - which is resolved now.

andig, 2023-09-25:
Waiting for log

closed 2023-09-25 by andig

arekpietrzyk, 2023-09-26:
Unfortunately, everything worked well in the last 24 hours. This is good in general, but not for the problem I've noticed. I'll reopen the issue when proper logs will be generated.

The only strange issue I found in logs is below when for some reason gateway DNS was unavailable when used for checking the hostname. But it was before some changes related to the network were made.

`Sep 22 00:17:44 hq evcc[362]: [main  ] ERROR 2023/09/22 00:17:44 creating vehicle tesla failed: cannot create vehicle 'template': cannot create vehicle 'tesla': cannot get vehicles: Get "https://owner-api.teslamotors.com/api/1/vehicles": Post "https://auth.tesla.com/oauth2/v3/token": dial tcp: lookup auth.tesla.com on 192.168.86.1:53: dial udp 192.168.86.1:53: connect: network is unreachable`