evcc-io/evcc #10057: Enable coarse current for vehicle templates (or for wallbox configuration)

FabianRig, 2023-09-25:
**Is your feature request related to a problem? Please describe.**
Coarse current (https://docs.evcc.io/docs/reference/configuration/vehicles/#features-coarsecurrent) cannot be used with vehicle templates. However, most vehicles are implemented using templates. One popular example is Tesla which doesn't support mA resolution.

**Describe the solution you'd like**
I'd like to have an option to either set the wallbox or the vehicle to "1A steps only".

**Describe alternatives you've considered**
None.

**Additional context**
My wallbox is capable of mA steps whereas my car is not. This leads to an oscillating behaviour (slightly positive / slightly negative power at the house connection point). EVCC should be configurable in such a way that it "knows" that only full 1A steps are accepted by the car.

StefanSchoof, 2023-09-25:
You can use every car without an template and configure the "coarsecurrent" feature there. The Tesla template has this already in the template: 
https://github.com/evcc-io/evcc/blob/c920893bf63acf1bf1d0be6e11dfb4273e911d4a/templates/definition/vehicle/tesla.yaml#L62C6-L62C6

Which car do you have, that need the coarsecurrent feature?

andig, 2023-09-25:
BMW i3 needs it, too.

FabianRig, 2023-09-25:
> The Tesla template has this already in the template: https://github.com/evcc-io/evcc/blob/c920893bf63acf1bf1d0be6e11dfb4273e911d4a/templates/definition/vehicle/tesla.yaml#L62C6-L62C6
> 
> Which car do you have, that need the coarsecurrent feature?

Thank you for your reply! Actually, I'm using the teslamate template -- which also has the coarsecurrent feature enabled (https://github.com/evcc-io/evcc/blob/c920893bf63acf1bf1d0be6e11dfb4273e911d4a/templates/definition/vehicle/teslamate.yaml). I missed that before, thanks for the hint!

Unfortunately, I'm still seeing oscillation when charging a Tesla Model Y. EVCC seems to allow more power, then it ends up drawing power from the grid and reduces it in the next step. I'll try to upload a Grafana snapshot to explain it better.

[image]

RTTTC, 2023-09-25:
What charger do you have and what is your control `interval` ?

FabianRig, 2023-09-25:
Charger is a WARP2 (https://docs.evcc.io/docs/devices/chargers/#warp-charger-smart). 

Interval is 20s, same problem with 10s and 13s.

RTTTC, 2023-09-25:
Then log would help..