evcc-io/evcc #10073: modbus/eastron broken?

RenatusRo, 2023-09-26:
### Describe the bug

after building version 0.120.2 from sources the eastron modbus meter readings ceased working
(applies to 0.120.1 as well, reverting to  0.120.0 fixes the issue)

### Steps to reproduce

0.120.2 sends 
modbus: send 01 04 00 34 00 02 30 05

0.120.0 sent
modbus: send 00 01 00 00 00 06 01 04 00 34 00 02

### Configuration details

```yaml
meters:
  - name: grid
    type: template
    template: eastron
    usage: grid    

    modbus: tcpip
    id: 1
    host: 192.168.1.237
    port: 502 # Port 

  - name: pv
    type: template
    template: eastron
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.1.238
    port: 502 # Port
```

### Log details

```text
NOT working in version 0.120.2

[site  ] DEBUG 2023/09/26 13:40:17 ----
[lp-1  ] DEBUG 2023/09/26 13:40:17 charge power: 0W
[modbus] TRACE 2023/09/26 13:40:18 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2023/09/26 13:40:28 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2023/09/26 13:40:38 modbus: send 01 04 00 34 00 02 30 05
[site  ] ERROR 2023/09/26 13:40:48 pv 1 power: read failed: read tcp 194.7.72.174:33148->192.168.1.238:502: i/o timeout
[site  ] DEBUG 2023/09/26 13:40:48 pv power: 0W
[modbus] TRACE 2023/09/26 13:40:48 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2023/09/26 13:40:58 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2023/09/26 13:41:09 modbus: send 01 04 00 34 00 02 30 05
[site  ] ERROR 2023/09/26 13:41:19 grid meter: read failed: read tcp 194.7.72.174:43052->192.168.1.237:502: i/o timeout
[site  ] ERROR 2023/09/26 13:41:19 grid meter: read failed: read tcp 194.7.72.174:43052->192.168.1.237:502: i/o timeout

WORKING in version 0.120.0
[site  ] DEBUG 2023/09/26 13:58:12 ----
[lp-1  ] DEBUG 2023/09/26 13:58:12 charge power: 0W
[main  ] INFO 2023/09/26 13:58:13 new version available: 0.120.2
[modbus] TRACE 2023/09/26 13:58:13 modbus: send 00 01 00 00 00 06 01 04 00 34 00 02
[modbus] TRACE 2023/09/26 13:58:13 modbus: recv 00 01 00 00 00 07 01 04 04 45 cb 3c 2c
[modbus] TRACE 2023/09/26 13:58:13 modbus: send 00 02 00 00 00 06 01 04 00 48 00 02
[modbus] TRACE 2023/09/26 13:58:13 modbus: recv 00 02 00 00 00 07 01 04 04 45 38 37 68
[site  ] DEBUG 2023/09/26 13:58:13 pv power: 6504W
[modbus] TRACE 2023/09/26 13:58:13 modbus: send 00 01 00 00 00 06 01 04 00 34 00 02
[modbus] TRACE 2023/09/26 13:58:13 modbus: recv 00 01 00 00 00 07 01 04 04 c5 7b 45 72
[site  ] DEBUG 2023/09/26 13:58:13 grid power: -4020W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.2 (9d8b94cb9a)

RenatusRo, 2023-09-26:
actually evcc version 0.120.2 (3b1d615239)

VolkerK62, 2023-09-26:
there was a change regarding modbus in 0.120.1
this is reverted with https://github.com/evcc-io/evcc/pull/10046
so, you should try this commit

RenatusRo, 2023-09-26:
ah, works again indeed - thx.

closed 2023-09-26 by RenatusRo