evcc-io/evcc #10075: Easee: outdated OP_MODE by charger reported on startup

GrimmiMeloni, 2023-09-26:
### Describe the bug

I have observed during initial negotiation on evcc startup, that the charger actually sends an outdated OP_MODE.
This left evcc with the false information, that the charger was waiting for the car, while the car was actually charging (charge was already ongoing, prior to restarting evcc).

I was able to manually fix this by requesting a poll of the OP_MODE from the charger via the API.
I suggest we put this into the startup sequence when we create the charger to ensure we get an up2date information about the charger's state.

### Steps to reproduce

Not sure.
I have seen numerous restarts with charges in flight just work. 

### Configuration details

```yaml
N/A
```

### Log details

```text
... this is the ProductUpdate received on evcc startup...
[easee ] TRACE 2023/09/26 17:19:59 ProductUpdate EHWHL6VE: (2023-09-26 12:07:30 +0000 UTC) CHARGER_OP_MODE 6

... and this is the LP date shown the very first interval...

[lp-1  ] DEBUG 2023/09/26 17:20:04 charge power: 1482W
[lp-1  ] DEBUG 2023/09/26 17:20:05 charge currents: [5.99 0.011 0.014]A
[lp-1  ] DEBUG 2023/09/26 17:20:07 charger status: B

... while car is (correctly) detetec as charging...
[lp-1  ] DEBUG 2023/09/26 17:20:08 vehicle status: C
```

### What type of operating system are you running?

Docker container

GrimmiMeloni, 2023-09-26:
@andig please assign

GrimmiMeloni, 2023-10-14:
Maybe related to https://github.com/evcc-io/evcc/issues/10265#issuecomment-1756309102 (Theory: same underlying issue - wrong op_mode ?)

GrimmiMeloni, 2023-11-23:
reopened.
The issue is actually not limited to restarts, but to re-negotiations.
So whenever the signalR connection breaks (which it tends to do from time to time, don't know why), we can get the same problem.
Caught this today. Charge started, evcc showed waiting for car, while 11kW charge was going.
Log shows renegotiation happening immediately after charge start, and then op_mode 7 (waiting for auth) is reported as most recent. 
D*** Easse, I really liked this being stable.... 

@andig I never dove deep into the SignalR stack. Is there a simple way to hook into the negotiation phase somehow, so that I could move the change I introduced into the charger startup into there?

andig, 2023-11-24:
@GrimmiMeloni es wird der Connector aufgerufen:

```go
	client, err := signalr.NewClient(context.Background(),
		signalr.WithConnector(c.connect(ts)),
		signalr.WithReceiver(c),
		signalr.Logger(easee.SignalrLogger(c.log.TRACE), false),
	)

```

GrimmiMeloni, 2023-12-02:
Closing, as this was addressed again by #10907

GrimmiMeloni, 2023-12-16:
Reopening this. Apparently the fix in #10907 does not suffice, because of wonky timing. Here's my car connecting yesterday evening. In this case, all the "goodness" comes together. Outdated status on SignalR (re)connect, plus conflicting op_mode state with identical timestamps. The effect was that the charge was happening, but the UI stated "waiting for vehicle".

```
[easee ] TRACE 2023/12/15 18:32:09 ProductUpdate EHWHL6VE: (2023-12-15 17:32:08 +0000 UTC) CHARGER_OP_MODE 2

--- SIGNALR reconnect

[easee ] TRACE 2023/12/15 18:32:11 ProductUpdate EHWHL6VE: (2023-12-15 17:32:01 +0000 UTC) CHARGER_OP_MODE 7
[easee ] TRACE 2023/12/15 18:32:11 ProductUpdate EHWHL6VE: (2023-12-15 17:32:09 +0000 UTC) CHARGER_OP_MODE 6
[easee ] TRACE 2023/12/15 18:32:12 ProductUpdate EHWHL6VE: (2023-12-15 17:32:10 +0000 UTC) CHARGER_OP_MODE 3
[easee ] TRACE 2023/12/15 18:32:12 ProductUpdate EHWHL6VE: (2023-12-15 17:32:10 +0000 UTC) CHARGER_OP_MODE 6
```

(Side note, the timestamps in the messages look strange. Ignoring the final update, the sequencing of the op_codes is correct, i.e. 2 -> 7 -> 6 -> 3. However, according to the timestamps, 7 (waiting for auth) happened 7s before the car was even connected.)

What this show is that we need to do the polling more proactively instead of just when the signalR connection is (re)established. The good news is, that we have enough status from the charger to detect inconsistencies. I will prepare a PR.

PR #10403 (2023-10-20): Easee: poll op_mode on startup
fix #10075 

tested locally, looking solid. Ready to merge.
closed 2024-01-11 by GrimmiMeloni