evcc-io/evcc #10078: "Schaltschutz aktiv" wenn ich die evcc Seite aufrufe

userwithoutpassword, 2023-09-26:
### Describe the bug

Jedesmal wenn ich die evcc Seite aufrufe zeigt evcc mir an "Schaltschutz aktiv für 18 s" und läuft dann runter. Lade ich die Seite erneut macht er es erneut. 

### Steps to reproduce

1.evcc aufrufen
2.
3.
...

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

v.0.120.2

mdkeil, 2023-09-26:
Das hatte ich auch mal mit einer älteren Version. Ein evcc Neustart hat damals das Problem gelöst.

VolkerK62, 2023-09-27:
hatte ich auch schon mal. Ich meine, ein kurzer Moduswechsel auf `aus`, reichte um die Verwirrtheit zu beenden.

VolkerK62, 2023-09-27:
siehe auch https://github.com/evcc-io/evcc/issues/8375 und https://github.com/evcc-io/evcc/issues/9431

closed 2023-09-27 by andig