evcc-io/evcc #10097: Check growatt-hybrid template

premultiply, 2023-09-28:
Hallo, ich wollte nun nochmal Rückmeldung zu dem Problem, dass der Batteriewechselrichter die falschen Erzeugungswerte liefert. Ich habe für meinen MOD 10KTL3-XH die config nun wie folgt angepasst:

[image]

Dadurch erhalte ich die korrekten Erzeugungswerte für diesen Wechselrichter (mypv1). Das sehe ich auch daran, dass die volle Erzeugungsleistung genutzt wird um die Batterie zu laden. Die anderen beiden WR übernehmen den Hausbedarf und Einspeisung. Die Werte decken sich auch mit den Eastron SDM Zählern die ich installiert habe. Der Wert (5253W), der aus dem Template bzw. Register 3001 kommt erschließt sich mir nicht.

[image]

Meiner Meinung nach ist als das Register 3001 für "PV total power" für den MOD 10KTL3-XH in meiner Konstellation nicht korrekt. Ich komme nur über die Summierung der einzelnen PV Eingänge auf den richtigen Wert.

_Originally posted by @k3flo in https://github.com/evcc-io/evcc/discussions/6819#discussioncomment-7110675_

andig, 2023-10-24:
@premultiply was fehlt hier- einfach Template anpassen?

premultiply, 2023-10-24:
Wenn dem so ist, ja.
Ich schaue mal heute Abend.

andig, 2023-10-26:
@k3flo könntest Du das bitte testen?

PR #10477 (2023-11-10): growatt-hybrid-tlxh: fix pv power
Fixes #10097
commit 4df3b14e (2023-11-10): Growatt-hybrid-tlxh: fix pv power (#10477)

Fixes #10097
closed 2023-11-10 by andig