evcc-io/evcc #10098: ENTSOE European Grid forecasting

hyperbart, 2023-09-28:
For non German countries there is no free/easy integration with dynamic grid pricing or CO2.
ENTSOE provides this for all European countries

**Describe the solution you'd like**
Enable ENTSOE data consumption for CO2 and dynamic pricing info (day ahead) to be used in charge planning.

**Describe alternatives you've considered**
Electricity Maps is not working with free tier.

**Additional context**
https://m-transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html
https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=true&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=28.09.2023+00:00|CET|DAY&biddingZone.values=CTY|10YBE----------2!BZN|10YBE----------2&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2)

bolisaf, 2024-01-07:
do I understand correctly that the ENTSOE integration has been implemented? I can't seem to find a mention of it in the documentation. how do I use it?

andig, 2024-01-07:
See https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml#L184

/cc @VolkerK62 haben wir die Tarife in der Doku?

VolkerK62, 2024-01-08:
diesen leider nicht