evcc-io/evcc #10100: Modbusproxy: Trotz "readonly" einige Schreibbefehle durchlassen

docolli, 2023-09-28:
Ich möchte einen Feature Request einreichen, da die Integration von EVCC in E3DC Systeme mit einer E3DC Wallbox (Wallbe) aktuell nur mit unschönen Einschränkungen möglich ist. So richtig stabil läuft es nur, wenn die Wallbox im E3DC Hauskraftwerk (HKW) dauerhaft gelöscht wird und nur über EVCC eingebunden ist. Der große Nachteil dieser Konfiguration ist aber, dass nun der Energiebedarf der Wallbox nicht mehr vom Hausverbrauch abgezogen wird und somit in sämtlichen Statistiken im E3DC Portal ununterscheidbar abgelegt wird. Das erdulde ich seit April 2023, aber nur weil ich die Ladung über EVCC so genial finde.

Es gibt einige Diskussionen hier im EVCC Forum wie es mit E3Dc und EVCC zusammen am besten klappt (https://github.com/evcc-io/evcc/discussions/2751 / https://github.com/evcc-io/evcc/discussions/7785). Der aktuell vorgeschlagene Weg läuft so, dass man in EVCC den Modbusproxy aktiviert und im readonly Modus betreibt. Anschliesend wird die Wallbox im HKW über die IP von EVCC konfiguriert, somit kann das HKW nur noch lesend auf die Wallbox zugreifen und EVCC kann die Leistung wie gewünscht regeln. Theoretisch sollte das so klappen, aber in der Praxis funktioniert das nicht.

1) Das HKW erkennt die Wallbox nur, wenn man die Einbindung während einer laufenden Ladung macht.
2) Nach 1 Ladung, maximal nach 2 Ladungen, hat das HKW jedoch ein Problem mit dem nur lesenden Zugriff und die Abfrage von Energiewerten von der Wallbox wird unterbrochen. Dabei kommt es häufig vor, dass EVCC die Ladung beendet, das HKW aber die letzte Ladeleistung behält und somit diesen Wert dauerhaft aufzeichnet, bis man z.B. die Wallbox im HKW löscht oder EVCC neu startet. Das versaut einem die E3DC Statistik der aufgewendeten Energie für die E-Auto Ladung dann komplett.

Anscheinend benötigt das E3DC zwingend einige Schreibzugriffe auf die Wallbox, bzw. (meine Vermutung) bricht die Kommunikation ab, wenn bestimmte Antworten nicht kommen.

### **Mein Feature Request bezieht sich daher auf dem Modbusproxy. Ist es möglich diesen so zu erweitern, dass bestimmte Befehle (konfigurierbar über evcc.yaml) doch zur Wallbox durchkommen und die Antworten zurück ans HKW gesendet werden?**

Selbstverständlich würde ich mich nach Implementierung selbst um Testung und um das Herausfinden, welche Schreibbefehle notwendig sind, kümmern. Aktuell fehlen mir einfach die Tools zur näheren Analyse.

@1b-w @lorbetzki

andig, 2023-09-29:
> Ist es möglich diesen so zu erweitern, dass bestimmte Befehle (konfigurierbar über evcc.yaml) doch zur Wallbox durchkommen und die Antworten zurück ans HKW gesendet werden?

Was würde das konkret heissen?

docolli, 2023-09-29:
Ich schau mir grad nochmals meinen anderen Beitrag zu Problemen mit dem readonly Modus des Modbusproxy an. https://github.com/evcc-io/evcc/discussions/7785
Dort sehe ich, dass es im Log beim Ende der Ladung zwei Nachrichten für die coil Adresse 400 gab (charge enable), die kurz vor dem Ende von 1500 auf 1400 sich geändert haben. Kenn mich da nicht aus, könnte aber heißen von "true" auf "false".
Da das E3DC HKW stets versucht die Wallbox zu konfigurieren, könnte hier das Problem liegen, dass das HKW das "false" erkennt, aber selbst der Meinung ist, die Ladung sollte auf "true" stehen und schickt auf coil Adresse 400 ein "true", was aber nicht bei der Wallbox ankommt, das HKW bekommt weiterhin "false" zurück. Es könnte sein, dass der E3DC Code dann in einen Zustand springt, in dem er die Wallbox nicht weiter abfrägt und somit permanent die letzten Werte loggt.
Wobei meine Beobachtung, dass ein Neustart von EVCC genügt, damit es wieder läuft, dem widerspricht, dass es ein Problem im E3DC Code ist. 

Mal eine konkrete Frage: Irgendwo zum Modbusproxy  habe ich mal gelesen "Coils not implemented". Ist das noch gültig?

Könnte man (testweise) Nachrichten für die coil Adresse 400 durch den Proxy durchleiten, ohne dass sie gefiltert werden?