evcc-io/evcc #10102: SoC Cupra Born wird nicht länger an openWB übertragen: Error from EVCC: 2023/09/28 12:25:02 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/[number removed]/vehicles/[number removed]/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout (LV0)

DrUdoZucker, 2023-09-29:
### Describe the bug

SoC Abruf in openWB Wallbox arbeitet seit 28.9.23 ca. 11:07 nicht mehr

### Steps to reproduce

Abruf SoC

### Configuration details

```yaml
Software 1.9.271, openWB mit Kabel am Ethernet, Internet verfügbar
Die Cupra App (Android) zeigt weiterhin den korrekten SoC, nur openWB wird nicht upgedated.
```

### Log details

```text
PID: 24104: Lp1: Error from EVCC: 2023/09/28 12:25:02 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/[number removed]/vehicles/[number removed]/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout (LV0)
```

### What type of operating system are you running?

Linux

StefanSchoof, 2023-09-29:
Klingt so wie #9860

DrUdoZucker, 2023-09-29:
#9860 wurde "repariert" und Fehlermeldung war etwas anders - nur so als Beobachtung, Problem kann dennoch gleich oder verwandt sein. AndyG wird das sicherlich korrekt beurteilen. Problem mit ausbleibendem SoC Update in openWB ist, dass die openWB (oder das EVCC Modul in openWB?) das Ausbleiben eines Updates nicht als Fehler detektiert. In anderen Worten, wird ein Fahrzeug nach einer Fahrt mit geringem SoC angesteckt, die openWB jedoch weiterhin einen veralteten Wert nutzt - da kein Update- kann ein automatischer Ladestart ausbleiben.

StefanSchoof, 2023-09-29:
Ich habe gerade lokal mal ein `nslookup ola.prod.code.seat.cloud.vwgroup.com` probiert und das schlägt ab und an fehl. Würde ein Fehler auf Seite von VW vermuten.

DrUdoZucker, 2023-09-29:
Die MyCupra App unter Android arbeitet jedoch korrekt - könnte aber ein anderer Zugang zum VW Server sein.

andig, 2023-09-29:
Siehe https://community.fly.io/t/dns-resolution-fails-inside-app-part-2/15675

mairj2345, 2023-10-01:
Gibts zu dem Thema Neuigkeiten? Hab das gleiche Problem.

andig, 2023-10-01:
Wenn es welche gäbe würden sie hier stehen.

mairj2345, 2023-10-02:
Hi Andi,
Gerade hat die Abfrage zum ersten Mal wieder funktioniert. Um 11.53 war sie noch gescheitert.
Sehr merkwürdig.

andig, 2023-10-02:
Temporäres Serverproblem bei uns. Hab leider noch keine Antwort im verlinkten Thread :(

andig, 2023-10-09:
Sollte behoben sein:

> Hi, we looked at this and found it was related to a DNSSEC configuration issue with our recursive DNS server. It should be working now.

Großartig :)

closed 2023-10-09 by andig