evcc-io/evcc #10115: Terra 11kw V1.6.6 OCPP Anbindung Fehler

Sixten85, 2023-09-30:
### Describe the bug

Habe folgende EVCC Fehlermeldung nach Update auf der Terra ABB Wallbox auf Version 1.6.6.
charger: cannot: enable transaction already running
charger logic error: disabled but charging

Zudem wird kein Lastenmanagement nach Solar Verfügbarkeit betrieben sondern das max. Mögliche oder das was ich per VW App vorgebe.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
#open evcc at http://evcc.local:7070
network:
  schema: http
  host: 192.168.x.x # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ocpp: trace
  lp-1: debug
  site: debug

#unique installation id
plant: d617592b4c49aeea96d7779a86b4021f8d0a7376712faaa357f7173fb0c92736

interval: 10s # control cycle interval

meters:
- name: grid1
  type: sma
  uri: 192.168.x.x
- name: pv2
  type: sma
  uri: 192.168.x.x
  password: xx
  

vehicles:
- type: id
  title: ID.4 Pure
  user: xxx
  password: xxx
  vin: WVxxx
  capacity: 52
  name: ev5
  
chargers:
- name: TERRA
  type: ocpp
  stationid: TACxxx
  GetConfiguration: false
  metervalues: Energy.Active.Import.Register,Power.Active.Import

site:
  title: HOME
  meters:
    grid: grid1
    pv:
    - pv2

loadpoints:
[... 101 lines cut ...]
[ocpp  ] TRACE 2023/09/30 12:22:19 sent JSON message to TACW###: [3,"5598260",{}]
[site  ] DEBUG 2023/09/30 12:22:19 ----
[lp-1  ] DEBUG 2023/09/30 12:22:19 charge power: 6499W
[site  ] DEBUG 2023/09/30 12:22:20 pv power: 7566W
[site  ] DEBUG 2023/09/30 12:22:20 grid power: 324W
[site  ] DEBUG 2023/09/30 12:22:20 grid powers: [-1355 861 819]W
[site  ] DEBUG 2023/09/30 12:22:20 grid currents: [-5.89 4.13 3.73]A
[site  ] DEBUG 2023/09/30 12:22:20 site power: 324W
[lp-1  ] DEBUG 2023/09/30 12:22:20 charge total import: 0.913kWh
[lp-1  ] DEBUG 2023/09/30 12:22:20 charger status: C
[lp-1  ] DEBUG 2023/09/30 12:22:20 soc estimated: 51.58% (vehicle: 50.00%)
[lp-1  ] DEBUG 2023/09/30 12:22:20 vehicle soc: 52%
[lp-1  ] DEBUG 2023/09/30 12:22:20 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/09/30 12:22:20 vehicle range: 143km
[lp-1  ] WARN 2023/09/30 12:22:20 charger logic error: disabled but charging
[lp-1  ] ERROR 2023/09/30 12:22:20 charger: cannot enable: transaction already running
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.120.3

andig, 2023-09-30:
> charger: cannot enable: transaction already running

Laufen Transaktionen können nicht übersteuert werden- es sei denn https://github.com/evcc-io/evcc/pull/10109 ändert das. Solange muss evcc die Transaktion starten.

closed 2023-09-30 by andig

Sixten85, 2023-09-30:
Aber was hat sich denn durch das Terra Update der Wallbox gändert?  Mit der 1.6.5. Version lief Evcc und Lastenmangement sauber durch machte absolut keine Probleme. Jetzt scheint man ja irgendwas umgestellt zu haben. Gibts schon Erfshrubg mit dem neueren ABB Update 1.6.9?