evcc-io/evcc #10123: simplebox test1

wb3103, 2023-09-30:
[test1.txt](https://github.com/evcc-io/evcc/files/12775422/test1.txt)

wb3103, 2023-09-30:
Pulsares simplebox Modbus Steuerung Ladestrom

closed 2023-09-30 by andig

wb3103, 2023-10-01:
Python, Steuerung des Ladestromes mit dem Ziel, während des Ladevorganges den Strombezug auf Null zu minimieren.
Es wird ein Shelly 3EM verwendet, der die momentane Leistung (Überschuss/Bezug) bereitstellt. 
Als Wallbox wird eine Pulsares SimpleBox Wallbox bis 11kW mit PulCharge Ladesteuerung verwendet.
Es wird modbus benutzt. Zur Kopplung mit dem Netz wird ein [HF Elfin-EW11 - RS485 to WIFI] genutzt.
Ausführung über cronjob mit 'python3 modbus_ladestrom'
[modbus_ladestrom.txt](https://github.com/evcc-io/evcc/files/12778110/modbus_ladestrom.txt)

andig, 2023-10-02:
Das bekommen wir hin 👍🏻

wb3103, 2023-10-02:
Ich verwende momentan nur einen Zähler (3EM) will aber über einen zweiten Zähler (EM31) den tatsächlichen Ladestrom sowie die eingespeiste Arbeit anzeigen. Ist ja eigentlich nicht nötig, ist eben in wenig Spielerei.