evcc-io/evcc #10132: go-e Charge session stopped and reauthenticate required

tpatzig, 2023-10-01:
I'm using the Go-e charger flex Gemini with evcc PV mode. In general this works very well charging only the energy overhead, including phase switch. Thanks for the great project and your effort.

The only annoying thing is, that when I configure the go-e charger to require authorization, then whenever evcc pauses/stops the charging, because the overhead is no longer enough, then a re-authenticate is required to continue. This is annoying and thus I turned authentication mode off to have it fully automated. 

Ideally I would like to have the go-e charger run in Authentication-required mode.
For me this sounds like the same as in https://github.com/evcc-io/evcc/pull/10077. Maybe the charge process just needs to be paused, not fully stopped to not require reauthenticate.
I only want to authenticate once, when I plug in my car.

closed 2023-10-01 by andig

tpatzig, 2023-10-01:
@andig closed because already completed within https://github.com/evcc-io/evcc/pull/10077 ?

andig, 2023-10-01:
No, because all requested information is missing.

tpatzig, 2023-10-01:
What exact information is missing? I tried to answer all the questions you have in the "New Feature Request" template:
------------
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
A clear and concise description of what you want to happen.
A clear and concise description of any alternative solutions or features you've considered.
Add any other context or screenshots about the feature request here.
------------
I explained the problem, described my expectation and the current workaround.

Just closing an issue without any comment gives me the impression, that user feedback/participation is not really welcome.

andig, 2023-10-01:
Hier geht es um einen vermeintlichen Fehler mit einem Charger. Mangels jeglicher Versionsinfo ist nicht mal klar worauf der sich eigentlich bezieht.

tpatzig, 2023-10-02:
Für mich war es eben kein Fehlverhalten, sondern ein fehlendes Feature, deswegen habe ich das "New Feature" issue template ausgefüllt (wo keine Versionsnummern abgefragt werden). 
Beim nächsten längeren Laden beobachte ich das Verhalten erneut mit der latest version und mache einen Bug auf mit logs und versionsinfos.

andig, 2023-10-02:
Vmtl haben wir jetzt catch 22: https://github.com/evcc-io/evcc/pull/9625. Bitte mal in dem Issue kommentieren.

premultiply, 2023-10-02:
Gut möglich.

tpatzig, 2023-10-06:
Ich habe das Verhalten jetzt erneut beobachtet, logs gesammelt und https://github.com/evcc-io/evcc/issues/10215 geöffnet.

tpatzig, 2023-10-09:
@andig habe ich mir angeschaut aber ist mir nicht ganz klar, ob der force state etwas mit der Autehntifizierung zu tun hat. Eventuell, wenn force off (frc=1) bedeuted, man muss in der go-e app nochmal "Authentifizieren" drücken.

andig, 2023-10-09:
Könntest du das evtl mal ohne evcc sondern nur mit go-e Api testen?