evcc-io/evcc #10140: Alfen Eve Single S-Line No route to Host nach zufälliger Ladedauer

Peter1984Git, 2023-10-01:
### Describe the bug

Nach einer nicht definierbaren Ladezeit ist die Wallbox nicht mehr erreichbar, evcc meldet dann "No Route to Host". 
Die Wallbox reagiert dann nicht mehr auf Ping, geht auf Fehler und lädt mit 11 kw unabhängig der Einstellung in evcc. Nach ebenfalls undefinierbarer Zeit ist die Wallbox dann wieder erreichbar.

### Steps to reproduce

1. Auto anstecken und laden
2. warten
3. Fehler tritt irgendwann während dem Laden auf.
...

### Configuration details

```yaml
GNU nano 5.4                                                                                                       evcc.yaml                                                                                                                
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc-pi.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 1234

interval: 30s # control cycle interval

sponsortoken: 1234

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.178.39
  port: 502
  name: grid1
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.178.39
  port: 502
  name: pv2
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.178.39
  port: 502
  name: battery3
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.178.38
  name: pv4

chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.178.110
  port: 502
  modbus: tcpip
  name: wallbox6

vehicles:
- type: template
  template: id
  title: ID.4 GTX
  user: asdf
  password: 1234
  vin: asdf
  capacity: 77
  name: ev5
  targetSoc: 80

loadpoints:
- title: Carport
  charger: wallbox6
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:
   threshold: 0
   delay: 2m
  disable:
   threshold: 200
   delay: 5m

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    battery:
    - battery3
  #bufferSoC: 100
  #prioritySoC: 50
  residualPower: 100

mqtt:
  broker: 192.168.178.82:1883
  topic: evcc
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/10/01 16:51:54 evcc 0.120.3
[main  ] INFO 2023/10/01 16:51:54 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/01 16:51:54 starting ui and api at :7070
[db    ] INFO 2023/10/01 16:51:55 using sqlite database: /home/peterreinhardt/.evcc/evcc.db
[mqtt  ] INFO 2023/10/01 16:51:55 connecting evcc-1217363604 at tcp://192.168.178.82:1883
[mqtt  ] DEBUG 2023/10/01 16:51:55 tcp://192.168.178.82:1883 connected
[alfen ] DEBUG 2023/10/01 16:51:58 detected 1p alfen
[main  ] FATAL 2023/10/01 16:51:58 cannot create charger 'wallbox6': cannot create charger 'template': cannot create charger 'alfen': dial tcp 192.168.178.110:502: connect: no route to host
[main  ] FATAL 2023/10/01 16:51:58 will attempt restart in: 5m0s
[alfen ] ERROR 2023/10/01 16:52:23 heartbeat: dial tcp 192.168.178.110:502: connect: no route to host
[alfen ] ERROR 2023/10/01 16:52:48 heartbeat: dial tcp 192.168.178.110:502: connect: no route to host
```

### What type of operating system are you running?

Linux

### Version

0.120.3

andig, 2023-10-01:
Gerät ist übers Netzwerk nicht zu erreichen. Da kann evcc nix tun :(

closed 2023-10-01 by andig

Peter1984Git, 2023-10-01:
Die Frage ist aber doch, warum passiert das nach einer gewissen Zeit? Es ist unabhängig vom Auto... Ist es eine Einstellung in der Alfen-Software, kennt das Verhalten noch jemand? Teilweise ist auch der Heartbeat weg, aber eben NUR, wenn ein Auto geladen wird.