evcc-io/evcc #10148: Berechnung OUT und Anzeige falsch

Schlafog0r, 2023-10-02:
### Describe the bug

Das Zeigt EVCC an sowohll in Text also auch im Balkendiagramm:
Erzeugung: 4,7 kW
Ladepunkt 3,7 kW
Einspeißung 3.7 kW

Richtig wäre:
Erzeugung: 4,7 kW
Ladepunkt 3,7 kW
Einspeißung 1,0 kW
Das Zeigt auch die Grafik oberhallb des Balkendiagrams an (also die Klammer mit dem Sonnensymbol.

### Steps to reproduce

Sonne scheint und PV produziert Strom. Auto lädt.
Läuft auuf raspberry Pi.
Lief vorher auf 0.120.1 problemlos

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: fd08429846c4411562d1bdd9eae549deb3bc4f056684b3a2edb8280f6a322a66

interval: 10s # control cycle interval

sponsortoken: XYZ

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:

- type: template
  template: kostal-piko 
  usage: pv  
  host: 192.168.0.42  
  name: pv2

- type: template
  template: tasmota 
  usage: grid  
  host: 192.168.0.51
  name: my_grid  

chargers:
- type: template
  template: abl 
  id: 1  
  host: 192.168.0.41  
  port: 502  
  modbus: rs485tcpip  
  name: wallbox5

loadpoints:
- title: ABLWallbox
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicle: Cupra

vehicles:
- name: Cupra
  type: template
  template: offline
  title: Cupra Leon
  phases: 1
  capacity: 13

site:
  title: Kochelsee
  meters:
    grid: my_grid
    pv:
    - pv2

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.36 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.12 # [currency]/kWh
```

### Log details

```text
[main  ] INFO 2023/10/02 12:42:15 evcc 0.120.3
[main  ] INFO 2023/10/02 12:42:15 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/02 12:42:15 starting ui and api at :7070
[db    ] INFO 2023/10/02 12:42:15 using sqlite database: /home/tobias/.evcc/evcc.db
[site  ] INFO 2023/10/02 12:42:15 site config:
[site  ] INFO 2023/10/02 12:42:15   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/10/02 12:42:15     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/02 12:42:15     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/02 12:42:15   vehicles:
[site  ] INFO 2023/10/02 12:42:15     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/02 12:42:15 loadpoint 1:
[lp-1  ] INFO 2023/10/02 12:42:15   mode:        pv
[lp-1  ] INFO 2023/10/02 12:42:15   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/10/02 12:42:15   meters:      charge ✓
[lp-1  ] INFO 2023/10/02 12:42:15     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/10/02 12:42:15 phase timer inactive
[lp-1  ] DEBUG 2023/10/02 12:42:15 pv timer inactive
[lp-1  ] DEBUG 2023/10/02 12:42:15 guard timer inactive
[lp-1  ] INFO 2023/10/02 12:42:15 vehicle updated: unknown -> Cupra Leon
[lp-1  ] DEBUG 2023/10/02 12:42:16 max charge current: 6A
[site  ] WARN 2023/10/02 12:42:16 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/02 12:42:16 ----
[lp-1  ] DEBUG 2023/10/02 12:42:16 charge power: 1449W
[site  ] DEBUG 2023/10/02 12:42:16 pv power: 4715W
[site  ] DEBUG 2023/10/02 12:42:16 grid power: -3741W
[site  ] DEBUG 2023/10/02 12:42:16 site power: -3741W
[lp-1  ] DEBUG 2023/10/02 12:42:16 charge currents: [6.3 0 0]A
[lp-1  ] DEBUG 2023/10/02 12:42:16 charger status: C
[lp-1  ] INFO 2023/10/02 12:42:16 car connected
[lp-1  ] DEBUG 2023/10/02 12:42:16 pv timer elapse
[lp-1  ] DEBUG 2023/10/02 12:42:16 pv timer inactive
[lp-1  ] DEBUG 2023/10/02 12:42:16 charger: guard elapse
[lp-1  ] DEBUG 2023/10/02 12:42:16 guard timer inactive
[lp-1  ] INFO 2023/10/02 12:42:16 start charging ->
[lp-1  ] DEBUG 2023/10/02 12:42:16 wake-up timer: stop
[lp-1  ] DEBUG 2023/10/02 12:42:16 pv charge current: 22.3A = 6A + 16.3A (-3741W @ 1p)
[lp-1  ] DEBUG 2023/10/02 12:42:16 pv timer reset
[lp-1  ] DEBUG 2023/10/02 12:42:16 pv timer inactive
[lp-1  ] DEBUG 2023/10/02 12:42:16 max charge current: 16A
[site  ] DEBUG 2023/10/02 12:42:26 ----
[lp-1  ] DEBUG 2023/10/02 12:42:26 charge power: 3703W
[site  ] DEBUG 2023/10/02 12:42:26 pv power: 4720W
[site  ] DEBUG 2023/10/02 12:42:26 grid power: -3741W
[site  ] DEBUG 2023/10/02 12:42:26 site power: -3741W
[lp-1  ] DEBUG 2023/10/02 12:42:26 charge currents: [16.1 0 0]A
[lp-1  ] DEBUG 2023/10/02 12:42:26 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/02 12:42:26 charger status: C
[lp-1  ] DEBUG 2023/10/02 12:42:26 pv charge current: 32.3A = 16A + 16.3A (-3741W @ 1p)
[site  ] DEBUG 2023/10/02 12:42:36 ----
[lp-1  ] DEBUG 2023/10/02 12:42:36 charge power: 3703W
[site  ] DEBUG 2023/10/02 12:42:36 pv power: 4727W
[site  ] DEBUG 2023/10/02 12:42:36 grid power: -3741W
[site  ] DEBUG 2023/10/02 12:42:36 site power: -3741W
[lp-1  ] DEBUG 2023/10/02 12:42:36 charge currents: [16.1 0 0]A
[lp-1  ] DEBUG 2023/10/02 12:42:36 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/02 12:42:36 charger status: C
[lp-1  ] DEBUG 2023/10/02 12:42:36 pv charge current: 32.3A = 16A + 16.3A (-3741W @ 1p)
[site  ] DEBUG 2023/10/02 12:42:46 ----
[lp-1  ] DEBUG 2023/10/02 12:42:46 charge power: 3703W
[site  ] DEBUG 2023/10/02 12:42:46 pv power: 4724W
[site  ] DEBUG 2023/10/02 12:42:46 grid power: -3741W
[site  ] DEBUG 2023/10/02 12:42:46 site power: -3741W
[lp-1  ] DEBUG 2023/10/02 12:42:46 charge currents: [16.1 0 0]A
[lp-1  ] DEBUG 2023/10/02 12:42:46 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/02 12:42:46 charger status: C
[lp-1  ] DEBUG 2023/10/02 12:42:46 pv charge current: 32.3A = 16A + 16.3A (-3741W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.3

andig, 2023-10-02:
Naja. Wenn du bei 5kw Erzeugung noch 4kW einspeist und trotzdem mit 4kW lädst stimmen wohl eher die Messwerte denn die Rechnung nicht ;)

andig, 2023-10-02:
Was sagt

    Evcc meter

Und welcher der Werte ist falsch?

Schlafog0r, 2023-10-02:
Oh man! @andig Du hattest recht. Der Meter war das Problem. Mein magnetisch gehaltener Tasmota Meter war verrutscht. Mit falschen Werten kann das natürlich nicht richtig gerechnet werden. Ich war nur verwirrt, weil bis auf einen Wert alle anderen richtig waren. Die kommen aber alle aus unterschiedlichen Quellen. Da habe ich nicht richtig geschalten.
Danke für die schnelle Rückmeldung.