evcc-io/evcc #10163: always 3p assumed for charging (although permanently at 1p by manual switch)

RodeoRoland, 2023-10-03:
### Describe the bug

Situation: I have a manual switch for 1p/3p, typically charging with 1p (over 90% of all charging sessions) and only a few times per year switching to 3p (when in a hurry).

Issue: Automatic phase detection works only AFTER the charging started, but not when the car is connected.
==> So when connecting the car, evcc assumes always 3 phases, although only 1 phase is connected (and was used during the last charging). As soon as the charge process starts (pv triggered, manual started, or by a timer) evcc realizes that only 1p is connected. Arising from that:
A) Unfortunately, for timed charging, the system starts too late (as 11kW are assumed, but only 3.7kW are available).
B) PV trigger is not working (waiting for 4.2kW, but only 1.4kW are required)

My current workaround: after connecting the car, I once force charging to start and stop it again after a few seconds. 1p is recognized then and saved for the CURRENT charging session (without reconnecting the car!)

Idea 1: My charger (WARP2) provides the CONNECTED number of phases, too. Possible to use this info instead of the ACITVE number of phases during charging (active = probably found by the currents)?!

From the log file, it looks like the phase detection is not implemented for WARP2 charger?!
[site  ] INFO 2023/10/03 12:34:27 site config:
[site  ] INFO 2023/10/03 12:34:27   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/03 12:34:27     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/03 12:34:27     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/03 12:34:27     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/10/03 12:34:27   vehicles:
[site  ] INFO 2023/10/03 12:34:27     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/10/03 12:34:27 loadpoint 1:
[lp-1  ] INFO 2023/10/03 12:34:27   mode:        off
[lp-1  ] INFO 2023/10/03 12:34:27   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/03 12:34:27   meters:      charge ✓
[lp-1  ] INFO 2023/10/03 12:34:27     charge:    power ✓ energy ✓ currents ✓

Idea 2: If idea 1 doesn't fit to the structure - would it make sense to simply save (non-persistent?) the active number of phases at the end of the last charging session, instead of assuming 3p in every situation?

### Steps to reproduce

1. Connect charger with 1p only
2. Connect car (without starting the charging manually, e.g. at night) => "3p" info in log file
3. Start charging manually (i.e. minpv) => "1p" is recognized by evcc (see log files below)

### Configuration details

```yaml
...

chargers:
- type: template
  template: tinkerforge-warp
  host: xxx.xxx.xxx.xxx
  port: 1883
  user: xxx
  password: xxx
  topic: warp2/xxx
  timeout: 60s
  name: WARP2

vehicles:
- type: template
  template: id
  title: Heidi
  user: xxx
  password: xxx
  vin: xxx
  capacity: 54
  phases: 3
  icon: car
  cache: 3m
  mode: pv
  minSoc: 20
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 16
  name: Auto_ID3

loadpoints:
[... 165 lines cut ...]
Okt 03 12:45:58 odroid evcc[26598]: [site  ] DEBUG 2023/10/03 12:45:58 grid power: 2W
Okt 03 12:45:58 odroid evcc[26598]: [site  ] DEBUG 2023/10/03 12:45:58 grid currents: [4.03 -2.19 -1.93]A
Okt 03 12:45:58 odroid evcc[26598]: [site  ] DEBUG 2023/10/03 12:45:58 site power: 887W
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 charge currents: [6.1 0 0]A
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 detected active phases: 1p
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 charge total import: 1429.058kWh
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 charger status: C
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 soc estimated: 80.04% (vehicle: 80.00%)
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 vehicle soc: 80%
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 vehicle soc limit: 100%
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 vehicle range: 291km
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 pv charge current: 2.14A = 6A + -3.86A (887W @ 1p)
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 site power 887W >= 200W disable threshold
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 pv disable in 1m58s
Okt 03 12:45:58 odroid evcc[26598]: [lp-1  ] DEBUG 2023/10/03 12:45:58 pv disable timer remaining: 1m58s
Okt 03 12:46:12 odroid evcc[26598]: [site  ] DEBUG 2023/10/03 12:46:12 ----
```

### What type of operating system are you running?

Linux

### Version

0.120.3

andig, 2023-10-03:
> I have a manual switch for 1p/3p

How would evcc know about your manual switch? 

> My current workaround: after connecting the car, I once force charging to start and stop it again after a few seconds. 

Usually you would need to *tell* evcc that its 1p now using the api. We can also self-diagnose that by looking at the charger voltages.

> Automatic phase detection works only AFTER the charging started, but not when the car is connected.

Because connecting a car cannot override the the chargers phyisical connection.

andig, 2023-10-03:
> We can also self-diagnose that by looking at the charger voltages.

Warp2 cann actually provide this if it has the `meter_all_values` feature: https://www.warp-charger.com/api.html?v=2#features_meter_all_values

RodeoRoland, 2023-10-03:
Hi andig,

thanks for your feedback! And yes, maybe I'm not an expert of all options of evcc... just starting with it (and it is a really great project!!!) and happy to learn more.

Telling via API would be also an option. But as far as everything in evcc works so nicely out of the box, I expected the issue I encounterd as a bug and not as a missing feature. That's why I didn't start to do something just for me, but reporting to the community :-)

My WARP2 setup provides the voltages ("in" to the box as well as "out" to the car), and yes, it has the "meter_all_values" enabled. The self-diagnosis (as you name it) meets exactly my first idea for a potential solution. 
Is it something that could be implemented? I guess, I'm not the only one using a small PV system that barely brings the power for 3p...

THX!

andig, 2023-10-03:
Could you check if the PR works for you?

RodeoRoland, 2023-10-03:
wow, that was quick. 
I installed evcc via package manager... my system is not set up for manually compiling the source code :-/ 
Will test it on a spare system, but may take until the weekend, sorry for delay.

andig, 2023-10-03:
Please try nightly version tomorrow

closed 2023-10-03 by andig

RodeoRoland, 2023-10-04:
Tried nightly today. Works perfectly! Evcc recognizes 1p/3p now for the WARP2. Thanks,  andig, for the ultra quick support.

andig, 2023-10-04:
That was easy 😉

RodeoRoland, 2024-02-09:
Hi all,

Sorry, there is a lot of information and also massive change in evcc in the past few weeks (which is absolutely great), so maybe I missed some basic changes...

Andig solved this issue long ago. Yesterday I tried to switch (hardware) from 3p to 1p but evcc 124.1 didn't realize... yes, I can switch in the software, I'm aware. But it worked automatically before.
Is there something I miss?

Thx!

andig, 2024-02-09:
Please open new discussion