evcc-io/evcc #10169: evcc ui zeigt Verbrauch am Ladepunkt wenn kein Auto angeschlossen ist

dirkclemens, 2023-10-03:
### Describe the bug

Seit einigen Versionen zeigt die evcc ui Verbrauch am Ladepunkt auch dann an, wenn **kein Auto angeschlossen** ist. 
Und der angezeigte Verbrauch ist identisch mit dem Netzbezug. Die Wallbox ist die go-eCharger Homefix 11kW
evcc.yaml nicht geändert zwischendurch. Wer hat Ideen? 

`evcc charger` zeigt
```
[main  ] INFO 2023/10/03 15:12:20 evcc 0.120.3
[main  ] INFO 2023/10/03 15:12:20 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2023/10/03 15:12:20 connecting evcc_goE104528 at tcp://localhost:1883
Power:          0W
Energy:         3923.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 237V 238V 237V
Charge status:  A
Enabled:        false
Charged:        9.6kWh
Identifier:     <none>
```

### Steps to reproduce

systemctl restart evcc

### Configuration details

```yaml
chargers:
- type: template
  name: go-eCharger
  template: go-e-v3
  host: 192.168.2.44 
```
```

### Log details

```text
Okt 03 15:18:12 smarthome evcc[586570]: [site  ] DEBUG 2023/10/03 15:18:12 site power: 4308W
Okt 03 15:18:12 smarthome evcc[586570]: [lp-1  ] DEBUG 2023/10/03 15:18:12 charge total import: 9508.159kWh
Okt 03 15:18:12 smarthome evcc[586570]: [lp-1  ] DEBUG 2023/10/03 15:18:12 charger status: A
Okt 03 15:18:42 smarthome evcc[586570]: [site  ] DEBUG 2023/10/03 15:18:42 ----
Okt 03 15:18:42 smarthome evcc[586570]: [lp-1  ] DEBUG 2023/10/03 15:18:42 charge power: 4081W
Okt 03 15:18:42 smarthome evcc[586570]: [site  ] DEBUG 2023/10/03 15:18:42 pv power: 1434W
Okt 03 15:18:42 smarthome evcc[586570]: [site  ] DEBUG 2023/10/03 15:18:42 grid power: 4081W
Okt 03 15:18:42 smarthome evcc[586570]: [site  ] DEBUG 2023/10/03 15:18:42 site power: 4081W
Okt 03 15:18:42 smarthome evcc[586570]: [lp-1  ] DEBUG 2023/10/03 15:18:42 charge total import: 9508.201kWh
Okt 03 15:18:42 smarthome evcc[586570]: [lp-1  ] DEBUG 2023/10/03 15:18:42 charger status: A
```
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.3

dirkclemens, 2023-10-03:
sorry, das issue wird sofort geclosed, warum ? 

Das log einer alten Version habe ich natürlich nicht, das war eine Version aus dem Sommer irgendwann, ich kann aber nicht mehr sagen, ab wann das aufgetreten ist. 

evcc meter
[main  ] INFO 2023/10/03 19:18:15 evcc 0.120.3
[main  ] INFO 2023/10/03 19:18:15 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2023/10/03 19:18:16 connecting evcc_goE104528 at tcp://localhost:1883
pv_growatt
----------
Power:  0W
Energy: 4.8kWh

tasmota_sml
-----------
Power:  293W
Energy: 9523.2kWh

andig, 2023-10-04:
> sorry, das issue wird sofort geclosed, warum ?

Wir sind ein kleines Team mit beschränkten Ressourcen- und können es uns nicht leisten, auf fehlende Informationen zu warten. Damit gehen wir auch ganz offen um:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.