evcc-io/evcc #10176: total kWh charged differs in GUI (Charge Energy Overview) and Charge Logs

supeerpeer, 2023-10-04:
### Describe the bug

The total amount of charged kWh for my site displayed in the GUI does not match the logged kWhs anymore, since I upgraded from V0.19.xx to V0.20.3.
Tesla was connected to my Easee Wallbox during the update, but not actively charging over night.
After the update the energy flow overview suddenly showed 4.343 kW at the charger ("out"), but no correlating energy flow on PV or grid on the "in" side of the overview. Obviously the displayed 4.343 kW was false, but I did not care yesterday. After unplugging the car in the morning everything was ok again.

I realized, the statistics on the bottom of the GUI showed only 75% solar usage instead of 88% yesterday.
Looking in detail, the statistics showed a plus of 43 kWh charged from grid over night (total 74kWh from grid), equalling the 4.343 kW for 10 hours (update until unplugging). Although I did not charge!
In my CSV charge session log there is no entry for charging during the night. The total sum charged from grid is still 29 kWh (real value).

For me it looks like the statistics have counted non-existent kWh during the night and EVCC did not realize the total energy flows did not match on the input and output side.
In consequence, the total kWh charged also differes (>300 according to statistics, 250 according to total in charge session log, currently 268 according to debug log)

EVCC is running on a raspberry, if that matters.

Question from my side: can I somehow correct the kWh in the statistics?

Logged energy flow via Influx DB (yellow line shows the "fake" charging load)
[image]

wrong data in GUI (marked red):
[image]

### Steps to reproduce

not able to reproduce currently, but wrong kWh values are currenty still displayed in GUI

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 7xxx

interval: 10s # control cycle interval

sponsortoken: eyxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: xxx  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: wechselrichterkostal
- type: template
  template: kostal-plenticore 
  id: 71  
  host: xxx 
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery1
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: xxx
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: energymeterkostal

chargers:
- type: template
  template: easee 
  user: xxx 
  password: xxx 
  charger: xxx 
  name: wallboxeasee

loadpoints:
- title: Wallbox
  charger: wallboxeasee
  mode: pv
  phases: 0 # automatische Phasenumschaltung
  mincurrent: 6 # 5 Ampere Mindestladestrom für Tesla
  maxcurrent: 16
  resetOnDisconnect: false # eingestelltes Setup bleibt bestehen
  enable: # regelt das Verhalten bei PV-Modus für Ladestart und Phasenumschaltung
    threshold: 0 # x Watt Überschuss müssen anliegen - grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    delay: 2m # für mind. x Minuten, bevor geschaltet wird
  disable: # regelt Verhalten im PV Modus, Ladeunterbrechung und Phasenumschaltung
    threshold: 100 # Definierter Schwellwert, erlaubter Netzbezug, max import power, 0=Netzbezug am Regelpunkt
    delay: 5m # Abschaltung nach x Minuten Überschreitung
  guardduration: 5m # zeitlicher Mindestabsstand für Sperrung und Freigabe Ladung

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3856 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1211 # [currency]/kWh Einspeisevergütung

vehicles:
  name: my_car
  type: template
  template: tesla
  title: Blitz # angezeigter Name
  accessToken: xxx
  refreshToken: xxx
  vin: LRWxxx
  capacity: 62 # Batteriekapazität

site: # Definition der lokalen Gegebenheiten
  title: xxx
  meters:
    grid: energymeterkostal
    pv:
    - wechselrichterkostal
    battery:
    - battery1
  
influx: # automatisches Schreiben der Daten in die Influx DB
  url: http://localhost:8086
  database: evcclog
  token: xxx
  org: xxx
```

### Log details

```text
[site  ] DEBUG 2023/10/04 11:00:53 ----
[lp-1  ] DEBUG 2023/10/04 11:00:53 charge power: 0W
[site  ] DEBUG 2023/10/04 11:00:54 pv power: 1873W
[site  ] DEBUG 2023/10/04 11:00:54 battery soc: 0%
[site  ] DEBUG 2023/10/04 11:00:54 battery power: 0W
[site  ] DEBUG 2023/10/04 11:00:54 grid power: -1601W
[site  ] DEBUG 2023/10/04 11:00:54 site power: -1601W
[lp-1  ] DEBUG 2023/10/04 11:00:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/04 11:00:54 charge total import: 268.489kWh
[lp-1  ] DEBUG 2023/10/04 11:00:54 charger status: A
[site  ] DEBUG 2023/10/04 11:01:03 ----
[lp-1  ] DEBUG 2023/10/04 11:01:03 charge power: 0W
[site  ] DEBUG 2023/10/04 11:01:04 pv power: 3447W
[site  ] DEBUG 2023/10/04 11:01:04 battery soc: 0%
[site  ] DEBUG 2023/10/04 11:01:04 battery power: 0W
[site  ] DEBUG 2023/10/04 11:01:04 grid power: -3506W
[site  ] DEBUG 2023/10/04 11:01:04 site power: -3506W
[lp-1  ] DEBUG 2023/10/04 11:01:04 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/04 11:01:04 charge total import: 268.489kWh
[lp-1  ] DEBUG 2023/10/04 11:01:04 charger status: A
```

### What type of operating system are you running?

Linux

### Version

v0.120.3

andig, 2023-10-04:
I don't understand what you're asking or how the logfile actually shows the problem?

naltatis, 2023-10-04:
The "Charge energy overview" currently uses data based on power values. In the charging log we use energy values we get from the chargers meter (if present). This is usually more accurate.

We are planning to rebuild the "Charge energy overview" so that it uses the same, more correct, data from the charging sessions.

The larger effect you are describing could also be related to https://github.com/evcc-io/evcc/issues/10175

andig, 2023-10-04:
Sounds like we should close this for time being?

supeerpeer, 2023-10-04:
Irgendwie hat mich das Formular verleitet auf Englisch zu Schreiben, auf Deutsch wärs vielleicht verständlicher gewesen :)

@naltatis: ja, das ist im Prinzip der gleiche Effekt wie in #10175, das die Werte bei In und Out nicht zusammenpassen, und in Folge die Statistik andere Daten auswirft, die nicht zu den Charge Session Logs passen.

Bei mir hat das dazu geführt, dass nicht nur der Charge Wert bei 3.434 KW eingefroren war bis zum Abstecken des Autos, sondern auch noch der Heimverbrauch mit 0KW dargestellt wurde (was ja definitiv nix mit der Easee Box zu tun hat, sondern von meinen Kostal geliefert wird).

Danke für die Antworten erstmal! Mich würd nur noch interessieren, kann ich irgendwo den Wert für die Statistik wieder (manuell) korrigieren, dass zumindest meine Statistik-Anzeige wieder stimmt?)

naltatis, 2023-10-04:
Die Werte stehen in der `settings` Tabelle der SQLite Datenbank (`~/.evcc/evcc.db`). 

[image]

Dass der Heimverbrauch für den Zeitraum bei 0 war ist auch nicht unplausible. Den messen wir ja nicht explizit sondern leiten den aus der Leistung am Hauszähler und allen Ladepunkten ab. Der Rest ist dann der Hausverbrauch (mind `0`).

Ich mach den hier dann erstmal zu. Konkreter Fix ist #10175. Der Mittelfristige die Umstellung der Ersparnisberechnung.

closed 2023-10-04 by naltatis

supeerpeer, 2023-10-04:
@naltatis Danke! echt klasse, wie schnell Ihr hier antwortet! :) Hast Du vielleicht noch einen letzten Tipp, wieso ich die markierten Werte in der SQLite Datenbank nicht ändern / überschreiben kann, auch wenn das schon langsam Richtung Offtopic geht? 
Ich greife mit dem DB Browser for SQLite drauf zu, Schreibrechte für die Datenbank habe ich auch hingekriegt. Nur ändern kann ich garnichts an den Werten...
Mit SQL hatte ich bislang noch nix am Hut, ich hatte aber verstanden, dass man die Werte mit dem DB Browser eigentlich editieren können müsste... Danke nochmals!
Vielleicht

naltatis, 2023-10-04:
Puh, ich kenn dein Tool nicht. Läuft evcc noch? Kann sein dass die DB gelockt ist und sich dein Client daher weigert.

supeerpeer, 2023-10-04:
Ne, EVCC hab ich alles gestoppt damit da nix querschiesst. Falls Du ne Tool Empfehlung hast versuch ich auch gern ein anderes...

naltatis, 2023-10-04:
Ich nutze aktuell [SQLiteFlow](https://www.sqliteflow.com) auf dem Mac. Ist kostenpflichtg, gefällt mir aber gut.

supeerpeer, 2023-10-04:
Hab das Problem gefunden, es lag daran, das der Datenbank-Editor kein Schreibrecht auf die Datenbank hatte, die im Ordner /var/lib lag und ich dort nicht editieren konnte. 
Workaround: Datenbank in Home kopiert, geändert und die alte Datenbank ersetzt, jetzt sind die Werte angepasst. Danke nochmals für die Hilfe und Tipps, so langsam werd ich vom ganz blutigen Anfänger zum blutigen Anfänger :)