evcc-io/evcc #10180: Victron Wallbox: evcc fatal error during system start

jubudu, 2023-10-04:
### Describe the bug

EVCC does not start due to missing template victron.
Running on Raspberry 4 with Home Assistant OS

### Steps to reproduce

1. Defined Victron Wallbox in evcs.yaml
2. Restart evcc

...

### Configuration details

```yaml
- name: Victron-Wallbox
    type: template
    template: victron
    # Modbus TCP
    modbus: tcpip
    id: 100
    host: 192.168.10.145 # Hostname
    port: 502 # Port
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/10/04 02:24:21 evcc 0.120.3
[main  ] INFO 2023/10/04 02:24:21 using config file: /config/evcc.yaml
[main  ] INFO 2023/10/04 02:24:21 starting ui and api at :7070
[mqtt  ] INFO 2023/10/04 02:24:22 connecting evcc-737072374 at tcp://localhost:1883
[main  ] FATAL 2023/10/04 02:24:25 cannot create charger 'Victron-Wallbox': cannot create charger 'template': template not found: victron
[main  ] FATAL 2023/10/04 02:24:25 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.120.3

mdkeil, 2023-10-04:
wrong definition 
```
chargers:    
    - name: Victron-Wallbox
    type: template
    template: victron
    # Modbus TCP
    modbus: tcpip
    id: 100
    host: 192.168.10.145 # Hostname
    port: 502 # Port
```
> Wallbox muss sich im Modus "Manual" befinden und Modbus ID 100 konfiguriert sein.

https://docs.evcc.io/docs/devices/chargers#victron-ev-charging-station

andig, 2023-10-04:
Victron ist bisher nur im nightly: https://github.com/evcc-io/evcc/compare/0.120.3...master

closed 2023-10-04 by andig