evcc-io/evcc #10182: Warnung, wenn Grid Meter als Charge Meter am Loadpoint eingetragen wird

VolkerK62, 2023-10-04:
Du hast dein grid Meter zusätzlich am Loadpoint als charger Meter eingetragen. Lösche das einmal und mit ein bisschen Glück ist das Problem dann weg. https://docs.evcc.io/docs/reference/configuration/loadpoints#meter

_Originally posted by @StefanSchoof in https://github.com/evcc-io/evcc/discussions/10179#discussioncomment-7186574_

VolkerK62, 2023-10-04:
Diese Art Fehlkonfiguration ist mir in letzter Zeit mehrmals aufgefallen.
Könnte man das erkennen und eine entsprechende Warnung im Log ausgeben?

andig, 2023-10-04:
Im Prinzip dürfte doch jeder Zähler in Site/Loadpoints nur 1x verwendet werden? Ich glaube das hatten wir schonmal, ist aber wieder raus geflogen. Evrl. weils Probleme mit Config UI gemacht hat, ich hätte aber ne neue Idee wo wir das einbauen könnten.

StefanSchoof, 2023-10-04:
Da es auch Fälle, gab wo Meter definiert, aber nicht genutzt wurde. Ich denke wenn möglich, wäre es auch gut bei den Fall von definierten aber nicht benutzen Meter, zu warnen.

andig, 2023-10-07:
> Da es auch Fälle, gab wo Meter definiert, aber nicht genutzt wurde. Ich denke wenn möglich, wäre es auch gut bei den Fall von definierten aber nicht benutzen Meter, zu warnen.

Das gabs mal, ist aber mit Config UI rausgeflogen. Der erste Fall oben wäre mir daher erstmal lieber.

andig, 2023-10-08:
Lässt sich leider aktuell nicht schön lösen da die Verwendung der Zähler an unterschiedlichen Stellen passiert :/

closed 2023-10-08 by andig