evcc-io/evcc #10201: modbus: response data size '4' does not match request quantity '6'

princemaxwell, 2023-10-05:
### Describe the bug

Hallo zusammen,

vorab ist wichtig, dass der Bug Report sich auf die aktuelle nightly Version bezieht.
In stable 0.120.3 war der Fehler noch **nicht** vorhanden!

Die Fehlermeldung erscheint nur bezogen auf die KEBA P30 x-series. Die Mennekes Amtron wirft keine Fehler aus.

`[lp-2  ] ERROR 2023/10/05 17:16:45 charge meter: modbus: response data size '4' does not match request quantity '6'`

Wenn noch weitere Logs oder Infos benötigt werden, bitte melden.

### Steps to reproduce

Nightly installieren, evcc starten.

### Configuration details

```yaml
unverändert zur stable
```

### Log details

```text
Okt 05 17:16:45 evcc evcc[64811]: [site  ] DEBUG 2023/10/05 17:16:45 pv power: 828W
Okt 05 17:16:45 evcc evcc[64811]: [site  ] DEBUG 2023/10/05 17:16:45 grid power: 83W
Okt 05 17:16:45 evcc evcc[64811]: [site  ] DEBUG 2023/10/05 17:16:45 grid powers: [-171 66 188]W
Okt 05 17:16:45 evcc evcc[64811]: [site  ] DEBUG 2023/10/05 17:16:45 grid currents: [-1.27 2.24 2.27]A
Okt 05 17:16:45 evcc evcc[64811]: [site  ] DEBUG 2023/10/05 17:16:45 site power: 83W
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: send 00 32 00 00 00 06 ff 03 03 f0 00 06
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: recv 00 32 00 00 00 07 ff 03 04 00 00 00 00
Okt 05 17:16:45 evcc evcc[64811]: [lp-2  ] ERROR 2023/10/05 17:16:45 charge meter: modbus: response data size '4' does not match request quantity '6'
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: send 00 33 00 00 00 06 ff 03 04 0c 00 02
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: recv 00 33 00 00 00 07 ff 03 04 00 1e f0 74
Okt 05 17:16:45 evcc evcc[64811]: [lp-2  ] DEBUG 2023/10/05 17:16:45 charge total import: 202.764kWh
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: send 00 34 00 00 00 06 ff 03 03 ec 00 02
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: recv 00 34 00 00 00 07 ff 03 04 00 00 00 03
Okt 05 17:16:45 evcc evcc[64811]: [lp-2  ] DEBUG 2023/10/05 17:16:45 charger status: A
Okt 05 17:16:45 evcc evcc[64811]: [lp-2  ] DEBUG 2023/10/05 17:16:45 next soc poll remaining time: 58m30s
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: send 00 35 00 00 00 06 ff 03 03 e8 00 02
Okt 05 17:16:45 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:45 modbus: recv 00 35 00 00 00 07 ff 03 04 00 00 00 01
Okt 05 17:16:49 evcc evcc[64811]: [site  ] DEBUG 2023/10/05 17:16:49 ----
Okt 05 17:16:49 evcc evcc[64811]: [amtron] TRACE 2023/10/05 17:16:49 modbus: send 00 2d 00 00 00 06 ff 04 03 0f 00 02
Okt 05 17:16:49 evcc evcc[64811]: [amtron] TRACE 2023/10/05 17:16:49 modbus: recv 00 2d 00 00 00 07 ff 04 04 00 00 00 00
Okt 05 17:16:49 evcc evcc[64811]: [lp-1  ] DEBUG 2023/10/05 17:16:49 charge power: 0W
Okt 05 17:16:49 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:49 modbus: send 00 36 00 00 00 06 ff 03 03 fc 00 02
Okt 05 17:16:50 evcc evcc[64811]: [keba  ] TRACE 2023/10/05 17:16:49 modbus: recv 00 36 00 00 00 07 ff 03 04 00 00 00 00
Okt 05 17:16:50 evcc evcc[64811]: [lp-2  ] DEBUG 2023/10/05 17:16:49 charge power: 0W
```

### What type of operating system are you running?

Linux

### Version

0.120.3 (03da1f12)

VolkerK62, 2023-10-05:
vermutlich #10036

andig, 2023-10-05:
> unverändert zur stable

Welche wäre das denn?

closed 2023-10-05 by andig