evcc-io/evcc #10210: panic: interface conversion: *struct { api.Meter; api.MeterEnergy } is not api.Battery: missing method Soc goroutine

tdangschulz, 2023-10-06:
### Describe the bug

Hi,

vorab erstmal vielen Dank für das coole Projekt.

Ich kann EVCC nicht starten, wenn ich eine Battery in der **site** Section aufnehme. Dann crash die Anwendung beim Starten.

Folgende Konfiguration habe ich:
 
- SH10RT Sungrow Hybrid Wechselrichter 
- BYD Speicher
- Elli Pro Wallbox
- ID.3 fahrzeug

### Steps to reproduce

1. Batterie in der Konfiguation aufnehmen
2. Anwendung starten

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s # control cycle interval

plant: xxx

log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.42.167
    port: 502
  - name: pv
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.42.167
    port: 502
  - name: battery
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.42.167
    port: 502

chargers:
  - name: elli
    type: template
    template: ellipro
[... 143 lines cut ...]
[lp-1  ] DEBUG 2023/10/06 10:59:58 charge power: 0W
[site  ] DEBUG 2023/10/06 10:59:58 pv power: 5W
panic: interface conversion: *struct { api.Meter; api.MeterEnergy } is not api.Battery: missing method Soc
goroutine 151 [running]:
github.com/evcc-io/evcc/core.(*Site).updateMeters(0xc0002cf800)
	github.com/evcc-io/evcc/core/site.go:489 +0x1585
github.com/evcc-io/evcc/core.(*Site).sitePower(0xc0002cf800, 0x0, 0x0)
	github.com/evcc-io/evcc/core/site.go:583 +0x67
github.com/evcc-io/evcc/core.(*Site).update(0xc0002cf800, {0x3429d38?, 0xc000b16000})
	github.com/evcc-io/evcc/core/site.go:764 +0x405
github.com/evcc-io/evcc/core.(*Site).Run(0xc0002cf800, 0xc000c363c0, 0x6fc23ac00)
	github.com/evcc-io/evcc/core/site.go:863 +0x206
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:282 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:281 +0x1a13
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.120.3

VolkerK62, 2023-10-06:
```
 - name: battery
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.42.167
    port: 502
```

dann solltest du bei der Batterie auch `usage: battery` nehmen und nicht `usage: grid`

tdangschulz, 2023-10-06:
danke! Das war wohl ganz klar mein Fehler :-)

andig, 2023-10-06:
Das panic ist allerdings trotzdem unschön- das geht bestimmt verträglicher.