evcc-io/evcc #10211: OCPP: charger disable: cannot disable: unknown transaction running

Maddogtet, 2023-10-06:
### Describe the bug

Seit der Version 120.3 bekomme ich jetzt die Meldung: `charger disable: cannot disable: unknown transaction running`

### Steps to reproduce

* Ladevorgang wird bei Überschuss gestartet
* Ladevorgang soll beendet werden, weil Batterie SOC unterschritten wurde und kein Überschuss mehr vorhanden ist
* Es taucht die beschriebene Meldung auf

### Configuration details

```yaml
ite:
  title: Mein Zuhause
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  bufferSoc: 85
  prioritySoc: 80
  #bufferStartSoc: 100 # hat die Hausbatterie Soc x& erreicht startet der ladev>
  residualPower: 200
 # maxGridSupplyWhileBatteryCharging: 650
  #smartCostLimit: 0.10 # 10 ct

loadpoints:
#- title: Garage
- title: Zum Lärchenwald 5A - 1 - ABL EMH2  # display name for UI
  charger: wallbox # Wallbox Referenz
  vehicle: ev1 # Referenz auf Standardfahrzeug
  mode: pv # c
  mincurrent: 6
  maxcurrent: 16
  phases: 3
  enable:
    threshold: -3000
    delay: 1m #1m
  disable:
    threshold: 100
    delay: 30s #low value because we have the battery
  meter: my_meter
  guardduration: 15m #Definiert den zeitlichen Mindestabstand in welchem der St>
```

### Log details

```text
Okt 06 12:48:34 srvdeb evcc[1275183]: [lp-1  ] DEBUG 2023/10/06 12:48:34 pv disable in 0s
Okt 06 12:48:34 srvdeb evcc[1275183]: [lp-1  ] DEBUG 2023/10/06 12:48:34 pv disable timer elapsed
Okt 06 12:48:34 srvdeb evcc[1275183]: [lp-1  ] ERROR 2023/10/06 12:48:34 charger disable: cannot disable: unknown transaction running
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.120.3

andig, 2023-10-06:
Es braucht ein vollständiges Logfile.

Maddogtet, 2023-10-06:
@andig ist das ausreichend?
```
Okt 06 14:00:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:00:27 charge total import: 4183.496kWh
Okt 06 14:00:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:00:27 charger status: B
Okt 06 14:00:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:00:27 write chargeTotalImport=map[value:4183.496] (map[loadpoint:ABL EMH2 vehicle:Model3 (dienstlich)])
Okt 06 14:00:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:00:27 pv charge current: 4.72A = 0A + 4.72A (-3257W @ 3p)
Okt 06 14:00:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:00:27 site power -3257W <= -3000W enable threshold
Okt 06 14:00:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:00:27 pv enable in 0s
Okt 06 14:00:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:00:27 pv enable timer elapsed
Okt 06 14:00:27 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:27 sent JSON message to ABL_10022482: [2,"543900425","RemoteStartTransaction",{"connectorId":1,"idTag":"D66D75F1","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
Okt 06 14:00:28 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:28 received JSON message from ABL_10022482: [2, "29171979", "StartTransaction" ,{"connectorId":1,"idTag":"D66D75F1","timestamp":"2023-10-06T12:00:26.547+00:00","meterStart":4183496}]
Okt 06 14:00:28 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:28 sent JSON message to ABL_10022482: [3,"29171979",{"idTagInfo":{"status":"Accepted"},"transactionId":1}]
Okt 06 14:00:28 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:28 received JSON message from ABL_10022482: [2, "60377054", "MeterValues" ,{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2023-10-06T12:00:26.547+00:00","sampledValue":[{"value":"4183.496","context":"Transaction.Begin","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"kWh"}]}]}]
Okt 06 14:00:28 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:28 sent JSON message to ABL_10022482: [3,"60377054",{}]
Okt 06 14:00:33 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:33 received JSON message from ABL_10022482: [2, "86714611", "StatusNotification" ,{"connectorId":1,"status":"SuspendedEV","errorCode":"NoError","info":"none","timestamp":"2023-10-06T12:00:32.821+00:00","vendorId":"ABL","vendorErrorCode":"none"}]
Okt 06 14:00:33 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:33 sent JSON message to ABL_10022482: [3,"86714611",{}]
Okt 06 14:00:35 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:35 received JSON message from ABL_10022482: [2, "98234863", "StatusNotification" ,{"connectorId":1,"status":"Charging","errorCode":"NoError","info":"none","timestamp":"2023-10-06T12:00:34.743+00:00","vendorId":"ABL","vendorErrorCode":"none"}]
Okt 06 14:00:35 srvdeb evcc[1633313]: [ocpp  ] TRACE 2023/10/06 14:00:35 sent JSON message to ABL_10022482: [3,"98234863",{}]
Okt 06 14:01:27 srvdeb evcc[1633313]: [lp-1  ] ERROR 2023/10/06 14:01:27 charger enable: ocpp message (543900425): GenericError - Request timed out
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write greenShareHome=map[value:1] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write greenShareLoadpoints=map[value:1] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [site  ] DEBUG 2023/10/06 14:01:27 ----
Okt 06 14:01:27 srvdeb evcc[1633313]: [http  ] TRACE 2023/10/06 14:01:27 GET http://srvdeb:8080/rest/items/ABL_Wallbox_Diagnose_Values_Ruckgabewert_PowerValue
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write tariffGrid=map[value:0.173] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write tariffFeedIn=map[value:0.173] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write tariffPriceHome=map[value:0.173] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write tariffCo2Home=map[value:0] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write tariffPriceLoadpoints=map[value:0.173] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write tariffCo2Loadpoints=map[value:0] (map[])
Okt 06 14:01:27 srvdeb evcc[1633313]: [http  ] TRACE 2023/10/06 14:01:27 {"link":"http://srvdeb:8080/rest/items/ABL_Wallbox_Diagnose_Values_Ruckgabewert_PowerValue","state":"0.0","stateDescription":{"pattern":"%s","readOnly":true,"options":[]},"editable":true,"type":"Number","name":"ABL_Wallbox_Diagnose_Values_Ruckgabewert_PowerValue","label":"ABL Wallbox Power Value","category":"","tags":["Point"],"groupNames":[]}
Okt 06 14:01:27 srvdeb evcc[1633313]: [lp-1  ] DEBUG 2023/10/06 14:01:27 charge power: 0W
Okt 06 14:01:27 srvdeb evcc[1633313]: [http  ] TRACE 2023/10/06 14:01:27 GET http://srvdeb:8080/rest/items/PVGeneratorPower
Okt 06 14:01:27 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:27 write chargePower=map[value:0] (map[loadpoint:ABL EMH2 vehicle:Model3 (dienstlich)])
Okt 06 14:01:27 srvdeb evcc[1633313]: [http  ] TRACE 2023/10/06 14:01:27 {"members":[{"link":"http://srvdeb:8080/rest/items/QCells_Inverter_HTTP_API_Bindung_PV2Power","state":"1744","stateDescription":{"readOnly":false,"options":[]},"type":"Number","name":"QCells_Inverter_HTTP_API_Bindung_PV2Po [... cut ...]
Okt 06 14:01:57 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:57 write vehicleSoc=map[value:53.05879999999998] (map[loadpoint:ABL EMH2 vehicle:Model3 (dienstlich)])
Okt 06 14:01:57 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:57 write vehicleTargetSoc=map[value:80] (map[loadpoint:ABL EMH2 vehicle:Model3 (dienstlich)])
Okt 06 14:01:57 srvdeb evcc[1633313]: [influx] TRACE 2023/10/06 14:01:57 write chargeRemainingEnergy=map[value:22451.000000000015] (map[loadpoint:ABL EMH2 vehicle:Model3 (dienstlich)])
Okt 06 14:01:57 srvdeb evcc[1633313]: [lp-1  ] ERROR 2023/10/06 14:01:57 charger: cannot enable: transaction already running

```

Maddogtet, 2023-10-06:
Man sieht im den Log wie er die Transaktion startet und auch bestätigt. Ganz unten im Log steht aber, dass er dann nochmal versucht die Transaktion zu starten :confused:

andig, 2023-10-06:
Bitte Log ab Start, ohne influx/http- das brauchts nicht.

closed 2023-10-06 by andig

Maddogtet, 2023-10-10:
Hallo @andig 
im Anhang das Log, wie gewünscht gekürzt. 
Habe danach zurückgewechselt auf eine Vorgängerversion.
[evcc_2023-10-10.log](https://github.com/evcc-io/evcc/files/12861093/evcc_2023-10-10.log)

andig, 2023-10-10:
Deine ABL weigert sich hier schon einen Ladevorgang zu starten:

```
[ocpp  ] TRACE 2023/10/06 13:40:29 sent JSON message to ABL_10022482: [2,"2461205222","RemoteStartTransaction",{"connectorId":1,"idTag":"D66D75F1","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[lp-1  ] ERROR 2023/10/06 13:41:29 charger enable: ocpp message (2461205222): GenericError - Request timed out
```

Bevor das Problem nicht gelöst ist müssen wir über potenzielle Doppelstarts (die ich in diesem Log auch nicht sehe) nicht reden. Hier stimmt irgendwas überhaupt gar nicht.

andig, 2023-10-13:
Hast Du noch was rausgefunden?

Maddogtet, 2023-10-21:
@andig nein leider nicht. Nur das es leider immer noch nicht wieder funktioniert :/
Habe nochmal eine Log angehangen, vielleicht hilft das ja noch.
Hatte eine Ladung mit 119.3 laufen und habe auf 121.1 aktualisiert. Ladung hätte eigentlich gestoppt werden müssen. Lässt sich aber auch manuell nicht per EVCC stoppen.
[evcc_2023-10-21.log](https://github.com/evcc-io/evcc/files/13060840/evcc_2023-10-21.log)

andig, 2023-10-22:
In https://github.com/evcc-io/evcc/commit/cd3e03e9083fa5154a9a4324cf1a0839a0881467 nochmal ein Versuch das hin zu biegen. Bitte morgen im Nightly probieren.

Maddogtet, 2023-10-23:
Hallo @andig,

sieht sehr gut aus. Habe es kurz getestet und für mich sieht es so aus, als ob es wieder super funktioniert!
Danke, im Anhang nochmal das Log.
Viele Grüße
Mario
[evcc_2023-10-23.log](https://github.com/evcc-io/evcc/files/13069223/evcc_2023-10-23.log)