evcc-io/evcc #10213: TWC3 - Error message indicated following a manual charger disconnection while still charging

alexonbandit, 2023-10-06:
### Describe the bug

If I disconnect the car ,a Tesla Model 3, from my Tesla Wall Connector 3 while it is charging I receive the following error caption in the EVCC UI:

- charger disable: vehicle not capable of start/stop 

The associated log file error messages match the UI error messages.
The error is persistent. Also I confirmed that the TWC3 is still in the network and if it is still providing data. 
The car itself was also providing data (my wife was driving after the disconnection as depicted above)

The behavior was seen in the past as well. So it was not a one of 

### Steps to reproduce

1. Ensure the car is charging 
2. Disconnect the Charging cable manually
3. See UI / log

### Configuration details

```yaml
# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.85  # Hostname
    port: 502 # Port 
    
    #PV
  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    # id: 1
    host: 192.168.178.85 # Hostname
    port: 502 # Port 
    
    #Battery
  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    # id: 1
    host: 192.168.178.85 # Hostname
    port: 502 # Port 
    
    # Shelly Pro 3EM
  - name: my_grid_shelly
    type: template
    template: shelly-pro-3em
    usage: grid
    host: 192.168.178.88 # IP-Adresse oder Hostname

chargers:
  - name: my_charger
    type: template
    template: twc3
    host: 192.168.178.83 # IP-Adresse oder Hostname 

# vehicle definitions     
vehicles:
  - name: my_car
    type: template
    template: tesla
    title: Tesla Model 3 # Wird in der Benutzeroberfläche angezeigt (Optional)
    accessToken: ???
    refreshToken: ???
    capacity: 82 # Akkukapazität in kWh (Optional)
    mode: pv
    mincurrent: 6
    maxcurrent: 14
    minSoc: 40
    targetSoc: 70

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: my_grid_shelly # grid meter
    pv:
      - my_pv # list of pv inverters/ meters
    battery:
      - my_battery # list of battery meters
    #aux:
    #  - aux # list of auxiliary meters for adjusting grid operating point
  residualPower: 150 # additional household usage margin
  prioritySoc: 40 # give home battery priority up to this soc (empty to disable)
  bufferSoc: 50 # continue charging on battery above soc (0 to disable)
  bufferStartSoc: 60 # start charging on battery above soc (0 to disable)
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value
  smartCostLimit: 0 # set cost limit for automatic charging in PV mode

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    #meter: charge # charge meter
    mode: pv # charge mode (off, now, minpv, pv)
    #mode: "off" # set default charge mode, use "off" to disable by default if charger is publicly available
    # vehicle: car1 # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 14 # maximum charge current (default 16A)

    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.349 # EUR/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # EUR/kWh

# mqtt message broker
mqtt:
   broker: localhost:???
   # broker: alexonbandit.duckdns.org:1883
   topic: evcc # root topic for publishing, set empty to disable
   clientid: evcc
   user: ???
   password: ???
```

### Log details

```text
site  ] DEBUG 2023/10/06 14:56:50 battery power: 0W
[site  ] DEBUG 2023/10/06 14:56:50 grid power: -3469W
[site  ] DEBUG 2023/10/06 14:56:50 grid powers: [-696 -1394 -1378]W
[site  ] DEBUG 2023/10/06 14:56:50 grid currents: [-3.26 -6.18 -6.13]A
[site  ] DEBUG 2023/10/06 14:56:50 site power: -3319W
[lp-1  ] DEBUG 2023/10/06 14:56:50 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2023/10/06 14:56:50 charge currents: [0.4 0 0.2]A
[lp-1  ] DEBUG 2023/10/06 14:56:50 charger status: A
[lp-1  ] ERROR 2023/10/06 14:56:50 charger disable: vehicle not capable of start/stop
[site  ] DEBUG 2023/10/06 14:57:00 ----
[lp-1  ] DEBUG 2023/10/06 14:57:00 charge power: 0W
[site  ] DEBUG 2023/10/06 14:57:00 pv power: 4436W
[site  ] DEBUG 2023/10/06 14:57:00 battery soc: 100%
[site  ] DEBUG 2023/10/06 14:57:00 battery power: 0W
[site  ] DEBUG 2023/10/06 14:57:00 grid power: -3649W
[site  ] DEBUG 2023/10/06 14:57:00 grid powers: [-755 -1455 -1439]W
[site  ] DEBUG 2023/10/06 14:57:00 grid currents: [-3.51 -6.45 -6.4]A
[site  ] DEBUG 2023/10/06 14:57:00 site power: -3499W
[lp-1  ] DEBUG 2023/10/06 14:57:00 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2023/10/06 14:57:00 charge currents: [0.4 0 0.2]A
[lp-1  ] DEBUG 2023/10/06 14:57:00 charger status: A
[lp-1  ] ERROR 2023/10/06 14:57:00 charger disable: vehicle not capable of start/stop
[site  ] DEBUG 2023/10/06 14:57:10 ----
[lp-1  ] DEBUG 2023/10/06 14:57:10 charge power: 0W
[site  ] DEBUG 2023/10/06 14:57:10 pv power: 4620W
[site  ] DEBUG 2023/10/06 14:57:10 battery soc: 100%
[site  ] DEBUG 2023/10/06 14:57:10 battery power: 0W
[site  ] DEBUG 2023/10/06 14:57:10 grid power: -3813W
[site  ] DEBUG 2023/10/06 14:57:10 grid powers: [-812 -1509 -1492]W
[site  ] DEBUG 2023/10/06 14:57:10 grid currents: [-3.73 -6.67 -6.63]A
[site  ] DEBUG 2023/10/06 14:57:10 site power: -3663W
[lp-1  ] DEBUG 2023/10/06 14:57:10 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2023/10/06 14:57:10 charge currents: [0.4 0 0.2]A
[lp-1  ] DEBUG 2023/10/06 14:57:10 charger status: A
[lp-1  ] ERROR 2023/10/06 14:57:10 charger disable: vehicle not capable of start/stop
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.120.3

andig, 2023-10-06:
Error comes from the fact that at this point, the charger can not longer see the vehicle as it is not connected. That is- as far as disabling the charger- however not a problem. Seems we should only check for detected vehicle if charger status is not "disconnected" when disabling.

closed 2023-10-06 by andig

alexonbandit, 2023-10-06:
Hi @andig, thanks for the explanation. As you said it is not really a problem just an error caption.
But if your proposal to check if the connection status for disabling is feasible and not overcomplicated to implement it would be great. 
Thanks 😊

andig, 2023-10-06:
Already done. You can test tomorrow‘s nightly.