evcc-io/evcc #10215: go-e Re-Authenticate nach Phasenwechsel benötigt

tpatzig, 2023-10-06:
### Describe the bug

Hin und wieder beim automatischen Phasenwechsel, wenn das Laden kurz stoppt/pausiert, muss ich mich am Go-e charger neu authentifizieren. evcc sagt dann "Warten auf Auto..." und es geht erst weiter, wenn man in der go-e app auf "Authentifizieren" drückt. Ich habe es nun beim Wechsel von 1p -> 3p beobachtet.
Schön wäre, wenn ich mich nur einmal nach dem Kabel anstecken authentifizieren muss.

### Steps to reproduce

1.  Go-e charger flex gemini 
2. autom. Phasenumschaltung konfiguriert (loadpoints.phases: 0)
3. PV Überschussladen (loadpoints.mode: pv)

### Configuration details

```yaml
meters:
- type: template
  template: rct-power
  usage: grid
  host: 192.168.10.10
  name: grid1
- type: template
  template: rct-power
  usage: pv
  host: 192.168.10.10
  name: pv2
- type: template
  template: rct-power
  usage: battery
  host: 192.168.10.10
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: 192.168.10.11
  name: wallbox5

vehicles:
- type: template
  template: id
  title: ID.4
  icon: car
  user: xxx
  password: xxx
  capacity: 77
  phases: 3
  cache: 15m
  mode: pv
  minSoc: 20
  targetSoc: 80
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
```

### Log details

```text
[lp-1  ] DEBUG 2023/10/06 15:26:50 charge currents: [15.6 0 0]A
[lp-1  ] DEBUG 2023/10/06 15:26:50 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/06 15:26:50 charge total import: 57.201kWh
[lp-1  ] DEBUG 2023/10/06 15:26:50 charger status: C
[lp-1  ] DEBUG 2023/10/06 15:26:50 soc estimated: 70.59% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/10/06 15:26:50 vehicle soc: 71%
[lp-1  ] DEBUG 2023/10/06 15:26:50 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/10/06 15:26:50 vehicle range: 295km
[lp-1  ] DEBUG 2023/10/06 15:26:50 available power 4275W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/10/06 15:26:50 phase scale3p in 30s
[lp-1  ] DEBUG 2023/10/06 15:26:50 pv charge current: 18.9A = 16A + 2.86A (-657W @ 1p)
[lp-1  ] DEBUG 2023/10/06 15:27:20 charge power: 3616W
[lp-1  ] DEBUG 2023/10/06 15:27:20 charge currents: [15.6 0 0]A
[lp-1  ] DEBUG 2023/10/06 15:27:20 detected active phases: 1p
[lp-1  ] DEBUG 2023/10/06 15:27:20 charge total import: 57.232kWh
[lp-1  ] DEBUG 2023/10/06 15:27:20 charger status: C
[lp-1  ] DEBUG 2023/10/06 15:27:20 soc estimated: 70.63% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/10/06 15:27:20 vehicle soc: 71%
[lp-1  ] DEBUG 2023/10/06 15:27:20 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/10/06 15:27:20 vehicle range: 295km
[lp-1  ] DEBUG 2023/10/06 15:27:20 available power 4586W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/10/06 15:27:20 phase scale3p in 0s
[lp-1  ] DEBUG 2023/10/06 15:27:20 !! scalePhases: GetPhases 1p <> 3p wanted
[lp-1  ] DEBUG 2023/10/06 15:27:20 switched phases: 3p
[lp-1  ] DEBUG 2023/10/06 15:27:20 phase timer inactive
[lp-1  ] DEBUG 2023/10/06 15:27:20 pv charge current: 17.4A = 16A + 1.41A (-970W @ 3p)
[lp-1  ] DEBUG 2023/10/06 15:27:50 charge power: 0W
[lp-1  ] DEBUG 2023/10/06 15:27:51 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/06 15:27:51 charge total import: 57.233kWh
[lp-1  ] DEBUG 2023/10/06 15:27:51 charger status: B
[lp-1  ] INFO 2023/10/06 15:27:51 stop charging <-
[lp-1  ] DEBUG 2023/10/06 15:27:51 wake-up timer: start
[lp-1  ] DEBUG 2023/10/06 15:27:51 charge total import: 57.233kWh
[lp-1  ] DEBUG 2023/10/06 15:27:52 soc estimated: 70.63% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/10/06 15:27:52 vehicle soc: 71%
[lp-1  ] DEBUG 2023/10/06 15:27:52 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/10/06 15:27:52 vehicle range: 298km
[lp-1  ] DEBUG 2023/10/06 15:27:52 pv charge current: 7.15A = 0A + 7.15A (-4934W @ 3p)
[lp-1  ] DEBUG 2023/10/06 15:27:52 site power -4934W <= 0W enable threshold
[lp-1  ] DEBUG 2023/10/06 15:27:52 pv enable timer start: 1m0s
[lp-1  ] DEBUG 2023/10/06 15:27:52 pv enable in 1m0s
[lp-1  ] DEBUG 2023/10/06 15:28:20 charge power: 0W
[lp-1  ] DEBUG 2023/10/06 15:28:20 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/06 15:28:20 charge total import: 57.233kWh
[lp-1  ] DEBUG 2023/10/06 15:28:20 charger status: B
[lp-1  ] DEBUG 2023/10/06 15:28:20 pv charge current: 7.25A = 0A + 7.25A (-5001W @ 3p)
[lp-1  ] DEBUG 2023/10/06 15:28:20 site power -5001W <= 0W enable threshold
[lp-1  ] DEBUG 2023/10/06 15:28:20 pv enable in 32s
[lp-1  ] DEBUG 2023/10/06 15:28:20 pv enable timer remaining: 32s
[lp-1  ] DEBUG 2023/10/06 15:28:50 charge power: 0W
[lp-1  ] DEBUG 2023/10/06 15:29:20 charge power: 0W
[lp-1  ] DEBUG 2023/10/06 15:29:20 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/06 15:29:20 charge total import: 57.233kWh
[lp-1  ] DEBUG 2023/10/06 15:29:20 charger status: B
[lp-1  ] DEBUG 2023/10/06 15:29:20 pv charge current: 6.95A = 0A + 6.95A (-4796W @ 3p)
[lp-1  ] DEBUG 2023/10/06 15:29:20 site power -4796W <= 0W enable threshold
[lp-1  ] DEBUG 2023/10/06 15:29:20 pv enable in 0s
[lp-1  ] DEBUG 2023/10/06 15:29:20 pv enable timer elapsed
[lp-1  ] DEBUG 2023/10/06 15:29:20 max charge current: 6A
[lp-1  ] DEBUG 2023/10/06 15:29:20 charger enable
[lp-1  ] DEBUG 2023/10/06 15:29:20 wake-up timer: start
[lp-1  ] DEBUG 2023/10/06 15:29:20 wake-up vehicle
[lp-1  ] DEBUG 2023/10/06 15:29:50 charge power: 4156W
[lp-1  ] DEBUG 2023/10/06 15:29:51 charge currents: [5.89 5.97 6]A
[lp-1  ] DEBUG 2023/10/06 15:29:51 charge total import: 57.255kWh
[lp-1  ] DEBUG 2023/10/06 15:29:51 charger status: C
[lp-1  ] INFO 2023/10/06 15:29:51 start charging ->
[lp-1  ] DEBUG 2023/10/06 15:29:51 wake-up timer: stop
[lp-1  ] DEBUG 2023/10/06 15:29:51 soc estimated: 70.63% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/10/06 15:29:51 vehicle soc: 71%
[lp-1  ] DEBUG 2023/10/06 15:29:51 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/10/06 15:29:51 vehicle range: 298km
[lp-1  ] DEBUG 2023/10/06 15:29:51 pv charge current: 6.32A = 6A + 0.319A (-220W @ 3p)
[lp-1  ] DEBUG 2023/10/06 15:29:51 pv timer reset
[lp-1  ] DEBUG 2023/10/06 15:29:51 pv timer inactive
[lp-1  ] DEBUG 2023/10/06 15:30:20 charge power: 4153W
[lp-1  ] DEBUG 2023/10/06 15:30:20 charge currents: [5.93 6 5.97]A
[lp-1  ] DEBUG 2023/10/06 15:30:20 detected active phases: 3p
[lp-1  ] DEBUG 2023/10/06 15:30:20 charge total import: 57.287kWh
[lp-1  ] DEBUG 2023/10/06 15:30:20 charger status: C
```

### What type of operating system are you running?

Docker container

### Version

v0.120.3

closed 2023-10-13 by andig

Schlichi83, 2023-12-08:
Ich habe das Problem auch. Das führt auch dazu, dass das Ziel-Laden in der Hälfte der Fälle nicht funktioniert, weil evcc eine Authentifizierung per RFID verlangt. Das Problem tritt bei mir allerdings unregelmäßig auf.

premultiply, 2023-12-09:
Bug in der Firmware?

Schlichi83, 2023-12-09:
Gut möglich. Ich bin auf FW 0.55.7
Könnte mal downgraden (0.55.5) oder auf die Beta 0.55.8 hoch.