evcc-io/evcc #10220: Improve documentation of domestic water heat pump use case

MikeTheTux, 2023-10-06:
By luck I found the description how to use EVCC to control my heat pump (in addition to my wallbox). I would like to share the following improvement ideas:

1. The heat pump use case is not well "advertised". E.g. it is not mentioned in [Home]( https://docs.evcc.io/docs/Home)
2. The setup is described in [Sonstiges](https://docs.evcc.io/docs/guides/faq) - a better title should be used for the page
3. [Sonstiges](https://docs.evcc.io/docs/guides/faq) is incomplete:
3.1. `chargers` misses an explanation of the `standbypower` parameter (`<0` vs. `>=0`)
3.2. In addition the parameters `integrateddevice`, `heating` as well as `icon` are not mentioned
3.3. I found nowhere a description, what icons can be used.
3.4. In the `loadpoint` section, the `priority` could be explained as useful option - Charged Car v.s. Warm Water ;-)
4. `meters` -> `WP_Pwr` should also support a `standbypower` parameter. Without, the standby of the heat pump (~15 Watt in my case) is interpreted as "Charging" and accumulates as charged energy.

andig, 2023-10-06:
Great input, thank you!

MikeTheTux, 2023-10-07:
For 4. I implemented a workaround using the following logic in openHAB and transmit the result via MQTT to EVCC:

```
if ( power < 100W )
{
   power = 0W
}
else
{
   power = power - 15W // typical standby
}
```

andig, 2024-01-13:
/cc @VolkerK62

VolkerK62, 2024-01-13:
@MikeTheTux 
This example was originally intended for use in Shellys, not specifically for heat pumps.
All parameters are described in the configuration area.
You are right about the poll mode, that should be changed.

For 4: `standbypower` on the chargemeter is still not necessary, it is enough, if it is entered on the charger. Try using `standbypower: 100`.

By the way, he documentary can be edited by any user 😄

andig, 2024-01-23:
Additional enhancements from https://github.com/evcc-io/evcc/discussions/11822#discussioncomment-8224247:

> Ja, Einerschritte macht hier durchaus Sinn. Kann ich mich gerne drum kümmern. Anpassung der Skala (lower/upper) ist schwieriger. Da haben wir aktuell (noch) keinen guten Ort für. Wäre etwas speziell für Heater. Da wir gerade dabei sind die yaml-Settings zu reduzieren bzw. in der UI zu migrieren würde ich das auch erst nach der Migration sehen.

MikeTheTux, 2024-01-27:
> For 4: `standbypower` on the chargemeter is still not necessary, it is enough, if it is entered on the charger. Try using `standbypower: 100`.

Muss ich am Charger nicht `-1` setzen, um in den Static-Switch Modus zu gelangen?
https://docs.evcc.io/docs/devices/chargers/#schaltbare-steckdosen

Ist die `poll-mode` Änderung aus https://github.com/evcc-io/docs/pull/515 nicht redundant aufgrund der Änderungen aus https://github.com/evcc-io/evcc/pull/10284?

MikeTheTux, 2024-01-29:
> > Ist die poll-mode Änderung aus [evcc-io/docs#515](https://github.com/evcc-io/docs/pull/515) nicht redundant aufgrund der Änderungen aus #10284?
> 
> Stimmt, daran hab ich gar nicht mehr gedacht ;-)

Ich denke, die Änderung aus https://github.com/evcc-io/docs/pull/515 sollte zurückgenommen werden. Damit entsteht der Eindruck, dass der `poll` Parameter immer noch eine Relevanz hat. Vielmehr sollte das via https://github.com/evcc-io/evcc/pull/10284 implementierte Verhalten beschrieben werden.

MikeTheTux, 2024-02-11:
> > Muss ich am Charger nicht -1 setzen, um in den Static-Switch Modus zu gelangen?
> 
> Wenn du ein meter nutzt, kannst du auch positive Werte nehmen.

@VolkerK62: Folgende Config, mit positiver `standbypower` hat bei mir nicht funktioniert:

```
chargers:
  - name: heatpump_switch
    type: template
    template: shelly
    host: 192.168.69.152
    standbypower: 15
    integrateddevice: true
    heating: true
    icon: waterheater

meters:
- name: heatpump_power
  type: custom
  power:
    source: mqtt
    topic: heatpump/power

loadpoints:
- title: Heatpump
  charger: heatpump_switch
  meter: heatpump_power
  vehicle: heatpump
  mode: pv
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0
    delay: 15m              # Laufzeit mindestens 15 Minuten
  guardduration: 15m        # Pause mindestens 15 Minunten
  priority: 10

vehicles:
  - name: heatpump
    type: custom
    title: Heatpump
    icon: waterheater
    phases: 1
    onIdentify:
      mode: pv
    soc:
      source: mqtt
      topic: heatpump/temperature
```

Warte auf Heizung, Geladene Energie 0 kWh ...

[image]

VolkerK62, 2024-02-15:
Ich weiß jetzt warum.  Siehe #12276  Hatte ich so nicht auf dem Schirm

michiproep, 2024-04-26:
Ich würde mich hier auch noch gerne dranhängen.
Das mit dem ```standbypower``` ist die eine Sache. Was jetzt noch dazukommt, ist der ```maxCurrent```, den evcc meldet.
Wenn zB die Wärmepumpe nachmittags ihre Arbeit erledigt hat und aufgrund ihrer Hysteresen (Warmwasser und/oder Heizung) jetzt im StandBy ist und es jetzt Abend wird und somit die PvErzeugung sinkt, meldet evcc weiterhin den maxmimalen maxCurrent für den Loadpoint. Dies führt dazu, dass der SG-/PV-Ready-Kontakt aktiv bleibt und die WP bei unterschreiten der Hysterese wieder mit Vollgas anfängt zu arbeiten.

Bei mir ist der WP Loadpoint auf min. 2,8kw und max. 4,6kw eingestellt.

Müsste der Loadpoint in diesem Fall (Screenshot) nicht disabled werden?

[image]

andig, 2024-07-06:
> Was jetzt noch dazukommt, ist der maxCurrent, den evcc meldet.

@michiproep der wird nicht gemeldet, sondern eingestellt. Dein Screenshot zeigt es: Du hast Überschuss, also wird geladen. Ob die WP das aufgrund Hysterese gerade konsumiert oder nicht ist da erstmal egal.

> Dies führt dazu, dass der SG-/PV-Ready-Kontakt aktiv bleibt und die WP bei unterschreiten der Hysterese wieder mit Vollgas anfängt zu arbeiten.

Das Hauptproblem hier ist m.E., dass evcc den Leistungsbedarf der WP nicht kennt. Du müsstest dafür sorgen, das indirekt mittels `minCurrent` am Loadpoint einzustellen. Wenn die zum WP-Bedarf passt dann wird der LP auch korrekt abgeschaltet.

Das scheint mir hier aber OT- gerne separate Diskussion dazu.

andig, 2024-07-06:
> 3.4 The default poll.mode is charging. You need it to set on always/connected, see

@MikeTheTux this should be obsolete now since charger can supply `soc` and will always be polled irrespective of poll mode.

For the rest- since the documentation was greatly enhanced- I'm not sure which points still apply.

andig, 2024-07-06:
> meters -> WP_Pwr should also support a standbypower parameter. Without, the standby of the heat pump (~15 Watt in my case) is interpreted as "Charging" and accumulates as charged energy.

@MikeTheTux https://github.com/evcc-io/evcc/pull/14546 allows defining power and energy on the custom charger. If you want to make status dependent on power, that can be done by using the `go` or `javascript` plugins to intercept the meter values and use those for calculating status. Not particularly pretty, but possible (see https://github.com/evcc-io/evcc/discussions/12761#discussioncomment-9944308 for a technical example).

michiproep, 2024-07-10:
Hi @andig 
Du schreibst:
> @michiproep ... Dein Screenshot zeigt es: Du hast Überschuss, also wird geladen. ...

Das kann ich so nicht erkennen. Hab ich da einen Denkfehler?
Im Screenshot sieht man, dass ein anderer Ladepunkt schon soviel zieht, dass die Batterie auch schon angezapft wird. Zu dem Zeitpunkt war der Batterie-Status im Bereich "Laden hat Vorrang".
Der MinCurrent des Ladepunkts "Warmwasser" steht auf 4A / 2,8kw ; Max: 7A / 4,8kw
Warum STELLT evcc den Ladepunkt in dem Fall auf 7A wenn von vorn herein nur maximal 1,8kw verfügbar sind?

andig, 2024-07-11:
> Das scheint mir hier aber OT- gerne separate Diskussion dazu.

andig, 2024-11-24:
> The heat pump use case is not well "advertised". E.g. it is not mentioned in [Home](https://docs.evcc.io/docs/Home)

@naltatis could you link the PR to change that here?

andig, 2025-01-03:
Update: as short-term measure we're planning to remove the soc slider for heating to better support the SG Ready use case. Are there any objections?

michiproep, 2025-01-03:
for every heating device? or just SG ready? Will I still be able to set the soc (temperature) via API?
because I need it for my heat pump use cases.

andig, 2025-01-03:
@michiproep which use cases are these?

michiproep, 2025-01-03:
Well, the room thermostats sync with the target soc. So, we can control the room temperature via thermostat and evcc.
The loadpoints/integrated devices ( e.g. infrared heater, towel radiator, etc) use the target soc to stop heating.
Not sure, if this answers your question.

MikeTheTux, 2025-01-03:
I use different target SOCs / temperatures on my heating device for two scenarios:

1. **Winter Heating:** During winter, my warm water is also heated by my pellet heating system. To avoid using pellets instead of PV energy, I maintain a higher target temperature. If the warm water drops below a certain threshold, the pellet system activates.
2. **PV Forecast Adjustment:** Depending on the PV energy forecast, I adjust the target temperatures. When there is limited PV energy available, I set a lower target temperature to allocate more energy for charging the car.

michiproep, 2025-01-03:
Similar here. I just didn't want to write the whole story.

MikeTheTux, 2025-02-23:
When switching to the new [heatpump guideline](https://docs.evcc.io/docs/devices/heating) (thanks for creating!), it seem that the parameter `integrateddevice` is specified at the wrong place: It has to be at [chargers](https://docs.evcc.io/docs/reference/configuration/chargers#integrateddevice), not at [loadpoints ](https://docs.evcc.io/docs/reference/configuration/loadpoints#poll).

```
loadpoints:
  - title: Meine Wärmepumpe
    charger: heatpump_control
    meter: heatpump_power
    integrateddevice: true # Fahrzeugauswahl deaktivieren
```

andig, 2025-02-23:
Please note that we're also about to add support for power-enveleope controlled heatpumps in   https://github.com/evcc-io/evcc/pull/19008.

MikeTheTux, 2025-02-23:
With the configuration from [heatpump guideline](https://docs.evcc.io/docs/devices/heating) I have the following issue:

Situation:

1. There is no pv surplus
2. `Heatpump_control` is in `normal` (non-`boost`) mode
3. Water temp drops under a certain limit or legionella program is due:
3.1. Waterpump activates, `heatpump_power` reports used power
3.2. Still, the used energy is not accumulated under 'charged':

[image]

-> The Energy shall be accumulated, independent if the sgready statemachine is in `normal` or in `boost` state.

MikeTheTux, 2025-02-23:
Can someone help me with with deriving the state from the power? I have some issues with the yaml syntax ...

Can it be easy as ...
```
    getmode:
      source: calc
      add:
        - source: const
          value: 1
        - source: mqtt
          topic: heatpump/power
```
... or do I need a more complex logic like in https://github.com/evcc-io/evcc/discussions/12761#discussioncomment-9944308 ?

MikeTheTux, 2025-02-23:
Co-Pilot proposes the following, but kex eq is not supported

[image]

MikeTheTux, 2025-02-23:
When implementing the above logic the status in the gui switches between "Connected" (1) and "Heating" (2). Which is in line with my expectation.
Unfortunately this goes along with the following warning:
`Heatpump: charger out of sync: expected disabled, got enabled`

@andig: Why is the current implementation of EVCC accumulating the heating power, only in case of BOOST? Why is it not per default accumulating also the energy in NORMAL mode?

andig, 2025-02-25:
> Why is the current implementation of EVCC accumulating the heating power, only in case of BOOST?

Because- for time being- it is following normal charger logic which only consumes energy while charging. This will be something we'll need to revisit.

MikeTheTux, 2025-02-26:
> > Why is the current implementation of EVCC accumulating the heating power, only in case of BOOST?
> 
> Because- for time being- it is following normal charger logic which only consumes energy while charging. This will be something we'll need to revisit.

Got it - the "own brain" in the heat pump is the main difference between a heating device and a wallbox.
But also in the wallbox use case, the car can decide to activate for example the parking heater. Will this energy be counted for the current charging session? Also in this case, the decision to consume it outside of EVCC.

MikeTheTux, 2025-04-15:
> In the meanwhile I replaced the external logic, by the following GO script directly in EVCC:
> 
> ```
> go:
>   - vm: sgready
>     script: |
>       import "fmt"
>       var globalPower float64
> 
> chargers:
>   - name: heatpump_switch
>     type: sgready
>     setmode:
>       source: switch
>       switch:
>       - case: 1 # normal
>         set:
>           source: const
>           value: false
>           set:
>             source: charger
>             config:
>               type: template
>               template: shelly # Boost-Kontakt der Wärmepumpe
>               host: 192.168.69.152
>       - case: 2 # boost
>         set:
>           source: const
>           value: true
>           set:
>             source: charger
>             config:
>               type: template
>               template: shelly # Boost-Kontakt der Wärmepumpe
>               host: 192.168.69.152
>       - case: 3 # dimm (not supported, yet)
>         set:
>           source: error
>           error: ErrNotAvailable
>     getmode:
>       source: go
>       vm: sgready
>       script: |
>         HP_SGready := 1
>         if HP_Shelly == 1 || globalPower > 0.0 {
>            HP_SGready = 2
>         }
>         // fmt.Println("HP_Shelly:", HP_Shelly, "globalPower:", globalPower, "-> HP_SGready:", HP_SGready)
>         HP_SGready
>       in:
>       - name: HP_Shelly
>         type: int
>         config: 
>           source: charger
>           config:
>             type: template
>             template: shelly # Boost-Kontakt der Wärmepumpe
>             host: 192.168.69.152
>     temp:
>       source: http              # read temperature from heating system
>       uri: http://192.168.69.25/ext/daqdata.cgi?key=1234abcd
>       jq: .[8]
>                 
> meters:
> - name: heatpump_power
>   type: custom
>   power:
>     source: go
>     vm: sgready
>     script: |
>       power := 0.0
>       if Shelly_Power >= 20.0 {  // ignore standby
>         power = Shelly_Power
>       }
>       globalPower = power
>       // fmt.Println("Shelly_Power:", Shelly_Power, "-> globalPower:", globalPower)
>       power
>     in:
>     - name: Shelly_Power
>       type: float
>       config:
>         source: http
>         uri: http://192.168.69.102/status
>         jq: .meters[0].power
> ```
> 
> This avoids the external dependency of EVCC towards the home automation.

@andig, kann es sein, dass das GO Scripting ein MemoryLeak verursacht?
Meine EVCC Instanz hat 6GB RAM verwendet, nachdem sie 3 Wochen gelaufen ist ...

andig, 2025-04-15:
Leider ja: https://github.com/traefik/yaegi/issues/1678