evcc-io/evcc #10222: completed planned charge, shows as pending in the past

GrimmiMeloni, 2023-10-07:
### Describe the bug

I charged the car via the planner last night. Based on GSI it completed the charge around 4 o'clock.
This morning I see that the UI still shows the charge as pending but in the past. The rest of the UI shows 100% SOC is reached, so the planned charge was successful.

[image]

(The screenshot is taken just now, roughly 7:30 in the morning)

@naltatis I assume this is probably more in your area. I have captured a full log, but given all the Easee goodness keeping me on my toes it is impossible to sanitize for public posting. Let me know if you need it, and I will send it over privately.

### Steps to reproduce

1. Plan a charge
2. Have it complete
3. UI shows the above

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  easee: trace

# unique installation id
plant: xxx
sponsortoken: xxx

interval: 15s # control cycle interval

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.3.38  
  password: xxx
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.3.38  
  password: xxx
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.3.38  
  password: xxx
  name: battery3

site:
  title: Zu Hause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
  prioritySoc: 70
  bufferSoc: 80

chargers:
  - name: Easee
    type: template
    template: easee
    user: xxx
    password: xxx
    charger: xxx
    timeout: 20s
    authorize: true

loadpoints:
  - title: Carport # display name for UI
    charger: Easee
    mode: off
    #mode: pv
    resetOnDisconnect: true
    maxcurrent: 32
    #maxcurrent: 16

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.3796 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.082 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: "33106"

vehicles:
  - name: Enyaq
    type: template
    template: enyaq
    title: Enyaq
    user: xxx
    password: xxx
    vin: xxx
    capacity: 77
    mode: pv 

mqtt:
  broker: ds218.fritz.box:1883
  topic: evcc
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Docker container

### Version

the usual master + pending PRs frankenstein *sigh*

andig, 2023-10-08:
@naltatis worauf triggert die Anzeige? Im Log ist der Plan weg wie erwartet?

andig, 2023-10-08:
@GrimmiMeloni was sagt /api/state zu diesem Zeitpunkt?

GrimmiMeloni, 2023-10-08:
> @GrimmiMeloni was sagt /api/state zu diesem Zeitpunkt?

Habe ich (leider) nicht geprüft. Ich habe das vollständige Log von gestern morgen weggesichert. Finde ich die Info ggf. dort?

mdkeil, 2023-10-11:
Dieses "Problem" hat auch weitere negative Seiteneffekte.. ich hatte heute Nacht eine Ladeplanung (Tibber-Preise) laufen und der Ziel-SOC (60%) wurde problemlos vor der eingestellten Abfahrtzeit erreicht.. Ich hatte heute morgen dann nochmal evcc neugestartet und der Ziel-SOC des Fahrzeuges wurde auf den in der evvc.yaml eingestellten 80% zurückgesetzt. Dies führte dazu, dass der Ladevorgang wieder direkt gestartet wurde, da die Abfahrtzeit noch nicht erreicht war und für die Ladeplanung ein neuer Ziel-SOC gesetzt wurde.

All-Ex-U, 2023-10-11:
Ein verwandtes Thema dazu habe ich auch:

Der Ziellade-Termin liegt in der Vergangenheit (hier heute 12:00 Uhr für SoC 60%), es ist jetzt 14:00 Uhr und der SoC beträgt inzwischen 64%, da ich nach dem Zielladen den Schieber auf 65% gestellt habe. Es wird weiterhin der in der Vergangenheit liegende Abfahrts-Termin angezeigt.
Erwartung: In der Vergangenheit liegende Ziellade-Termine sollte nicht angezeigt werden.

[image]

Ich verwende v.0.120.3 (kein nightly). Hier die API-Ausgabe für state:

```json{
	"result": {
		"auth": {
			"vehicles": {}
		},
		"batteryConfigured": false,
		"bufferSoc": 0,
		"bufferStartSoc": 0,
		"currency": "EUR",
		"greenShareHome": 0.7200398686679175,
		"greenShareLoadpoints": 0,
		"gridConfigured": true,
		"gridEnergy": 3181586.2,
		"gridPower": 191,
		"homePower": 682.24,
		"loadpoints": [
			{
				"chargeCurrent": 0,
				"chargeCurrents": [
					0,
					0,
					0
				],
				"chargeDuration": 8035000000000,
				"chargePower": 0,
				"chargeRemainingDuration": 0,
				"chargeRemainingEnergy": 629.5555555555613,
				"chargeTotalImport": 162.968,
				"chargeVoltages": [
					233.10000610351563,
					232.39999389648438,
					231.5
				],
				"chargedEnergy": 7673.999999999978,
				"chargerFeatureHeating": false,
				"chargerFeatureIntegratedDevice": false,
				"chargerIcon": null,
				"charging": false,
				"climaterActive": false,
				"connected": true,
				"connectedDuration": 0,
				"enabled": false,
				"guardAction": "inactive",
				"guardRemaining": 0,
				"maxCurrent": 16,
				"minCurrent": 6,
				"minSoc": 20,
				"mode": "pv",
				"phaseAction": "inactive",
				"phaseRemaining": 0,
				"phasesActive": 3,
				"phasesConfigured": null,
				"phasesEnabled": 3,
				"planActive": false,
				"planProjectedStart": "0001-01-01T00:00:00Z",
				"priority": 0,
				"pvAction": "inactive",
				"pvRemaining": 0,
				"sessionCo2PerKWh": 167.01361403090136,
				"sessionEnergy": 7673.999999999978,
				"sessionPrice": 2.8457985963814966,
				"sessionPricePerKWh": 0.37083640818106658,
				"sessionSolarPercentage": 1.215661113195097,
				"targetEnergy": 0,
				"targetSoc": 65,
				"targetTime": "2023-10-11T10:00:00Z",
				"title": "Wallbox",
				"vehicleCapacity": 77,
				"vehicleDetectionActive": false,
				"vehicleFeatureOffline": false,
				"vehicleIcon": "car",
				"vehicleOdometer": 861,
				"vehiclePresent": true,
				"vehicleRange": 294,
				"vehicleSoc": 64.26415584415584,
				"vehicleTargetSoc": 100,
				"vehicleTitle": "ID4"
			}
		],
		"prioritySoc": 0,
		"pv": [
			{
				"power": 491.24,
				"energy": 26.7457
			}
		],
		"pvConfigured": true,
		"pvEnergy": 26.7457,
		"pvPower": 491.24,
		"residualPower": -4100,
		"savingsAmount": 0.5611475602098968,
		"savingsEffectivePrice": 0.37187126496210695,
		"savingsGridCharged": 157.52750050310119,
		"savingsSelfConsumptionCharged": 1.4947990415820363,
		"savingsSelfConsumptionPercent": 0.9399933505313303,
		"savingsSince": "2023-09-20T10:38:06+02:00",
		"savingsTotalCharged": 159.02229954468323,
		"siteTitle": "Zuhause",
		"smartCostLimit": 145,
		"smartCostType": "co2",
		"sponsor": "All-Ex-U",
		"sponsorTokenExpires": 0,
		"tariffCo2": 163,
		"tariffCo2Home": 45.63350140712945,
		"tariffCo2Loadpoints": 163,
		"tariffFeedIn": 0,
		"tariffGrid": 0.3754,
		"tariffPriceHome": 0.10509703330206379,
		"tariffPriceLoadpoints": 0.3754,
		"vehicles": [
			"ID4"
		],
		"version": "0.120.3"
	}
}`

andig, 2023-10-16:
Sollte in 0.121.1 behoben sein.

closed 2023-10-20 by andig